photobooth-app/photobooth-app #1: Missing directory after install

yaazzz, 2022-12-06:
Hi,  after installing :
- data, log and web folders are missing.
- The transisitions module is missing in the install description.
- ./config/config.json is missing
- how to disable the WS2812

Not real issues ;)

yaazzz, 2022-12-06:
Thanks for reactivity. Well i
- if web and log and log/qbooth.log are not existing then the app do not start
- I have to install pynput module
- when I access to http://localhost:8000/index.html I get {"detail":"Not Found"
- When I try to take a picture from photobooth I get:
Started server process [20217]
```
INFO:     Waiting for application startup.
INFO:     Application startup complete.
INFO:     Uvicorn running on http://0.0.0.0:8000 (Press CTRL+C to quit)
INFO:     127.0.0.1:47732 - "GET /cmd/capturePrepare HTTP/1.1" 404 Not Found
INFO:     127.0.0.1:47734 - "POST /capture HTTP/1.1" 405 Method Not Allowed
```
Seems to be a really good app. Thank so much for sharing this with us.

mgrl, 2022-12-06:
When you use the webfrontend you can "persist" your config. Then you have a config.json with all settings and can edit the file directly.

yaazzz, 2022-12-06:
Whaoouuu!!!! this is very promising. 
Thanks. Remove everything. Restart the install. Working fine now ;)

closed 2022-12-06 by yaazzz

mgrl, 2022-12-06:
Glad it works. I used the setup earlier on 2 events with no problems. Let me know if you run into issues…