photobooth-app/photobooth-app #4: Green screen

loudjoudi, 2023-02-24:
Hi,

I try to install again as you made lot of improvement.
The script break my arduno 64mp cam. But nevermind I reinstall from scratch and try to install manually.
I have a green screen when using the application. Is there anything to do to use the arduno_64mp?

Regards.

mgrl, 2023-02-24:
Script updates all packages, it is likely to break the Arducam installations by simple system updates. There are some complaints about this in Arducam forum currently.
Please go to localist:8000 and check the config tab backends. Change backend from simulation to picam2.

loudjoudi, 2023-02-24:
Well, did it but now I have no preview. I don't know how to troubleshoot the issue.

mgrl, 2023-02-24:
this gives the last 100 lines console log from service:
`journalctl --unit=imageserver.service -n 100 --no-pager`

also in imageserver/log there should be logfiles (the one without a number is the most current one, if any number suffixes exist).

and if the service is working regular, there are messages logged in the webfrontend http://HOSTNAME:8000/#/admin/status

mgrl, 2023-02-24:
also good to start directly from console to see whats going on.
Stop service if installed:
`systemctl --user stop imageserver`
and in directory imageserver start
`python start.py`

loudjoudi, 2023-02-24:
Here is what I get from the console start:
`
INFO:     Started server process [1899]
INFO:     Waiting for application startup.
INFO:     Application startup complete.
INFO:     Uvicorn running on http://0.0.0.0:8000 (Press CTRL+C to quit)
Exception during process_requests()
Traceback (most recent call last):
  File "/home/yaz/.local/lib/python3.9/site-packages/picamera2/previews/null_pre                                                                                                                view.py", line 85, in handle_request
    picam2.process_requests(self)
  File "/home/yaz/.local/lib/python3.9/site-packages/picamera2/picamera2.py", li                                                                                                                ne 1104, in process_requests
    self.pre_callback(req)
  File "/home/yaz/apps/photobooth-imageserver/src/ImageServerPicam2.py", line 20                                                                                                                0, in _pre_callback_overlay
    if settings.debugging.picam2_stats_overlay:
AttributeError: 'ConfigSettings' object has no attribute 'debugging'
Exception in thread Thread-8:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/yaz/.local/lib/python3.9/site-packages/picamera2/previews/null_pre                                                                                                                view.py", line 29, in thread_func
    callback(picam2)
  File "/home/yaz/.local/lib/python3.9/site-packages/picamera2/previews/null_pre                                                                                                                view.py", line 85, in handle_request
    picam2.process_requests(self)
  File "/home/yaz/.local/lib/python3.9/site-packages/picamera2/picamera2.py", li                                                                                                                ne 1104, in process_requests
    self.pre_callback(req)
  File "/home/yaz/apps/photobooth-imageserver/src/ImageServerPicam2.py", line 20                                                                                                                0, in _pre_callback_overlay
    if settings.debugging.picam2_stats_overlay:
AttributeError: 'ConfigSettings' object has no attribute 'debugging'
2023-02-24 16:44:30,473 [INFO] __main__ subscribe() L138  Client connected Addre                                                                                                                ss(host='IP', port=57539)
2023-02-24 16:44:30,474 [DEBUG] __main__ add_subscriptions() L98   add subscript                                                                                                                ion for publishSSE
INFO:     IP:57539 - "GET /eventstream HTTP/1.1" 200 OK
2023-02-24 16:44:34,461 [WARNING] src.ImageServerPicam2AddonCustomAutofocus stat                                                                                                                sThread() L131  imageserver did not deliver frames, aborting cycle (should occur                                                                                                                 only if service stopped!)
2023-02-24 16:44:34,463 [DEBUG] src.ImageServerPicam2AddonCustomAutofocus statsT                                                                                                                hread() L184  autofocus run finished, sharpnessList=[]

`

mgrl, 2023-02-24:
updated the main branch.
please try 
`git pull`
to update and check console again.

yaazzz, 2023-02-24:
Hi, very strange. i get it working. Then I made a change in the front-end web page,,reload service and now I am back to the same situation even after reboot.

mgrl, 2023-02-24:
What is the log from console after stopping the service like above again?

mgrl, 2023-02-24:
Could be an illegal setting, there is no proper validation yet

yaazzz, 2023-02-24:
`
2023-02-24 21:29:55,928 [INFO] __main__ <module>() L323  Welcome to qPhotobooth
2023-02-24 21:29:55,929 [INFO] __main__ <module>() L324  platform.system=Linux
2023-02-24 21:29:55,930 [INFO] __main__ <module>() L325  platform.release=5.15.84-v8+
2023-02-24 21:29:55,931 [INFO] __main__ <module>() L326  platform.machine=aarch64
2023-02-24 21:29:55,932 [INFO] __main__ <module>() L327  platform.python_version=3.9.2
2023-02-24 21:29:55,932 [INFO] __main__ <module>() L328  hostname=raspberrypi
2023-02-24 21:29:55,933 [INFO] __main__ <module>() L329  psutil.cpu_count logical=4
2023-02-24 21:29:55,936 [INFO] __main__ <module>() L330  psutil.cpu_count cores=4
2023-02-24 21:29:55,938 [INFO] __main__ <module>() L331  psutil.disk_partitions=[sdiskpart(device='/dev/sda2', mountpoint='/', fstype='ext4', opts='rw,noatime', maxfile=255, maxpath=4096), sdiskpart(device='/dev/sda1', mountpoint='/boot', fstype='vfat', opts='rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro', maxfile=1530, maxpath=4096)]
2023-02-24 21:29:55,939 [INFO] __main__ <module>() L333  psutil.disk_usage /=sdiskusage(total=491968536576, used=4897472512, free=467021656064, percent=1.0)
2023-02-24 21:29:55,941 [INFO] __main__ <module>() L336  psutil.net_if_addrs={'lo': [snicaddr(family=<AddressFamily.AF_INET: 2>, address='127.0.0.1', netmask='255.0.0.0', broadcast=None, ptp=None), snicaddr(family=<AddressFamily.AF_INET6: 10>, address='::1', netmask='ffff:ffff:ffff:ffff:ffff:ffff:ffff:ffff', broadcast=None, ptp=None), snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='00:00:00:00:00:00', netmask=None, broadcast=None, ptp=None)], 'wlan0': [snicaddr(family=<AddressFamily.AF_INET: 2>, address='192.168.1.29', netmask='255.255.255.0', broadcast='192.168.1.255', ptp=None), snicaddr(family=<AddressFamily.AF_INET6: 10>, address='fdfc:9f65:b816:f837:fdf5:b06f:41ab:ae68', netmask='ffff:ffff:ffff:ffff::', broadcast=None, ptp=None), snicaddr(family=<AddressFamily.AF_INET6: 10>, address='fe80::d72d:3373:1af7:498b%wlan0', netmask='ffff:ffff:ffff:ffff::', broadcast=None, ptp=None), snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='dc:a6:32:54:6a:b9', netmask=None, broadcast='ff:ff:ff:ff:ff:ff', ptp=None)], 'eth0': [snicaddr(family=<AddressFamily.AF_PACKET: 17>, address='dc:a6:32:54:6a:b8', netmask=None, broadcast='ff:ff:ff:ff:ff:ff', ptp=None)]}
2023-02-24 21:29:55,942 [INFO] __main__ <module>() L337  psutil.virtual_memory=svmem(total=3977621504, available=3531653120, percent=11.2, used=331317248, free=2889871360, active=176803840, inactive=743796736, buffers=44142592, cached=712290304, shared=35577856, slab=89845760)
2023-02-24 21:29:55,943 [INFO] __main__ <module>() L339  __debug__=True
2023-02-24 21:29:55,944 [INFO] src.WledService __init__() L31   WledService disabled
[0:15:42.187588512] [1908]  INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4186-d1dc0373
[0:15:42.191468897] [1909]  WARN CameraSensorProperties camera_sensor_properties.cpp:205 No static properties available for 'arducam_64mp'
[0:15:42.191520526] [1909]  WARN CameraSensorProperties camera_sensor_properties.cpp:207 Please consider updating the camera sensor properties database
[0:15:42.214083296] [1909]  WARN RPI raspberrypi.cpp:1308 Mismatch between Unicam and CamHelper for embedded data usage!
[0:15:42.215041222] [1909]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/arducam_64mp@1a to Unicam device /dev/media2 and ISP device /dev/media3
2023-02-24 21:29:56,170 [DEBUG] src.ImageServerPicam2 _onPreviewMode() L163  change to preview mode
[0:15:42.238910263] [1908]  INFO Camera camera.cpp:1028 configuring streams: (0) 2312x1736-XBGR8888 (1) 1280x720-YUV420
[0:15:42.239636452] [1909]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/arducam_64mp@1a - Selected sensor format: 2312x1736-SBGGR10_1X10 - Selected unicam format: 2312x1736-pBAA
2023-02-24 21:29:56,196 [INFO] src.ImageServerPicam2 __init__() L71   camera_config: {'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 2, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (2312, 1736), 'stride': 9280, 'framesize': 16110080}, 'lores': {'format': 'YUV420', 'size': (1280, 720), 'stride': 1280, 'framesize': 1382400}, 'raw': None, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}, 'display': 'lores', 'encode': 'lores'}
2023-02-24 21:29:56,197 [INFO] src.ImageServerPicam2 __init__() L72   camera_controls: {'Sharpness': (0.0, 16.0, 1.0), 'AwbEnable': (False, True, None), 'Brightness': (-1.0, 1.0, 0.0), 'AfWindows': (libcamera.Rectangle(0, 0, 0, 0), libcamera.Rectangle(65535, 65535, 65535, 65535), libcamera.Rectangle(0, 0, 0, 0)), 'AeExposureMode': (0, 3, 0), 'AfSpeed': (0, 1, 0), 'AfTrigger': (0, 1, 0), 'AeConstraintMode': (0, 3, 0), 'AfMode': (0, 2, 0), 'ExposureTime': (131, 122583595, None), 'AfPause': (0, 2, 0), 'NoiseReductionMode': (0, 4, 0), 'AeEnable': (False, True, None), 'FrameDurationLimits': (33333, 122583654, None), 'ColourCorrectionMatrix': (-16.0, 16.0, None), 'ScalerCrop': (libcamera.Rectangle(0, 0, 256, 256), libcamera.Rectangle(0, 0, 9248, 6944), None), 'ExposureValue': (-8.0, 8.0, 0.0), 'AwbMode': (0, 7, 0), 'AfMetering': (0, 1, 0), 'AnalogueGain': (1.0, 10.079999923706055, None), 'ColourGains': (0.0, 32.0, None), 'Contrast': (0.0, 32.0, 1.0), 'AeMeteringMode': (0, 3, 0), 'Saturation': (0.0, 32. [... cut ...]
  File "/home/yaz/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 249, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.
2023-02-24 21:30:03,200 [WARNING] src.ImageServerPicam2AddonCustomAutofocus statsThread() L131  imageserver did not deliver frames, aborting cycle (should occur only if service stopped!)
2023-02-24 21:30:03,203 [DEBUG] src.ImageServerPicam2AddonCustomAutofocus statsThread() L184  autofocus run finished, sharpnessList=[]
`

mgrl, 2023-02-24:
Seems you set the picam backend as main and secondary. You can use picam only once as main backend. There is no benefit using it two times.

mgrl, 2023-02-24:
Remove LIVE_BACKEND key from config.json in config dir again.

yaazzz, 2023-02-24:
Done it! It solved the issue.

I can play to get everything running. Thank also for the 3d model: all printed and assembled ;)

mgrl, 2023-02-24:
Oh nice, would love if you post some images and / describe differences and highlights. You saw the wled integration for ring light integration?

yaazzz, 2023-02-24:
Yes I have implemented the wled. It is connected directly to the pi. I will probably need to buy an esp32. I will send you some pictures later ;)

mgrl, 2023-02-25:
https://kno.wled.ge/ esp32 or esp8266 would work.
Think this issue can be closed now :)

closed 2023-02-25 by mgrl