photobooth-app/photobooth-app #5: Auto focus not working on arducam_64mp

lyazide, 2023-02-25:
Hi,
Thanks again for all the support you are providing.
I think the last driver for arducam_64mp made a change for the autofocus.
Here is the log:
`
INFO:     Started server process [4131]
INFO:     Waiting for application startup.
INFO:     Application startup complete.
INFO:     Uvicorn running on http://0.0.0.0:8000 (Press CTRL+C to quit)
Exception in thread AutofocusStats:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/yaz/apps/photobooth-imageserver/src/ImageServerPicam2AddonCustomAutofocus.py", line 152, in statsThread
    roi_frame = getROIFrame(roi, frame)
  File "/home/yaz/apps/photobooth-imageserver/src/ImageServerPicam2AddonCustomAutofocus.py", line 102, in getROIFrame
    x_end = x_start + int(w * roi[2])
TypeError: int() argument must be a string, a bytes-like object or a number, not 'tuple'
2023-02-25 10:45:21,052 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
2023-02-25 10:45:26,052 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
2023-02-25 10:45:31,053 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
INFO:     192.168.1.23:34034 - "GET /stream.mjpg HTTP/1.1" 200 OK
2023-02-25 10:45:36,053 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
INFO:     192.168.1.23:45420 - "GET /config/schema?type=dereferenced HTTP/1.1" 200 OK
2023-02-25 10:45:41,055 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
2023-02-25 10:45:46,055 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
2023-02-25 10:45:51,055 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
2023-02-25 10:45:56,056 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
2023-02-25 10:46:01,056 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
2023-02-25 10:46:06,056 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
2023-02-25 10:46:11,057 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
2023-02-25 10:46:16,058 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
2023-02-25 10:46:21,058 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
2023-02-25 10:46:26,059 [WARNING] src.ImageServerPicam2AddonCustomAutofocus doFocus() L81   Focus is not done yet or in standby.
`
And the available option for focus from libacamera-still:
`
                                        %focus (focus FoM value)
  --autofocus-mode arg (=default)       Control to set the mode of the AF (autofocus) algorithm.(manual, auto, 
  --autofocus-range arg (=normal)       Set the range of focus distances that is scanned.(normal, macro, full)
  --autofocus-speed arg (=normal)       Control that determines whether the AF algorithm is to move the lens as quickly
  --autofocus-window arg (=0,0,0,0)     Sets AfMetering to  AfMeteringWindows an set region used, e.g. 
  --lens-position arg                   Set the lens to a particular focus position, expressed as a reciprocal distance
  --autofocus-on-capture [=arg(=1)] (=0)

`

mgrl, 2023-02-25:
Yes, I think in the latest releases arducam implemented native autofocus in libcamera.
Please just disable the "focuser", then picam2 defaults to internal autofocus algorithms implemented in libcamera2.
that should do the trick here. needs better description :)

lyazide, 2023-02-25:
Thanks for your reply. 
How do you make picam2 defaults to internal autofocus algorithms implemented in libcamera2?
Sorry I am a noob with those cameras?

mgrl, 2023-02-25:
in file ImageServerPicam2AddonLibcamAutofocus.py it's just setting the focus commands provided by libcamera.
So if these are supported you don't need the custom focuser any more.

yaazzz, 2023-02-25:
[image]

yaazzz, 2023-02-25:
[image]

lyazide, 2023-03-01:
Hi,
I made the update and install the WLEd with ESP. Now I am facing in hang when I try to take a picture. Here is the log. Sorry again for not being able to solve by myself.
`
INFO:     192.168.1.243:62184 - "GET /config/schema?type=dereferenced HTTP/1.1" 200 OK
2023-03-01 15:16:29,161 [INFO] src.camstatemachine run() L62   trigger event arm
INFO:     192.168.1.243:62187 - "GET /chose/1pic HTTP/1.1" 200 OK
2023-03-01 15:16:29,165 [DEBUG] src.camstatemachine on_exit_idle() L88   now locked
2023-03-01 15:16:29,169 [INFO] src.wledservice preset_countdown() L115  WledService preset_countdown triggered
2023-03-01 15:16:29,170 [DEBUG] src.wledservice _request_preset() L135  wled request preset id=2
2023-03-01 15:16:29,172 [DEBUG] src.imageserverpicam2 _on_capture_mode() L213  change to capture mode
2023-03-01 15:16:29,193 [INFO] src.imageserverpicam2 _generate_images_fun() L286  switch_mode invoked
[3:54:38.739238220] [4161]  INFO Camera camera.cpp:1028 configuring streams: (0) 1280x720-BGR888 (1) 1280x720-YUV420
[3:54:38.740220084] [4149]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/arducam_64mp@1a - Selected sensor format: 1280x720-SBGGR10_1X10 - Selected unicam format: 1280x720-pBAA
ERROR:    Exception in ASGI application
Traceback (most recent call last):
  File "/home/yaz/.local/lib/python3.9/site-packages/uvicorn/protocols/http/h11_impl.py", line 407, in run_asgi
    result = await app(  # type: ignore[func-returns-value]
  File "/home/yaz/.local/lib/python3.9/site-packages/uvicorn/middleware/proxy_headers.py", line 78, in __call__
    return await self.app(scope, receive, send)
  File "/home/yaz/.local/lib/python3.9/site-packages/fastapi/applications.py", line 270, in __call__
    await super().__call__(scope, receive, send)
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/applications.py", line 124, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/middleware/errors.py", line 184, in __call__
    raise exc
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/middleware/errors.py", line 162, in __call__
    await self.app(scope, receive, _send)
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/middleware/exceptions.py", line 79, in __call__
    raise exc
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/middleware/exceptions.py", line 68, in __call__
    await self.app(scope, receive, sender)
  File "/home/yaz/.local/lib/python3.9/site-packages/fastapi/middleware/asyncexitstack.py", line 21, in __call__
    raise e
  File "/home/yaz/.local/lib/python3.9/site-packages/fastapi/middleware/asyncexitstack.py", line 18, in __call__
    await self.app(scope, receive, send)
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/routing.py", line 706, in __call__
    await route.handle(scope, receive, send)
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/routing.py", line 276, in handle
    await self.app(scope, receive, send)
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/routing.py", line 69, in app
    await response(scope, receive, send)
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/responses.py", line 273, in __call__
    await wrap(partial(self.listen_for_disconnect, receive))
  File "/home/yaz/.local/lib/python3.9/site-packages/anyio/_backends/_asyncio.py", line 662, in __aexit__
    raise exceptions[0]
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/responses.py", line 269, in wrap
    await func()
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/responses.py", line 258, in stream_response
    async for chunk in self.body_iterator:
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/concurrency.py", line 63, in iterate_in_threadpool
    yield await anyio.to_thread.run_sync(_next, iterator)
  File "/home/yaz/.local/lib/python3.9/site-packages/anyio/to_thread.py", line 31, in run_sync
    return await get_asynclib().run_sync_in_worker_thread(
  File "/home/yaz/.local/lib/python3.9/site-packages/anyio/_backends/_asyncio.py", line 937, in run_sync_in_worker_thread
    return await future
  File "/home/yaz/.local/lib/python3.9/site-packages/anyio/_backends/_asyncio.py", line 867, in run
    result = context.run(func, *args)
  File "/home/yaz/.local/lib/python3.9/site-packages/starlette/concurrency.py", line 53, in _next
    return next(iterator)
  File "/home/yaz/apps/photobooth-imageserver/src/imageserverpicam2.py", line 171, in gen_stream
    array = self._wait_for_lores_frame()
  File "/home/yaz/apps/photobooth-imageserver/src/imageserverpicam2.py", line 209, in _wait_for_lores_frame
    raise IOError("timeout receiving frames")
OSError: timeout receiving frames
2023-03-01 15:16:32,482 [DEBUG] src.imagedb capture_hq_image() L236  capture to filename: 20230301_151632.jpg
2023-03-01 15:16:34,483 [ERROR] src.imagedb capture_hq_image() L273  timeout receiving frames
Traceback (most recent call last):
  File "/home/yaz/apps/photobooth-imageserver/src/imagedb.py", line 244, in capture_hq_image
    jpeg_buffer = self._imageserver.wait_for_hq_image()
  File "/home/yaz/apps/photobooth-imageserver/src/imageserverpicam2.py", line 161, in wait_for_hq_image
    raise IOError("timeout receiving frames")
OSError: timeout receiving frames
2023-03-01 15:16:34,494 [DEBUG] src.camstatemachine on_enter_idle() L82   now UNlocked
2023-03-01 15:16:34,496 [DEBUG] src.camstatemachine on_enter_idle() L83   waiting to start take pics!

`

mgrl, 2023-03-01:
sorry for this experience.

I find similar issue on my system after updating to latest apt-packages.
how much RAM is your pi?
when was your last time apt-get update was run?
when I start the browser on the system at the same time, it dies with the same error messages you provided (timeout receiving frames). additionally I see 
```
[253:16:56.316395881] [351901] ERROR V4L2 v4l2_videodevice.cpp:1248 /dev/video14[20:cap]: Not enough buffers provided by V4L2VideoDevice
Failed to allocate buffers.
```
in the logs. same true for you?
not sure, whether I can do something about this or it's a problem in the system.
when using a remote browser connecting to the photobooth it works just fine.

mgrl, 2023-03-01:
here is another report about similar issue: https://github.com/raspberrypi/picamera2/issues/581
I restarted the PI and the problem is solved for now... when was your last reboot?

loudjoudi, 2023-03-01:
Hi,

My rpi4 is à 4 gig.
I made several reboot and had the same issue.
I use it with a remote Web brosser : it both case I get such crash.
I have not performed an apr update recently.
The only thing is the update from you gît and the install of the wled : for both I don´t see any link with the crash.
As I played with the config is there a way to load à default booth config: I am questioning myself about a wrong parameter somewhere.
Thanks again for your time and support.

mgrl, 2023-03-01:
Config is located in config/config.json. When deleted the defaults are reverted.

mgrl, 2023-03-01:
When you take a picture the camera is switching to still picture mode. That needs more buffer if resolution is high. That is why it might fail when taking a picture

loudjoudi, 2023-03-01:
> When you take a picture the camera is switching to still picture mode. That needs more buffer if resolution is high. That is why it might fail when taking a picture 

Ok, that made the trick. Delete the config. Lower resolution seems to crash. I was expected the opposite ;)

Next step is to get the CP1500 printing. That one will not be for you. I confirm the esp is much more better that Direct rpi connection. Thanks again. This project is amazing.

mgrl, 2023-03-02:
I'll continue to watch this topic and see if any improvements possible on my codebasis.

Regarding the cp1500 are there driver available in the meantime? Let me know what commands you use for printing as I still need to implement my cp1300 also

loudjoudi, 2023-03-02:
I am not able to make any print from Linux. I always get a message saying the format is not correct. The drivers are the cups one, no gunterprint yet.
Do you have any script you use to prepare the picture?

mgrl, 2023-03-25:
I close this for now - feel free to open another issue if neccessary.

closed 2023-03-25 by mgrl

loudjoudi, 2023-03-25:
Sorry, I did not came back to you but was sure I did.
Everything is working now for me for several week. Last issue was the printing but seems to come from the printer : returned for repair.
This is amazing what you did ;)