photobooth-app/photobooth-app #38: Random focus in images

mookie-, 2023-05-13:
Hello,

thank you very much for this project! I currently try to get a working Photobooth with an arducam 64mp.
Unfortunately I have an issue and I'm not sure if it's normal.

When the live preview is running with `LibcamAfInterval` as focuser, it will refocus any 10 seconds.
If I now take a picture and it gets executed while the camera runs the autofocus, the picture is pretty bad.
Is it possible to make sure the autofocus process and picture taking don't overlap?

mgrl, 2023-05-14:
Right every 10 seconds, you can change that time in settings, but that’s not helpful here I guess.
When switching to capture mode on countdown the resolution changes to capture resolution. If you chose highest res, the camera only delivers like 2 frames/sec, which makes autofocus take pretty long and image would be blurry.
I check whether possible to abort focus run before capture if in progress during shoot.
Another option would be longer countdown (no new focus runs are startet in that mode) .

mgrl, 2023-05-14:
I updated the source to cancel an ongoing focus run if active right before capture. This should fix it in first place.
Might refine later and handle smarter based on the actual AfState.

mgrl, 2023-05-21:
Did the update make it work for you?

mookie-, 2023-05-25:
Hello @mgrl , I'm sorry for the late response and thank you very much for the fast help!
Unfortunately I got errors when using the updated code. Unfortunately I didn't had the time to debug it and I have to wait another 2 weeks until I can test it again (and provide you the error output).

I think one problem was that I did use https://github.com/PhotoboothProject/photobooth as frontend. Due to that the countdown did not run in the photobooth app.
I will dig into this again and will let you know what I find.

mgrl, 2023-05-26:
Using the photobbothproject as frontend is totally fine. It would be good to know what commands you set in the frontend to interact with the backend. The api changed a little in latest code, maybe it’s just that.

mgrl, 2023-05-26:
https://github.com/mgrl/photobooth-app#integrate-photobooth-project-and-this-photobooth-app

mgrl, 2023-05-26:
Error messages would be very welcome 🙂

mgrl, 2023-08-26:
Closing this for now. Feel free to reopen if you have still issues.

PR #41 (2023-05-14): Ensure non-focusing state on capture for picamera2
could fix #38
closed 2023-08-26 by mgrl