photobooth-app/photobooth-app #44: removebg?

mgrl, 2023-05-21:


mgrl, 2023-10-06:
will not implement background removal for now.

closed 2023-10-06 by mgrl

mgineer85, 2025-09-28:
Only reason it is not yet there is because I would assume it's too slow on a Pi? Not a big fan of greenscreens also.

- Which hardware did you test?
- How did the installation work for you? 
- Any additional dependencies needed?

Depending on how hard it is to install, we could make it an optional dependency and if it is available, it could be used...

mgineer85, 2025-09-30:
Input Image is 1442x1080:

Result from actions runner:
```
2025-09-30T10:08:33.2840279Z --------------------------------------------------------------------------- benchmark 'background_removal_ai': 3 tests ---------------------------------------------------------------------------
2025-09-30T10:08:33.2840956Z Name (time in ms)                    Min                 Max                Mean            StdDev              Median               IQR            Outliers     OPS            Rounds  Iterations
2025-09-30T10:08:33.2841551Z --------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
2025-09-30T10:08:33.2842358Z test_rembg[u2netp]              432.3502 (1.0)      443.8766 (1.0)      435.7654 (1.0)      4.6099 (1.0)      434.1830 (1.0)      3.3525 (1.0)           1;1  2.2948 (1.0)           5           1
2025-09-30T10:08:33.2843135Z test_rembg[u2net_human_seg]     826.3637 (1.91)     843.2014 (1.90)     831.3208 (1.91)     6.8714 (1.49)     828.6580 (1.91)     7.0116 (2.09)          1;0  1.2029 (0.52)          5           1
2025-09-30T10:08:33.2843787Z test_rembg[u2net]               828.3656 (1.92)     846.2815 (1.91)     832.3009 (1.91)     7.8215 (1.70)     828.8729 (1.91)     4.8350 (1.44)          1;1  1.2015 (0.52)          5           1
2025-09-30T10:08:33.2844405Z --------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
```

Result from Pi4 runner:

```
2025-09-30T10:24:36.0198460Z ----------------------------------------------------------------------- benchmark 'background_removal_ai': 3 tests -----------------------------------------------------------------------
2025-09-30T10:24:36.0201647Z Name (time in s)                   Min               Max              Mean            StdDev            Median               IQR            Outliers     OPS            Rounds  Iterations
2025-09-30T10:24:36.0205070Z ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
2025-09-30T10:24:36.0208070Z test_rembg[u2netp]              2.7248 (1.0)      2.8296 (1.0)      2.7466 (1.0)      0.0464 (1.0)      2.7264 (1.0)      0.0270 (1.0)           1;1  0.3641 (1.0)           5           1
2025-09-30T10:24:36.0211011Z test_rembg[u2net]               5.6881 (2.09)     5.9671 (2.11)     5.7730 (2.10)     0.1223 (2.63)     5.6947 (2.09)     0.1664 (6.16)          1;0  0.1732 (0.48)          5           1
2025-09-30T10:24:36.0213936Z test_rembg[u2net_human_seg]     5.7845 (2.12)     6.0402 (2.13)     5.8902 (2.14)     0.1068 (2.30)     5.8983 (2.16)     0.1728 (6.40)          1;0  0.1698 (0.47)          5           1
2025-09-30T10:24:36.0216866Z ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
2025-09-30T10:24:36.0218547Z 
2025-09-30T10:24:36.0220381Z ------------------------------------------------------------------------------- benchmark 'background_removal_chromakey': 2 tests --------------------------------------------------------------------------------
2025-09-30T10:24:36.0223734Z Name (time in ms)                                      Min                Max               Mean            StdDev             Median               IQR            Outliers      OPS            Rounds  Iterations
2025-09-30T10:24:36.0226650Z ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
2025-09-30T10:24:36.0230145Z test_opencv_greenscreen[opencv_chromakey_live]     62.3608 (1.0)      63.4870 (1.0)      62.9082 (1.0)      0.3338 (1.0)      62.9663 (1.0)      0.4417 (1.0)           5;0  15.8962 (1.0)          16           1
2025-09-30T10:24:36.0233723Z test_opencv_greenscreen[opencv_chromakey]          79.7394 (1.28)     83.0095 (1.31)     81.5369 (1.30)     0.9854 (2.95)     81.7083 (1.30)     1.2655 (2.87)          3;0  12.2644 (0.77)         10           1
2025-09-30T10:24:36.0237033Z ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
```

mgineer85, 2025-10-05:
Just adding some more numbers: 

```
experimental/rembgAndwithoutbg/main.py

Hello from rembg!
13fe244a730f483f9b5ff20767c011dd.jpg: duration: 0.912s
1a4d424374144731826d75d086e03260.jpg: duration: 0.863s
26a47ca532ab433d9e6902827826c57c.jpg: duration: 0.643s
2d1351cff3f7459bac2716a51f8140ec.jpg: duration: 0.79s
9faccb7fad044ea7bc544575bb3e8800.jpg: duration: 0.839s

Hello from rembg-noalphamatting!
13fe244a730f483f9b5ff20767c011dd.jpg: duration: 0.679s
1a4d424374144731826d75d086e03260.jpg: duration: 0.728s
26a47ca532ab433d9e6902827826c57c.jpg: duration: 0.584s
2d1351cff3f7459bac2716a51f8140ec.jpg: duration: 0.783s
9faccb7fad044ea7bc544575bb3e8800.jpg: duration: 0.663s

Hello from withoutbg!
13fe244a730f483f9b5ff20767c011dd.jpg: duration: 2.63s
1a4d424374144731826d75d086e03260.jpg: duration: 2.39s
26a47ca532ab433d9e6902827826c57c.jpg: duration: 1.95s
2d1351cff3f7459bac2716a51f8140ec.jpg: duration: 2.76s
9faccb7fad044ea7bc544575bb3e8800.jpg: duration: 2.39s

Hello from eugene!
2d1351cff3f7459bac2716a51f8140ec.jpg: duration: 8.85s
9faccb7fad044ea7bc544575bb3e8800.jpg: duration: 9.36s
13fe244a730f483f9b5ff20767c011dd.jpg: duration: 9.15s
1a4d424374144731826d75d086e03260.jpg: duration: 9.21s
26a47ca532ab433d9e6902827826c57c.jpg: duration: 1.38s
```

Some more references:
- outdated code, not directly usable: https://medium.com/analytics-vidhya/how-to-change-image-background-using-tensorflow-in-python-62b1b2d570b9
- withoutbg seems to be better at alphamatting, but is way slower and segmentation is bit worse compared to rembg: https://github.com/withoutbg/withoutbg
- "eugene" implementation, is slow and deeplabv3 is not optimized for background removal, which shows during testing. https://colab.research.google.com/github/eugenesiow/practical-ml/blob/master/notebooks/Remove_Image_Background_DeepLabV3.ipynb

PixelAlarm, 2025-10-05:
This is awesome. Thank you so much!

I already pulled the changes. How do I get the extra package? I have rembg installed as Package and also used "uv tool install rembg". Just ignore this, if this will come to the readme :)

Another question:
Wouldn't it be good to move the background removal to the action itself. This way it could be configured per action and the possibility is given to the guest, but a normal picture is also possible via another action.

mgineer85, 2025-10-05:
when using uv in my dev I installed rembg  extra package using `uv sync --all-extras`

when you install with pipx I think it should be `pipx install photobooth-app[rembg]`

I also thought about having it in the actions, but thought it would be a setup/placement decision, not an action-related config.

I will release a beta in a few minutes to test it with global config.   Maybe it can be added later in the final release when it's working ok.