photobooth-app/photobooth-app #76: Incompatible with opencv-python

mart-e, 2023-08-08:
Hello,

I tried to install photobooth-app on my raspberry (raspbian bullseye, python 3.9.2) yesterday and I got the following error:
```
$ pip install photobooth-app
...
Collecting photobooth-app
  Downloading https://www.piwheels.org/simple/photobooth-app/photobooth_app-0.1.0b4-py3-none-any.whl (8.4 MB)
     |████████████████████████████████| 8.4 MB 8.6 MB/s 
  Downloading https://www.piwheels.org/simple/photobooth-app/photobooth_app-0.1.0b3-py3-none-any.whl (8.4 MB)
     |████████████████████████████████| 8.4 MB 8.5 MB/s 
  Downloading https://www.piwheels.org/simple/photobooth-app/photobooth_app-0.1.0b2-py3-none-any.whl (8.4 MB)
     |████████████████████████████████| 8.4 MB 3.2 MB/s 
  Downloading https://www.piwheels.org/simple/photobooth-app/photobooth_app-0.1.0b1-py3-none-any.whl (8.4 MB)
     |████████████████████████████████| 8.4 MB 10.2 MB/s 
ERROR: Cannot install photobooth-app==0.1.0b1, photobooth-app==0.1.0b2, photobooth-app==0.1.0b3, photobooth-app==0.1.0b4, photobooth-app==0.1.0b5, photobooth-app==0.1.0rc12, photobooth-app==0.1.0rc3, photobooth-app==0.1.0rc4, photobooth-app==0.1.0rc5, photobooth-app==0.1.0rc6, photobooth-app==0.1.0rc7, photobooth-app==0.1.0rc8 and photobooth-app==0.1.0rc9 because these package versions have conflicting dependencies.

The conflict is caused by:
    photobooth-app 0.1.0rc12 depends on opencv-python<4.9 and >=4.7
    photobooth-app 0.1.0rc9 depends on opencv-python<4.9 and >=4.7
    photobooth-app 0.1.0rc8 depends on opencv-python<4.9 and >=4.7
    photobooth-app 0.1.0rc7 depends on opencv-python<4.9 and >=4.7
    photobooth-app 0.1.0rc6 depends on opencv-python~=4.7.0
    photobooth-app 0.1.0rc5 depends on opencv-python~=4.7.0
    photobooth-app 0.1.0rc4 depends on opencv-python~=4.7.0
    photobooth-app 0.1.0rc3 depends on opencv-python~=4.7.0
    photobooth-app 0.1.0b5 depends on opencv-python<5.0.0.0 and >=4.7.0.72
    photobooth-app 0.1.0b4 depends on opencv-python<5.0.0.0 and >=4.7.0.72
    photobooth-app 0.1.0b3 depends on opencv-python<5.0.0.0 and >=4.7.0.72
    photobooth-app 0.1.0b2 depends on opencv-python<5.0.0.0 and >=4.7.0.72
    photobooth-app 0.1.0b1 depends on opencv-python<5.0.0.0 and >=4.7.0.72

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/user_guide/#fixing-conflicting-dependencies
```
[Full error logs](https://github.com/mgrl/photobooth-app/files/12287465/out.log)

Thank you

mgrl, 2023-08-08:
Hi, what system you use? Apt is up to date?
See the full log, there is a problem in pip cmake building opencv . That's why it complains.

mgrl, 2023-08-11:
I find this is an issue on the 32bit version of raspberry pi os. 
I highly recommend to change to the 64Bit version, on that version installation works fine.
Otherwise if you get it to run on 32bit machine I would be glad if you could provide the procedure to solve.

mart-e, 2023-08-11:
Hmm I will try again with a fresh 64bits install this weekend, thanks

mgrl, 2023-08-13:
Confirmed that 64bit works and has no drawbacks so far. 
I close this issue. Please reopen if needed. Have fun ✨

closed 2023-08-13 by mgrl

mart-e, 2023-08-13:
Hi,
Yes I just tried again and got no issue with 64 bits rapsbian. Not sure what changed since when I tested in spring but I have my solution :)