photobooth-app/photobooth-app #83: wled stopped working

dtimes, 2023-08-28:
Hi, 

wled was working fine , but stopped today after I did a "pip install --upgrade photobooth-app" as mentioned in the update docs , 
I´ve also tried swichting between and dev branch, both  are not triggering the LED in my setup or generating any output relating to wledservice.py (which was the case before , see log below). 

could you please give me a hint how to troubleshoot this ? log level is already set to debug. permissions to /dev/ttyUSB0 seems to be fine and I think dmesg output looks also good 

`[ 5826.390447] usb 1-1.2: new full-speed USB device number 4 using xhci_hcd
[ 5826.524151] usb 1-1.2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 5826.524186] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5826.524203] usb 1-1.2: Product: CP2102 USB to UART Bridge Controller
[ 5826.524218] usb 1-1.2: Manufacturer: Silicon Labs
[ 5826.524231] usb 1-1.2: SerialNumber: 0001
[ 5826.536923] cp210x 1-1.2:1.0: cp210x converter detected
[ 5826.540573] usb 1-1.2: cp210x converter now attached to ttyUSB0
`

thanks a lot for your great software and support ! 

->journalctl --user --unit photobooth-app.service|grep wled
`Jul 30 11:06:38 photobooth python[800]: 2023-07-30 11:06:38,463 [    INFO] WledService preset_shoot triggered (wledservice.py:130)
Jul 30 11:06:38 photobooth python[800]: 2023-07-30 11:06:38,768 [    INFO] WledService preset_standby triggered (wledservice.py:120)
Jul 30 11:06:43 photobooth python[800]: 2023-07-30 11:06:43,956 [    INFO] WledService preset_thrill triggered (wledservice.py:125)
Jul 30 11:06:46 photobooth python[800]: 2023-07-30 11:06:46,952 [    INFO] WledService preset_shoot triggered (wledservice.py:130)
Jul 30 11:06:47 photobooth python[800]: 2023-07-30 11:06:47,256 [    INFO] WledService preset_standby triggered (wledservice.py:120)
Jul 30 11:06:58 photobooth python[800]: 2023-07-30 11:06:58,829 [    INFO] WledService preset_standby triggered (wledservice.py:120)
Jul 30 11:13:45 photobooth python[800]: 2023-07-30 11:13:45,637 [    INFO] close port to WLED module (wledservice.py:113)
Jul 30 11:14:14 photobooth python[794]: 2023-07-30 11:14:14,742 [    INFO] WledService enabled, trying to setup and connect, self._serial_port='/dev/ttyUSB0' (wledservice.py:36)
Jul 30 11:14:14 photobooth python[794]: 2023-07-30 11:14:14,758 [    INFO] WLED version response: b'WLED 2306210\r\n' (wledservice.py:80)
Jul 30 11:14:14 photobooth python[794]: 2023-07-30 11:14:14,760 [    INFO] register events for WLED (wledservice.py:96)
Jul 30 11:14:14 photobooth python[794]: 2023-07-30 11:14:14,761 [    INFO] WledService preset_standby triggered (wledservice.py:120)
Jul 30 11:19:34 photobooth python[794]: 2023-07-30 11:19:34,086 [    INFO] WledService preset_thrill triggered (wledservice.py:125)
Jul 30 11:19:37 photobooth python[794]: 2023-07-30 11:19:37,169 [    INFO] WledService preset_shoot triggered (wledservice.py:130)
Jul 30 11:19:37 photobooth python[794]: 2023-07-30 11:19:37,575 [    INFO] WledService preset_standby triggered (wledservice.py:120)
Aug 01 13:36:05 photobooth python[888]: 2023-08-01 13:36:05,737 [    INFO] WledService enabled, trying to setup and connect, self._serial_port='/dev/ttyUSB0' (wledservice.py:36)
Aug 01 13:36:05 photobooth python[888]: 2023-08-01 13:36:05,756 [    INFO] WLED version response: b'WLED 2306210\r\n' (wledservice.py:80)
Aug 01 13:36:05 photobooth python[888]: 2023-08-01 13:36:05,757 [    INFO] register events for WLED (wledservice.py:96)
Aug 01 13:36:05 photobooth python[888]: 2023-08-01 13:36:05,759 [    INFO] WledService preset_standby triggered (wledservice.py:120)
Aug 01 13:37:27 photobooth python[888]: 2023-08-01 13:37:27,539 [    INFO] close port to WLED module (wledservice.py:113)
Aug 01 13:37:52 photobooth python[796]: 2023-08-01 13:37:52,321 [    INFO] WledService enabled, trying to setup and connect, self._serial_port='/dev/ttyUSB0' (wledservice.py:36)
Aug 01 13:37:52 photobooth python[796]: 2023-08-01 13:37:52,336 [    INFO] WLED version response: b'WLED 2306210\r\n' (wledservice.py:80)
Aug 01 13:37:52 photobooth python[796]: 2023-08-01 13:37:52,337 [    INFO] register events for WLED (wledservice.py:96)
Aug 01 13:37:52 photobooth python[796]: 2023-08-01 13:37:52,338 [    INFO] WledService preset_standby triggered (wledservice.py:120)
Aug 07 21:44:39 photobooth python[796]: 2023-08-07 21:44:39,639 [    INFO] close port to WLED module (wledservice.py:113)
Aug 07 21:45:06 photobooth python[797]: 2023-08-07 21:45:06,979 [    INFO] WledService enabled, trying to setup and connect, self._serial_port='/dev/ttyUSB0' (wledservice.py:36)
Aug 07 21:45:06 photobooth python[797]: 2023-08-07 21:45:06,995 [    INFO] WLED version response: b'WLED 2306210\r\n' (wledservice.py:80)
Aug 07 21:45:06 photobooth python[797]: 2023-08-07 21:45:06,997 [    INFO] register events for WLED (wledservice.py:96)
`

mgrl, 2023-08-29:
Sorry, initialization of some services did not work well any more.
It's fixed now and PyPi packages are updated in a few minutes.

mgrl, 2023-08-29:
Ah, since last update some config settings for backends moved it's location - you might need to reconfigure your backend resolution and so on again after the update yesterday.

closed 2023-08-29 by mgrl

dtimes, 2023-08-29:
thanks a lot for the fast reply , I will update & try gain

commit 4204dd48 (2023-08-29): fix service initialization fixes #83