photobooth-app/photobooth-app #86: Customize UI

mgrl, 2023-09-09:
- [ ] custom colors https://quasar.dev/style/color-palette#dynamic-change-of-brand-colors-dynamic-theme-colors-
- [ ] custom fonts https://quasar.dev/style/typography

DominicWindisch, 2024-02-08:
Regarding user css, do you plan on allowing a user provided css stylesheet or to add css to specific components? 

In the dev branch of my fork, I've added user css to some components ([action buttons](https://github.com/DominicWindisch/photobooth-frontend/blob/54a1a9675d3756f466554a48eebd550b0e331004/src/pages/IndexPage.vue#L41), gallery button, toolbar), but that quickly gets out of hand with an input for each tiny component... On the other hand, to make a user stylesheet more accessible, we should at least add IDs to some of the core components.

mgineer85, 2024-02-08:
Jeah Right.
I would love to allow the user upload a file to a userdata folder. The folder shall not be deleted during updates. Also a custom i18n file might be possible, if some very specific language shall be used.
Not sure yet what the simplest way is to load additional files in ui if avail from server…

closed 2024-02-10 by mgineer85