photobooth-app/photobooth-app #87: install fail

Frontkiller, 2023-10-02:
mennes@raspberrypi:~ $ pip install photobooth-app
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting photobooth-app
  Using cached https://www.piwheels.org/simple/photobooth-app/photobooth_app-0.9.1-py3-none-any.whl (8.9 MB)
Collecting dependency-injector~=4.41.0 (from photobooth-app)
  Using cached https://www.piwheels.org/simple/dependency-injector/dependency_injector-4.41.0-cp39-cp39-linux_armv7l.whl (4.0 MB)
Collecting fastapi<0.104.0,>=0.100.0 (from photobooth-app)
  Using cached https://www.piwheels.org/simple/fastapi/fastapi-0.103.2-py3-none-any.whl (66 kB)
Collecting gphoto2~=2.5.0 (from photobooth-app)
  Using cached https://www.piwheels.org/simple/gphoto2/gphoto2-2.5.0-cp39-cp39-linux_armv7l.whl (1.2 MB)
Requirement already satisfied: gpiozero~=1.6.2 in /usr/lib/python3/dist-packages (from photobooth-app) (1.6.2)
Collecting jsonref~=1.1.0 (from photobooth-app)
  Using cached https://www.piwheels.org/simple/jsonref/jsonref-1.1.0-py3-none-any.whl (9.4 kB)
Collecting numpy>=1.25.0 (from photobooth-app)
  Using cached numpy-1.26.0.tar.gz (15.6 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [47 lines of output]
      + /usr/bin/python3 /tmp/pip-install-2jve2p4c/numpy_cf8120598a53465cad9b74c0b76372ba/vendored-meson/meson/meson.py setup /tmp/pip-install-2jve2p4c/numpy_cf8120598a53465cad9b74c0b76372ba /tmp/pip-install-2jve2p4c/numpy_cf8120598a53465cad9b74c0b76372ba/.mesonpy-kv1bebar/build -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --native-file=/tmp/pip-install-2jve2p4c/numpy_cf8120598a53465cad9b74c0b76372ba/.mesonpy-kv1bebar/build/meson-python-native-file.ini
      The Meson build system
      Version: 1.2.99
      Source dir: /tmp/pip-install-2jve2p4c/numpy_cf8120598a53465cad9b74c0b76372ba
      Build dir: /tmp/pip-install-2jve2p4c/numpy_cf8120598a53465cad9b74c0b76372ba/.mesonpy-kv1bebar/build
      Build type: native build
      Project name: NumPy
      Project version: 1.26.0
      C compiler for the host machine: cc (gcc 10.2.1 "cc (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
      C linker for the host machine: cc ld.bfd 2.35.2
      C++ compiler for the host machine: c++ (gcc 10.2.1 "c++ (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
      C++ linker for the host machine: c++ ld.bfd 2.35.2
      Cython compiler for the host machine: cython (cython 3.0.2)
      Host machine cpu family: arm
      Host machine cpu: arm
      Program python found: YES (/usr/bin/python3)
      Found pkg-config: /usr/bin/pkg-config (0.29.2)
      Run-time dependency python found: YES 3.9
      Has header "Python.h" with dependency python-3.9: YES
      Compiler for C supports arguments -fno-strict-aliasing: YES
      Message: During parsing cpu-dispatch: The following CPU features were ignored due to platform incompatibility or lack of support:
      "XOP FMA4"
      Test features "NEON" : Supported
      Test features "NEON_FP16" : Supported
      Test features "NEON_VFPV4" : Supported
      Test features "ASIMD" : Unsupported due to Compiler fails against the test code of "ASIMD"
      Test features "ASIMDHP" : Unsupported due to Implied feature "ASIMD" is not supported
      Test features "ASIMDFHM" : Unsupported due to Implied feature "ASIMD" is not supported
      Configuring npy_cpu_dispatch_config.h using configuration
      Message:
      CPU Optimization Options
        baseline:
          Requested : min
          Enabled   :
        dispatch:
          Requested : max -xop -fma4
          Enabled   : NEON NEON_FP16 NEON_VFPV4
      
      Library m found: YES
      Did not find CMake 'cmake'
      Found CMake: NO
      Run-time dependency openblas found: NO (tried pkgconfig and cmake)
      Run-time dependency openblas found: NO (tried pkgconfig and cmake)
      
      ../../numpy/meson.build:207:4: ERROR: Problem encountered: No BLAS library detected! Install one, or use the `allow-noblas` build option (note, this may be up to 100x slower for some linear algebra operations).
      
      A full log can be found at /tmp/pip-install-2jve2p4c/numpy_cf8120598a53465cad9b74c0b76372ba/.mesonpy-kv1bebar/build/meson-logs/meson-log.txt
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

mgrl, 2023-10-02:
Cmake not found… what system you are on? Debian and Ubuntu should be fine

Frontkiller, 2023-10-02:
> Cmake not found… what system you are on? Debian and Ubuntu should be fine

Debian Bullseye with Raspberry Pi Desktop

mgrl, 2023-10-02:
There might be an issue in numpy: https://github.com/numpy/numpy/issues/24703

can have a look into this tomorrow

Frontkiller, 2023-10-02:
> There might be an issue in numpy: https://github.com/numpy/numpy/issues/24703
> 
> 
> 
> can have a look into this tomorrow 

Ok, thanks in advance for your help.
I really appreciate it!

mgrl, 2023-10-02:
Are you on 32bit? I highly recommend 64bit

Frontkiller, 2023-10-03:
> Are you on 32bit? I highly recommend 64bit

Yes 32bit, i will do a clean install with bullseye 64bit and try again

mgrl, 2023-10-03:
I reran github actions with most current numpy on 64bit systems. That works for me. Is the issue solved with 64bit system on your side also?

Frontkiller, 2023-10-03:
Yes it works now on the 64bit, i got some lag (1-2 seconds) and the resolution is not that good. Can I solve this ?

mgrl, 2023-10-03:
You can set the resolution in the admin backend. Images are streamed directly when available to the browser, there is no buffer. How do you display the stream? Any network involved?

Frontkiller, 2023-10-03:
Thanks for the help but now after installing the service I got another problem. My cam doesn't work anymore. When I do Libcamera-hello it gives ERROR V4L2 : unable to set controls : device or resource busy.

Of course photobooth also doesn't startup now.

Can hou help me out please ? 

I really appreciate your help!

mgrl, 2023-10-03:
Maybe you have another instance running on a terminal? Did you restart once? Could be some process that stalled for whatever reason.

mgrl, 2023-10-03:
Ah wait. When you installed the service, the app is permanently accessing the camera. If you want to use the libcamera apps you need to stop the service. The cam is open for the livestream

Frontkiller, 2023-10-03:
Yeah I did a reboot and a shutdown without succes. 

Damn it was working so well

Frontkiller, 2023-10-03:
How do I stop the service ? I can't run photobooth in terminal 

Thanks

Frontkiller, 2023-10-03:
Ok I see, just go to the Localhost 8000

Thank you for helping !!!

mgrl, 2023-10-03:
Ok, so you got it working?
Here you might find additional help: https://mgrl.github.io/photobooth-docs/support/troubleshooting/

mgrl, 2023-10-03:
Also it would be great if you have feedback on the documentation where it could be improved to support you better.

Frontkiller, 2023-10-04:
I'm sorry to bother you again but now i'm having a new problem. 

The locolhost8000 page stopped but i can't startup photobooth in terminal again because it says there is still a process running.

Is there a way to stop the process and start it again in Terminal ?

Thank a lot !

mgrl, 2023-10-04:
Might be misconfiguration.
Try journalctl —user stop photobooth and start from terminal. Ensure you start from within the application working directory.

Frontkiller, 2023-10-05:
> Might be misconfiguration.
> 
> Try journalctl —user stop photobooth and start from terminal. Ensure you start from within the application working directory.

Zhen i try the journalctl I get following red message 😀 

Failed to add match "stop" : invalid argument

mgrl, 2023-10-05:
Sorry autocorrect. try again.

mgrl, 2023-10-05:
Clarified the commands to troubleshoot here a bit:
https://mgrl.github.io/photobooth-docs/support/troubleshooting/#gather-information-to-troubleshoot

mgrl, 2023-10-15:
Closing this, as latest release is tested with RPI Bookworm now and works fine.
Feel free to reopen if any issues remain.

closed 2023-10-15 by mgrl