photobooth-app/photobooth-app #107: New Ui Settings

gitfoehn, 2023-10-26:
Hi @mgrl 
thanks for this amazing app. 

I was wondering if you were planing on the following settings in the admin panel. 

Enable/ disable touch buttons on main page. I was planing on using hardware buttons. So it would be nice not to show any touchscreen options.

For the "gallery" I was planing on using a smaller display with touch. I don't want the user to be able to navigate from the "gallery" to the "main page", bypassing my hardware buttons. 

Therefore I would like the option to enable/disable the "< start" button, which Leads back to the "main page".

Also I think it might be need to be able to go back to "main page" from the "capture page" via gpio button. 
I know I can trigger a new picture from this page as well but the preview and the animation are missing which I really like. 

Not really important but I think it would be nice, if the smily shown before taking the picture, is shown after the countdown instead of while. But that's just my personal preference and I understand if you would like to keep it that way.

Anyway, thanks again for your work! Cheers :)

mgrl, 2023-10-26:
I find following suggestions from your post
- buttons on main screen -> already implemented in latest release
- Improve process using gpio only -> I think about it. It’s very touch optimized, I barely use gpio and don’t want to add too many buttons. Usually capture page closes after a configurable amount of time. Why doesn’t this work for your case?
- Kind of standalone gallery-> could be simple, I check how to hide the button.
- Timing smile icon. -> I prefer to stick with current implementation. At 0 in best case the camera captures. Display the smile symbol would be too late then.

gitfoehn, 2023-10-26:
Sorry for not checking the latest version. Nice, that this is already implemented. 

A stand-alone gallery would be awesome. 

I agree that the other two issues are more of a personal preference and I totally understand, if you would rather stick to the current implementation.

mgrl, 2023-10-27:
Hi, 
please see above post about current implementation.

For standalone gallery I openend #108 because needs bit more refactoring later.
Other improvements are addressed or I intend not to change.

Please update to the latest version (will be pushed very soon) and let me know if it's working as intended for you.

commit f029b8c5 (2023-10-27): update frontend, fix #107
closed 2023-10-27 by mgrl