photobooth-app/photobooth-app #123: Slideshow

mgrl, 2023-11-04:
### Discussed in https://github.com/photobooth-app/photobooth-app/discussions/122

<div type='discussions-op-text'>

<sup>Originally posted by **laguinch** November  4, 2023</sup>
Hy, is it possible to make a slideshow from the galley ? Thx</div>

DominicWindisch, 2024-02-08:
Do you already have an approach in mind?

From the top of my head, I'd probably add a new page for that (or is that what standalone is meant for?) displaying the galleryimagedetail component.

Further brainstorming:
- Auto start slideshow after timeout on main page
- Auto start slideshow after timeout on gallery page
- button to start slideshow in gallery page
- when clicking the slideshow, return to main page (maybe prompt "Stop the slideshow?")
- configurable slideshow mode (random, in-order)
- configurable timeouts (timeout to start, time per image)
- probably fade transition between images 
- hide buttons (i.e. no toolbar in slideshow mode)
- no qr code in slideshow mode
- handle carousel button presses and swipes not as "cancel the slide show" intent
- probably exclude videos and gifs from the slideshow?
- config stop of the slideshow (after time X, when all images were shown, repeat forever)

Let me know what you think about it :) unfortunately I won't have time to work on it anyways in the upcoming days.

mgineer85, 2024-02-08:
There is a very simple standalone layout already implemented, which could be extended.

- https://github.com/photobooth-app/photobooth-frontend/blob/main/src/layouts/StandaloneLayout.vue
- https://photobooth-app.org/reference/foldersandurls/#special-urls
- quasar also has some components that could be reused or at least give inspiration:
- https://quasar.dev/vue-components/carousel#controls
- https://quasar.dev/vue-components/carousel#infinite-and-autoplay
- 

I would implement all these additions in the separate pages and only after a timeout forward the router there.
The separate pages are also great if multiple devices are used. You can point additional devices to a specific standalone Url and it would play random slides permanently. … or it could display a gallery where user can choose for printing.

DominicWindisch, 2024-02-13:
See https://github.com/photobooth-app/photobooth-frontend/pull/9.

DominicWindisch, 2024-02-17:
Can be closed, I guess ✌️

closed 2024-03-26 by mgineer85