photobooth-app/photobooth-app #131: Issue with Image Flipping in Photobooth-App libphoto2 Backend

Peda1996, 2023-12-12:
The image flipping feature is not functioning correctly in the Photobooth-App when using the Canon EOS 2000D camera. The application utilizes the libphoto2 backend. When the flipping feature is activated, either no change occurs, or the feature does not respond at all.

### Steps to Reproduce

  Set up the Photobooth-App with a Canon EOS 2000D camera.
    Ensure that the app is configured to use the libphoto2 backend.
    Attempt to use the image flipping feature within the app.

### Expected Behavior
The image should flip horizontally, allowing for a mirrored effect as typically seen in photobooth applications.

### Actual Behavior
The flipping feature does not alter the image. No error messages are displayed, and the app continues to function normally aside from this issue.

### Proposed Solutions

    Adjustment in gphoto2 Backend: Investigate the integration with the libphoto2 backend. It might be possible that the flipping functionality is not properly implemented or there is a compatibility issue with the EOS 2000D model.
    CSS-Based Flipping: As a workaround or alternative solution, implement the flipping feature using CSS. This approach would flip the image on the frontend. After the image capture, process the image to apply a permanent flip before saving or further processing. This method ensures the flipped image is consistent throughout the application's workflow.

mgineer85, 2023-12-12:
Hi, 
please see that the hflip / vflip options are bound to the backend capabilities currently as this the most efficient way to proces.
The settings you found are probably for the picamera2 backend or the opencv2 backend.

Maybe it could be changed so the setting is separate for main and secondary backend. If the backend has native option available it would use that or fallback to flip the resulting jpg.

Peda1996, 2023-12-12:
I think we could also flip the preview and captured image by just using cv2 (eg. for gphoto2). I don't think that this will introduce that much latency (and of course quality loss).

Eg.: similar edit to the gphoto2.py file like this:
```
        # Convert bytes to a numpy array and then to an OpenCV image
        nparr = np.frombuffer(img_bytes, np.uint8)
        frame = cv2.imdecode(nparr, cv2.IMREAD_COLOR)

        # Apply horizontal flip (1 for horizontal, 0 for vertical, -1 for both)
        if self.config.apply_hflip:
            frame = cv2.flip(frame, 1)

        # Apply vertical flip
        if self.config.apply_vflip:
            frame = cv2.flip(frame, 0)

        # Convert back to bytes
        _, flipped_img_bytes = cv2.imencode('.jpg', frame)

        # Now assign the flipped image data to the respective data holder
        with self._lores_data.condition:
            self._lores_data.data = flipped_img_bytes.tobytes()
            self._lores_data.condition.notify_all()
```

mgineer85, 2023-12-13:
Yes, need to find a general method. Cv2 would be one. Other could be pil or another jpeg lib. 
I will compare speed in a benchmark and choose fastest.
The booth would work on rpi 3 also so it needs most efficiency.
CPU load on rpi3 with livestream and display is only ~20% due to the optimizations.

mgineer85, 2023-12-13:
Apply above function is expensive and needs to run at live preview framerate.

PIL seems to be more efficient than cv2 in this.

```
--------------------------------------------------------------------------------------- benchmark 'flip_hires': 2 tests ---------------------------------------------------------------------------------------
Name (time in ms)                                 Min                 Max                Mean            StdDev              Median               IQR            Outliers     OPS            Rounds  Iterations
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
test_libraries_encode_hires[pillow_flip]     138.1011 (1.0)      139.9108 (1.0)      138.9313 (1.0)      0.6957 (1.13)     138.6959 (1.0)      1.2106 (1.14)          3;0  7.1978 (1.0)           8           1
test_libraries_encode_hires[cv2_flip]        281.6042 (2.04)     282.9538 (2.02)     282.4346 (2.03)     0.6155 (1.0)      282.7434 (2.04)     1.0574 (1.0)           1;0  3.5406 (0.49)          5           1
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

-------------------------------------------------------------------------------------- benchmark 'flip_lores': 2 tests --------------------------------------------------------------------------------------  
Name (time in ms)                                Min                Max               Mean            StdDev             Median               IQR            Outliers       OPS            Rounds  Iterations  
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------  
test_libraries_encode_lores[pillow_flip]      7.7765 (1.0)       8.9868 (1.0)       8.1391 (1.0)      0.2737 (1.08)      8.1142 (1.0)      0.3730 (1.07)         21;2  122.8641 (1.0)          67           1  
test_libraries_encode_lores[cv2_flip]        23.9784 (3.08)     25.0324 (2.79)     24.2686 (2.98)     0.2536 (1.0)      24.1327 (2.97)     0.3488 (1.0)           5;1   41.2056 (0.34)         41           1  
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------  

```
```

def pillow_flip(jpeg_bytes):
    # read image
    image = Image.open(io.BytesIO(jpeg_bytes))

    flipped_img = ImageOps.mirror(image)

    # encode to jpeg again
    byte_io = io.BytesIO()
    flipped_img.save(byte_io, format="JPEG", quality=85)
    bytes_full = byte_io.getbuffer()

    return bytes_full

def cv2_flip(jpeg_bytes):
    # convert to cv2 format that can be resized by cv2
    nparr = numpy.frombuffer(jpeg_bytes, numpy.uint8)
    img_np = cv2.imdecode(nparr, cv2.IMREAD_COLOR)

    # Apply horizontal flip (1 for horizontal, 0 for vertical, -1 for both)
    img_np_flipped = cv2.flip(img_np, 1)

    # and encode to jpeg again
    encode_param = [int(cv2.IMWRITE_JPEG_QUALITY), 90]
    result, encimg = cv2.imencode(".jpg", img_np_flipped, encode_param)

    return encimg

```

Try to find another one involving turbojpeg as this is usually the fastest.

Peda1996, 2023-12-13:
Sadly I don't have an raspberry pi 3 for testing, therefore I cannot test it myself.

Here would be another one involving turbojpeg:
`pip install PyTurboJPEG`

Here would be a code outline (in this case generated by chatgpt 😅), I will also try it out this evening.

```
from turbojpeg import TurboJPEG
import numpy as np

# Initialize TurboJPEG instance
jpeg = TurboJPEG()

# Your existing code to capture the image bytes (img_bytes)
# ...

# Decode JPEG image to BGR format
bgr_image = jpeg.decode(img_bytes)

# Apply flips using numpy
if self.config.apply_hflip:
    bgr_image = np.fliplr(bgr_image)
if self.config.apply_vflip:
    bgr_image = np.flipud(bgr_image)

# Encode the image back to JPEG
flipped_img_bytes = jpeg.encode(bgr_image)

# Now, use flipped_img_bytes as needed
# ...

```

mgineer85, 2023-12-13:
Yes, we have right now

- turbojpeg
- PIL
- opencv and
- numpy
that could do the flip and/or decode/encode again.

Based on the backend sometimes we could skip decode/encode because backend delivers just frames (picamera/cv2). Some deliver jpeg exclusive.

Maybe there are libs out there that can flip without de/encode the image and operate just on the compressed data. That would probably be most efficient...

Peda1996, 2023-12-13:
A quick Google search sadly didn't realy yield any results for native jpeg
flipping transformations.

For now I think the best option would be to  incooperate pil or turbojpeg,
(depending on what is faster).

mgineer85, 2023-12-13:
I for now chose to implement a mirror effect in css which is purely on display only, no images are changed actually.

This is somewhat in line with how phones take picture. Preview mirrored, actual photo is taken as is (not mirrored). This way if some text is in the background of the image, it can be read in the right direction :)

Revisit this later when pipelines for livepreview get more advanced (example live chromakey/backgroundremoval) ...

closed 2023-12-13 by mgineer85

Peda1996, 2023-12-13:
Thanks for implementing this :)

mgineer85, 2023-12-13:
Released: https://github.com/photobooth-app/photobooth-app/releases/tag/v0.22.0

Peda1996, 2023-12-14:
Is it possible that with this release the filter option is no longer working?