photobooth-app/photobooth-app #132: Feature Request: Dynamic ISO Setting for Live View and Photo Capture with EOS 2000D on gphoto2 Backend

Peda1996, 2023-12-12:
I am proposing the implementation of a dynamic ISO setting feature for the Photobooth-App when using the Canon EOS 2000D camera with the gphoto2 backend. This feature is especially important as the EOS 2000D does not support auto-brightness in live view mode, leading to challenges in achieving the correct exposure during live view and photo capture.

### Problem Statement
Currently, when setting a fixed ISO value, the live view tends to be either too dark (if the ISO is set low) or the final photo is overexposed (if the ISO is set high, especially when using an external flash).

### Proposed Workflow

1. Auto ISO in Live View: When the camera is in live view mode, set the ISO to auto. This will allow the camera to automatically adjust the ISO based on the ambient light, providing a clearer and more accurately exposed live view image for the user.
2. Fixed ISO for Photo Capture: When the user captures the photo, switch the ISO setting from auto to a predefined fixed value. This ensures that the photo is taken with a consistent exposure setting, suitable for the lighting conditions and the use of external flash if any.

mgineer85, 2023-12-12:
Under the hood https://github.com/jim-easterbrook/python-gphoto2 is used to control the camera.
It seems possible to add ISO or other settings when switch from preview to still capture mode and reverse.

Feel free to send a pull request as I do not have the plan to implement currently.
I have EOS1000D and usually use external permanent light to have the preview look good also.

Peda1996, 2023-12-12:
I will try to implement this, and create a PR then :).

Peda1996, 2023-12-12:
I created a PR for this :)
https://github.com/photobooth-app/photobooth-app/pull/135

mgineer85, 2023-12-13:
merged #135. Testing on camera now and soon going to release.

mgineer85, 2023-12-13:
My camera is in auto mode and just ignores the command. Testing manual tomorrow.
Seems it works for you, so I close.

Thanks for the patch!

closed 2023-12-13 by mgineer85

mgineer85, 2023-12-13:
Released: https://github.com/photobooth-app/photobooth-app/releases/tag/v0.22.0