photobooth-app/photobooth-app #133: Bug Report: Application Crash on Image Capture Failure in gphoto2 Backend

Peda1996, 2023-12-12:
## Description
The application crashes when attempting to capture an image using the gphoto2 backend, specifically when the image is black, and the maximum retry limit of 3 is reached. This issue typically occurs when the camera, such as the Canon EOS 2000D, is unable to find focus and fails to take a picture, likely leading to a timeout. An HTTP Exception is raised with the error message: `'NoneType' object has no attribute 'create_fileset_unprocessed'` (originating from `application.py`). Subsequently, the application requires a restart to resume functionality.

## Steps to Reproduce
1. Configure the application with a Canon EOS 2000D camera using the gphoto2 backend.
2. Attempt to capture an image in conditions where the camera cannot find focus, leading to a black image output.
3. Observe the behavior of the application as it retries image capture up to the maximum limit (3 attempts).

## Expected Behavior
The application should handle the failure gracefully, logging the error without crashing. It should then just end the cycle.

## Actual Behavior
The application crashes after failing to capture an image within 3 attempts, raising an HTTP Exception related to a `'NoneType' object` and necessitating a restart.

## Error Messages
```
2023-12-12 12:32:54,086 [CRITICAL] finally failed after 3 attempts to capture image! (processingservice.py:182)
2023-12-12 12:32:54,086 [   ERROR] finally failed after 3 attempts to capture image! (processingservice.py:310)
Traceback (most recent call last):
  File "/home/fotobox/.local/lib/python3.10/site-packages/photobooth/services/processingservice.py", line 308, in start_job_1pic
    self.start(JobModel.Typ.image, 1)
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/event.py", line 54, in trigger_event
    return event_instance.trigger(self, *args, **kwargs)
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/event.py", line 29, in trigger
    return machine._process(trigger_wrapper)
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/statemachine.py", line 288, in _process
    return self._processing_loop()
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/statemachine.py", line 303, in _processing_loop
    result = trigger()
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/event.py", line 27, in trigger_wrapper
    return self._trigger(trigger_data)
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/event.py", line 39, in _trigger
    if transition.execute(event_data):
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/transition.py", line 140, in execute
    result = machine._activate(event_data)
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/statemachine.py", line 336, in _activate
    self._callbacks_registry[target.enter].call(
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/callbacks.py", line 228, in call
    return [
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/callbacks.py", line 229, in <listcomp>
    callback(*args, **kwargs)
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/callbacks.py", line 29, in __call__
    result = self._callback(*args, **kwargs)
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/dispatcher.py", line 48, in __call__
    return self._cache(*args, **kwds)
  File "/home/fotobox/.local/lib/python3.10/site-packages/statemachine/signature.py", line 56, in method_wrapper
    return method(*ba.args, **ba.kwargs)
  File "/home/fotobox/.local/lib/python3.10/site-packages/photobooth/services/processingservice.py", line 183, in on_enter_capture
    raise RuntimeError(f"finally failed after {MAX_ATTEMPTS} attempts to capture image!")
RuntimeError: finally failed after 3 attempts to capture image!
2023-12-12 12:32:54,088 [    INFO] start postprocessing phase 1 (processingservice.py:196)
2023-12-12 12:32:54,088 [    INFO] postprocessing last capture: mediaitem=None (processingservice.py:200)
2023-12-12 12:32:54,088 [CRITICAL] 'NoneType' object has no attribute 'create_fileset_unprocessed' (processing.py:31)
2023-12-12 12:32:54,088 [   ERROR] HTTPException: HTTPException(status_code=500, detail="something went wrong, Exception: 'NoneType' object has no attribute 'create_fileset_unprocessed'") (application.py:91)
2023-12-12 12:32:54,089 [    INFO] 127.0.0.1:46652 - "GET /processing/chose/1pic HTTP/1.1" 500 (h11_impl.py:478)
2023-12-12 12:32:54,153 [   ERROR] error capture lores image for stream. timeout expired attempt=3/3, retrying (abstractbackend.py:141)
2023-12-12 12:32:54,153 [CRITICAL] critical error getting stream. failed to get lores image after 3 attempts. giving up! (abstractbackend.py:149)
2023-12-12 12:32:54,766 [    INFO] (gp_camera_capture [gphoto2-camera.c:1339]) 'camera->functions->capture (camera, type, path, context)' failed: -110 (port_log.py:20)
2023-12-12 12:32:54,768 [CRITICAL] error capture! check logs for errors. [-110] I/O in progress (gphoto2.py:285)
```

## Additional Information
- This issue has been consistently reproducible under specific conditions with the Canon EOS 2000D.

mgineer85, 2023-12-12:
I will try with my Canon ESO 1000D later. Please standby.

mgineer85, 2023-12-12:
I see your point, it's same with my Canon.
Two things going on:
1) trying to get a high quality image. due to no focus after 3 times it fails finally.
2) in the background the lores stream tries to continue but because blocked by high quality image processing no preview image can be sent. finally the stream fails.

1) I changed logging to look a bit less drastic and removed the exception. There are many errors still coming up, but we cannot say it's from autofocus failing because the camera does not provide that information.
2) I changed the timeout handling for stream generation slightly (timeout 2 times than high quality capture timeout). This way the stream recovers after the image failed.

In future one can add a notification about the failed image on the screen.
Will release a new version in a few minutes.

Please test and let me know if it solves your issue.

mgineer85, 2023-12-12:
Released: https://github.com/photobooth-app/photobooth-app/releases/tag/v0.21.0

Peda1996, 2023-12-12:
Hi,

I tried out your new fix, works flawlessly for this case, thanks for this
lighting fast update :).

Still, wouldn't it be better to fully retry eg forever? And reopen gphoto2
as eg this also then fixes the case when the camera gets disconnected and
reconnected (or turned off and on, eg on power outage).

Lg

mgineer85, 2023-12-12:
I understand what you mean.

When you mean to retry forever, you probably mean to check permanently in the background if the camera is unavailable and reconnect?
Thats different from trying to finish the capture job in a limited number of times.

Background check is implemented already. When you turn off the camera, check the logs - it will go to a failsafe mode and notices the camera is not available. It checks regularly to reconnect. Turn on the camera again and wait little time until it reconnects.

Whats missing unfortunately is to inform the frontend, that the stream is available again. Press F5 in the browser and the live-preview is back again :)

I leave this open, so it can be fixed one day properly.

Peda1996, 2023-12-13:
So the one thing missing is just the websocket (fastapi + vue)? 
eg. workflow would then be:
- on start, frontend subscribes to the backend websocket
- frontend listens to this endpoint for changes
- if some major changes happen, the websocket informs the frontend with a message (eg. refresh page)
- frontend reacts to this message and refreshes page
 
I could implement the serverside part. Sadly I'm not familiar with vuejs, so I will have to look into how to implement this.

mgineer85, 2023-12-13:
I tried to fix it but ran into issues.

It would be sufficient to hide the preview and show again on main screen. That would work, but unfortunately the browser (here chrome) keeps the old stream active and starts another stream. Thats not intended, did not find a way around this yet.

mgineer85, 2023-12-13:
Actual issue is solved, followup created #137

commit ab2a6738 (2023-12-12): fix #133
closed 2023-12-13 by mgineer85