photobooth-app/photobooth-app #134: Assistance with Setting Up Development Environment in PyCharm and Frontend Editing

Peda1996, 2023-12-12:
## Description
Hello, I've recently submitted several Bug/Feature Requests that I am interested in addressing myself. However, I'm encountering challenges in setting up the project in my local development environment for contribution.

## Issues Faced
1. **Local Environment Setup:** I attempted to install dependencies using `python -m pip install .`, but this approach doesn't seem to be working effectively.
2. **Frontend Editing:** I am unable to locate the source files for frontend development. Currently, I only find compressed JavaScript files.

## Specific Questions
1. Are there any specific steps or additional configurations required to set up this project in PyCharm?
2. Could you provide guidance on where to find the editable frontend files or how to work with the existing compressed JS files?

## Attempts Made
- Installed project dependencies via pip.
- Searched the repository for frontend source files, but only found compressed .js files.

Any assistance or direction would be greatly appreciated, as I am eager to contribute to this project. Thank you!

mgineer85, 2023-12-12:
There is a contribute.md in the root folder, but somewhat outdated ;)
Usually I use vscode and settings are commited in the project - maybe it would be easier to start with that as I do not know pycharm.

mgineer85, 2023-12-12:
The frontend you find over there: https://github.com/photobooth-app/photobooth-frontend
It's a quasar framework project, so their setup guide will work.

Once installed fire up the dev-server via `quasar dev` and it will start a proxy with hotreload accessing the local photobooth in the background.

Peda1996, 2023-12-12:
Had some issues with setting up the python project using `pdm` as I never used it before, added therefore documentation for it in my PR https://github.com/photobooth-app/photobooth-app/pull/135 

Thanks for your help, somehow also overlooked the contribute.md .. ^^

mgineer85, 2023-12-13:
Thanks!

I applied for pycharm free open source license. Maybe pycharm is better than VSCode? I will have a look. But application needs like 2 weeks maybe ^^

Peda1996, 2023-12-13:
I just use PyCharm Community Edition, as this is enough for me ^^. But I think choosing the right IDE just comes down to your preference. In my opinion PyCharm should have a better understanding of the underlying code (eg. autocompletion, ..) , and better git integration.

Peda1996, 2023-12-18:
As I already was able to setup my environment (backend), I will close this issue

closed 2023-12-18 by Peda1996