photobooth-app/photobooth-app #137: Reload stream if failed

mgineer85, 2023-12-13:
**Describe the bug**

#133 if backend crashes temporarily or network is down, ensure to reconnect the stream preview properly.

Peda1996, 2023-12-17:
You mentioned 
> 
> I tried to fix it but ran into issues.
> 
> It would be sufficient to hide the preview and show again on main screen. That would work, but unfortunately the browser (here chrome) keeps the old stream active and starts another stream. Thats not intended, did not find a way around this yet.
> 

Wouldn't an alternative (but little hacky) approach be, to just generate a White Image / or last buffered image, with an error message on it. (eg.: Live Preview Error), instead of creating an HTTPException, that basically kills the stream?

mgineer85, 2023-12-18:
Hm, yeah, maybe something similar like the simulated backend displaying something...
But it would need to try restart the backend to stream again then internally somehow - otherwise an f5 would be needed also.
I think about that...

Peda1996, 2023-12-18:
I was thinking of just editing the abstractbackend.py

instead of this section:
```
            else:
                # we failed finally all the attempts - deal with the consequences.
                logger.critical("critical error getting stream. " f"failed to get lores image after {MAX_ATTEMPTS} attempts. giving up!")

                # return to signal stop yielding frames to calling function
                return
```

we could just insert the last frame here, or an alternative frame here optionally, so this reason the stream isn't closed, and restarting the other stream is still handled the same way.

Peda1996, 2023-12-18:
Also started to work on it, in a way that the last Frame is used, and no SimulatedBackend for now

https://github.com/photobooth-app/photobooth-app/pull/144

mgineer85, 2023-12-18:
Do you actually have issues still with the liveview to fail? In which situations?

Peda1996, 2023-12-18:
Also added this first initialization fix.

Case 1: Well, when I initially start my Notebook, and the power cord isn't plugged in, the Camera has no power, and therefore the stream cannot start.
Case 2: Somebody removes the power cord while the party is running, therefore the camera gets disconnected, and the stream isn't continuing anymore.

so this PR should fix these issues, but does not add Error messages to the front-end: (eg. Camera not found or stuff like this..)

Peda1996, 2023-12-18:
@mgineer85 I think I am done with my changes :)

So pls consider reviewing and merging the PR :)

mgineer85, 2023-12-18:
> Also added this first initialization fix.
> 
> Case 1: Well, when I initially start my Notebook, and the power cord isn't plugged in, the Camera has no power, and therefore the stream cannot start. (my photobooth has a notebook with akku integrated) Case 2: Somebody removes the power cord while the party is running, therefore the camera gets disconnected, notebook stays on (because akku), and the stream isn't continuing anymore.
> 
> so this PR should fix these issues, but does not add Error messages to the front-end: (eg. Camera not found or stuff like this..)
> 
> edit: and yes, had the second case at my brothers party ^^

Ah, I see. Your booth is not kind of a box but loosely wired? 

I assumed a closed box with all devices protected an start on power on automatically.

nevermind, will review and merge later, thx!

Peda1996, 2023-12-18:
If u want, I can create an example Project setup for the Documentation of my booth :)

mgineer85, 2023-12-18:
https://photobooth-app.org/projects/

Could add it here if you want to. Just post whatever you want to have added.

mgineer85, 2023-12-18:
Or here https://github.com/photobooth-app/photobooth-app/discussions/categories/show-and-tell

mgineer85, 2023-12-20:
implemented streaming a substitude image in case backends time out.
this way the stream is kept upright and the client displays the livestream again when the camera is connected again.

closed 2023-12-20 by mgineer85