photobooth-app/photobooth-app #138: Feature Request: Automated Photo Copying to External USB Device

Peda1996, 2023-12-13:
The photobooth application currently allows users to take and store photos locally. However, there might be a need for immediate and easy transfer of these photos to an external storage device, like a USB stick.

### Feature Description:
I propose a feature that automatically detects the connection of an external USB device and copies all new photos (or photos that haven't been copied yet) to this device.

This could then be just run in another thread / daemon.

mgineer85, 2023-12-13:
Currently there are 2 options
- QR code shareservice using single php script on the other side and
- Download zip in files tab in the backend.

I see it could be useful. Maybe use a simple Linux service? Wouldn’t work on windows though.

Peda1996, 2023-12-14:
This is my take on it :)
https://github.com/photobooth-app/photobooth-app/pull/139

mgineer85, 2023-12-17:
Slightly modified version is added to repo now. Thx for the pull request.
Will release a new version very soon.

Peda1996, 2023-12-17:
Thanks for refactoring my code, and accepting this PR :)

closed 2023-12-17 by Peda1996