photobooth-app/photobooth-app #140: Help needed: I somehow cannot get my

Peda1996, 2023-12-14:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - OS: [e.g. Raspberry Pi Bookworm 64bit]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mgineer85, 2023-12-14:
You sure the php server is setup properly? If I try to access the URL I get timeout also.

mgineer85, 2023-12-14:
Highly recommend just any random shared hoster for this service.

Peda1996, 2023-12-14:
Well, the thing is I can open my test.php file that includes my phpinfo(); eg.: /test.php, so basically my php server should run without problems.

mgineer85, 2023-12-14:
Is SQLite supported by the server? 
Whats in the PhotoBooth logs? 
to debug check the php script and enable logging temporarily to file also

Peda1996, 2023-12-14:
SQLite is supported.

error logs are enabled, but nothing realy of interest can be found:
```
[14-Dec-2023 21:42:55 UTC] runtime error: photobooth did not upload the requested image within time :( no internet? service disabled?
[14-Dec-2023 21:42:55 UTC] {"action":"download","id":"test"}
[14-Dec-2023 21:42:55 UTC] []
[14-Dec-2023 21:42:55 UTC] []
```

i kind of found, that shareservice.py is accessing `upload_queue` from the dl.php which has a timeout of 240 seconds. After waiting for these 240 seconds, also the browser returned a valid response, with a lot of ping? messages.
eg.:
`{"ping":1702590549} {"ping":1702590549} {"ping":1702590550} {"ping":1702590550} {"ping":1702590551} {"ping":1702590551} {"ping":1702590552} {"ping":1702590552} {"ping":1702590553} {"ping":1702590553} {"ping":1702590554},...
`

reducing this max loop time to 5 seconds also wasn't that much helpful, but changed the error of the backend (eg. shareservice.py)

```
2023-12-14 23:02:16,065 [    INFO] successfully connected to shareservice dl.php script (shareservice.py:98)
2023-12-14 23:02:16,065 [   DEBUG] dl.php script finished after some time. stopiteration issued-reconnect (shareservice.py:109)
2023-12-14 23:02:16,065 [    INFO] request timed out, error occured or shutdown requested (shareservice.py:174)
2023-12-14 23:02:16,066 [    INFO] restarting loop wait 1 second (shareservice.py:176)
```

Peda1996, 2023-12-14:
Issue can be closed, it was related to streaming responses and long-running requests. I fixed this by changing my nginx .php part config to this

```
    location / {
        try_files $uri $uri/ /index.php$is_args$args;
    }

	location ~ \.php$ {
    	include snippets/fastcgi-php.conf;
    	fastcgi_pass unix:/run/php/php8.2-fpm.sock;
    	fastcgi_read_timeout 500;  # Set this to a value that suits your needs
        fastcgi_buffering off;
	}

    location ~ /\.ht {
        deny all;
    }
```

closed 2023-12-14 by Peda1996

mgineer85, 2023-12-14:
Ah, good to know!
My hoster uses apache and I did not take care about any config.
I added a link to the docs in section shareservice troubleshooting description to this issue for future reference.

WebfinityDesign, 2024-07-05:
I think I do have the same problem, only my server is Apache. Any idea how to solve this on an Apache server?

mgineer85, 2024-07-05:
There was a lot going on here and I am not very sure what the problem is exactly. Please create a new discussion and post all logs from the app as well as from the server to start debugging .