photobooth-app/photobooth-app #141: Wled Flash Time

skar-cgn, 2023-12-16:
**Is your feature request related to a problem? Please describe.**
I use the photobox app together with an led stripe powered by wled. I added the proper presets in wled. Everything is working as expected. Only the flash time is super short and therefore does not have a big effect. The camera can not react on this additional light. 

**Describe the solution you'd like**
Define the time the flash preset of the wled should be activated

**Describe alternatives you've considered**
Tried to change other settings in wled like transition time, but no success.

mgineer85, 2023-12-16:
Probably it would make sense to have the strip light during countdown already at like 80%? When shot raise to 90% do the user has feedback but doesn’t interfere with the capture?
This way during countdown camera has all time to settle. 
does this work for you?

skar-cgn, 2023-12-16:
I am using a color stripe and play a "waiting" rainbow like effect to indicate the preperation for the picture. I could change it to play white during preparation but would prefer the color effect.

mgineer85, 2023-12-16:
I think it’s possible to have two separate strips to wled and apply different settings for the same preset to them. So you could keep the rainbow plus have additional light.

mgineer85, 2023-12-16:
To have a better understanding could you describe how many lights you have, their type and the camera?

mgineer85, 2023-12-20:
closing this as I think it can be resolved within WLED. Please feel free to reopen if you need additional customization.

closed 2023-12-20 by mgineer85

WebfinityDesign, 2024-07-03:
I think I have more or less the same setup. I do have a LED-ring around the camera that is running colors while waiting and it is switching to white when actually taking the picture. It seems like the timing of the white light is not always at the exact moment the picture is taken. When I take a photo with a mirror sometimes I see a white ring and other times I see the colored waiting light. Is there an adjustment I can try to tweak?

mgineer85, 2024-07-03:
No it’s not. There is no way to sync like a flash.
You might want to change your presets so the countdown is white-ish moving to signal countdown and then have an animation with static white. The amount of light should be similar so the cam can adjust properly.

WebfinityDesign, 2024-07-03:
Thank you for your very quick response. Do I understand it correctly it could have something to do with the difference in the amount of light that is used for the countdown and the amount of light used for the "flash"?
My "flash" is a lot brighter then the countdown. I use a RGBW ring. For countdown the RGB part is set to red/green. In the flash mode the RGB is set to white and the white LEDs come on as well.
I will see if I can tweak something there.

skar-cgn, 2024-07-03:
Hi all, 
I solved the topic with the Playlist feature of Wled. 
So I am calling the Playlist instead of the preset and just organize the countdown timing is similar to the playlist playtime. 
In the Playlist I start with some blinking lights and afterwards white very bright leds. In the second phase the camera has enough time to adjust to the "new" brightness. 
Regards
Ralf

WebfinityDesign, 2024-07-03:
A playlist was my previous approach. I switched to the control by the photobooth app because I thought that it would be more accurate. I will play a bit with it and see what works best.