photobooth-app/photobooth-app #142: Change QR Code visibility, based on Internet connectivitiy

Peda1996, 2023-12-18:
Only show the QR Code when the fotobox is connected to the internet, eg. (make a configurable ping address, or ping the shareservice), and only show the QR code when the internet connectivity is given.

mgineer85, 2023-12-18:
Good idea. Right now there is mechanism to inform the frontend if shareservice is working fine and to display/not display the qr code.   I leave it here for now as I would consider a missing internet connection as error that needs recovered.

Peda1996, 2023-12-18:
How often is the frontend about the status of the shareservice informed?

I somehow thought of pinging the server everytime the image gets loaded. So it also automatically would recover.

An even more advanced approach would be to also somehow set a uploaded flag and then also make the qr code for these images visible, that were already uploaded. (ensures visibility despite a missing internet connection for already uploaded images.)

mgineer85, 2023-12-18:
The service knows whether it’s working or not properly. It’s just there is no mechanism to inform the frontend.

regarding the display of the QR code after upload: the mechanism is actually smarter. When the code is scanned, the file is requested from the booth, uploaded to php host and served from there again. Needs little extra time but saves data.

Peda1996, 2023-12-18:
Okay, I was thinking of creating a web socket anyways, eg. for informing the user when eg. file transfers to an usb stick are done, or there is no space left on the stick etc. There this could be also then send as message.

But we have to decide on a message structure for the websocket then^^.

mgineer85, 2023-12-18:
There is the sse service for this. And the informationservice… but it feels all somewhat cluttered already right now. 😂

Peda1996, 2023-12-18:
Oh, somehow I completely missed this, but it isn't the typical fastapi websocket implementaion or, (as my search didn't yield any results with searching for "WebSocket")?

Will have a look at this later

mgineer85, 2023-12-18:
Its sse, serverside events. No bidirectional communication yet, because not really needed

mgineer85, 2023-12-18:
Maybe sse could be replaced in frontend and backend because websockets seem like first class citizen to fastapi and vue with good support

Peda1996, 2023-12-18:
Never worked with sse to be honest, so never heard of it

Peda1996, 2023-12-18:
Sorry, accidentally pressed close with comment instead of comment.

Striker62, 2024-03-31:
maybe i did get it right but the QRcode makes also sence without internet connection.
I setup an acces point and this give the chance to get the photo to your mobile device.

or do you meant if there is also no acces point is setup.

mgineer85, 2024-03-31:
I think you will want to configure method b: https://photobooth-app.org/setup/shareservice/#method-b-share-via-local-wifi

Striker62, 2024-03-31:
oh I'm fine with the actual solution. I only wanted to inform you that from my point of view a internet connection should not be the only reason for the QRCode.

mgineer85, 2025-06-03:
I will not actively implement this right now. Maybe it happens one day as many items from my (previously internal) list have happened.

I added an item to the project list to keep it in mind but closing this one now.

If anybody is willing to add this feature, I think we should keep in mind that QR codes in the future could be printed on hardcopies also. With that in mind we need to rely on the images being available at one point anyways and checking for the actual availability is not really needed.

-> https://github.com/orgs/photobooth-app/projects/2/views/1?pane=issue&itemId=113646742

closed 2025-06-03 by mgineer85