photobooth-app/photobooth-app #148: Camera - application shuts down itself after a few hours

RobertLukan, 2023-12-23:
The applications works very well and I am using it for computer vision(machine learning) in combination with Frigate and Double Take and several cameras. Double take make an API call to take a picture and it works very well. However, I noticed that the application stops talking pictures after a while. I checked the logs and I found following output:

2023-12-22 18:16:06,788 [   DEBUG] Allocated 2 buffers for stream 2 with fds [56, 59] (dmaallocator.py:64)
2023-12-22 18:16:06,810 [    INFO] Camera started (picamera2.py:1126)
2023-12-22 18:16:06,833 [    INFO] switchmode finished successfully (picamera2_.py:344)
2023-12-22 18:16:07,712 [    INFO] Saved <picamera2.request.Helpers object at 0x7f64b5dad0> to file <_io.BytesIO object at 0x7f557878d0>. (request.py:289)
2023-12-22 18:16:07,713 [    INFO] Time taken for encode: 222.40730302291922 ms. (request.py:290)
2023-12-22 18:16:07,725 [   DEBUG] change to preview mode requested (picamera2_.py:309)
2023-12-22 18:16:07,734 [    INFO] aquired still_image, 2377062bytes to be sent to client (aquisition.py:58)
2023-12-22 18:16:07,736 [    INFO] 192.168.123.10:37382 - "GET /aquisition/still HTTP/1.1" 200 (h11_impl.py:478)
2023-12-22 18:16:07,835 [    INFO] switch_mode invoked, stopping stream encoder, switch mode and restart encoder (picamera2_.py:329)
2023-12-22 18:16:08,197 [    INFO] Camera stopped (picamera2.py:1180)
2023-12-22 18:16:08,215 [    INFO] Configuration successful! (picamera2.py:1051)
2023-12-22 18:16:08,254 [   DEBUG] Allocated 2 buffers for stream 0 with fds [38, 41] (dmaallocator.py:64)
2023-12-22 18:16:08,258 [   DEBUG] Allocated 2 buffers for stream 1 with fds [44, 47] (dmaallocator.py:64)
2023-12-22 18:16:08,270 [   DEBUG] Allocated 2 buffers for stream 2 with fds [50, 53] (dmaallocator.py:64)
2023-12-22 18:16:08,295 [    INFO] Camera started (picamera2.py:1126)
2023-12-22 18:16:08,315 [    INFO] switchmode finished successfully (picamera2_.py:344)
2023-12-22 20:19:11,060 [    INFO] Shutting down (server.py:264)
2023-12-22 20:19:11,162 [    INFO] Finished server process [48416] (server.py:86)
2023-12-22 20:19:11,168 [   ERROR] Traceback (most recent call last):
  File "/home/pi/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/starlette/routing.py", line 686, in lifespan
    await receive()
  File "/home/pi/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/uvicorn/lifespan/on.py", line 137, in receive
    return await self.receive_queue.get()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/queues.py", line 158, in get
    await getter
asyncio.exceptions.CancelledError
 (on.py:134)
2023-12-22 20:19:11,179 [    INFO] FileTransferService stopped. (filtetransferservice.py:41)
2023-12-22 20:19:11,256 [    INFO] _generate_images_fun left (picamera2_.py:413)
2023-12-22 20:19:11,783 [    INFO] Camera stopped (picamera2.py:1180)
2023-12-22 20:19:12,183 [    INFO] Camera closed successfully. (picamera2.py:625)
2023-12-22 20:19:12,184 [   DEBUG] photobooth.services.backends.picamera2_ stopped,  self._generate_images_thread.is_alive()=False,  self._stats_thread.is_alive()=False (picamera2_.py:231)

Sometimes the application recovers by itself or by visiting the web interface. But sometimes it does not. 

The log output after I refreshed website a few times:

2023-12-23 08:35:41,627 [   DEBUG] loggingservice __init__ finished (loggingservice.py:130)
2023-12-23 08:35:41,628 [   DEBUG] registered handlers: [<StreamHandler <stderr> (NOTSET)>, <FileHandler /home/pi/photobooth-data/log/photobooth_20231223.log (NOTSET)>, <EventstreamLogHandler (NOTSET)>] (loggingservice.py:131)
2023-12-23 08:35:42,084 [    INFO] photobooth directory: /home/pi/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth (__main__.py:81)
2023-12-23 08:35:42,086 [    INFO] working directory: /home/pi/photobooth-data (__main__.py:83)
2023-12-23 08:35:42,087 [    INFO] app version started: 0.90.0 (__main__.py:84)
2023-12-23 08:35:42,217 [    INFO] picamera2_focuser_module is disabled. Select a focuser module in config to enable autofocus. (picamera2_.py:103)
2023-12-23 08:35:42,381 [    INFO] Initialization successful. (picamera2.py:447)
2023-12-23 08:35:42,383 [    INFO] Camera now open. (picamera2.py:468)
2023-12-23 08:35:42,387 [   DEBUG] change to preview mode requested (picamera2_.py:309)
2023-12-23 08:35:42,389 [    INFO] Camera configuration has been adjusted! (picamera2.py:1046)
2023-12-23 08:35:42,392 [    INFO] Configuration successful! (picamera2.py:1051)
2023-12-23 08:35:42,398 [   DEBUG] Allocated 2 buffers for stream 0 with fds [38, 41] (dmaallocator.py:64)
2023-12-23 08:35:42,401 [   DEBUG] Allocated 2 buffers for stream 1 with fds [44, 47] (dmaallocator.py:64)
2023-12-23 08:35:42,406 [   DEBUG] Allocated 2 buffers for stream 2 with fds [50, 53] (dmaallocator.py:64)
2023-12-23 08:35:42,407 [    INFO] camera_config: {'use_case': 'video', 'transform': <libcamera.Transform 'hvflip'>, 'colour_space': <libcamera.ColorSpace 'Rec709'>, 'buffer_count': 2, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (1280, 720), '>
2023-12-23 08:35:42,408 [    INFO] camera_controls: {'ExposureValue': (-8.0, 8.0, 0.0), 'AeExposureMode': (0, 3, 0), 'AeMeteringMode': (0, 3, 0), 'AwbMode': (0, 7, 0), 'AeFlickerMode': (0, 1, 0), 'AnalogueGain': (1.0, 22.2608699798584, None), 'Sharpness>
2023-12-23 08:35:42,408 [    INFO] controls: <Controls: {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (33333, 33333)}> (picamera2_.py:175)
2023-12-23 08:35:42,409 [    INFO] set_ae_exposure, try to set to 1 (picamera2_.py:319)
2023-12-23 08:35:42,409 [    INFO] current picamera2.controls.get_libcamera_controls():{libcamera.ControlId(44, NoiseReductionMode, ControlType.Integer32): <NoiseReductionModeEnum.Fast: 1>, libcamera.ControlId(28, FrameDurationLimits, ControlType.Intege>
2023-12-23 08:35:42,410 [    INFO] stream quality Quality[self._config.backends.picamera2_stream_quality.name]=<Quality.MEDIUM: 2> (picamera2_.py:178)
2023-12-23 08:35:42,677 [    INFO] Camera started (picamera2.py:1126)
2023-12-23 08:35:42,949 [   DEBUG] photobooth.services.backends.picamera2_ started (picamera2_.py:209)
2023-12-23 08:35:42,951 [    INFO] init self.primary_backend=<class 'photobooth.services.backends.picamera2_.Picamera2Backend'> (aquisitionservice.py:39)
2023-12-23 08:35:42,951 [    INFO] init self.secondary_backend=None (aquisitionservice.py:40)
2023-12-23 08:35:42,953 [    INFO] init database and creating missing scaled images. this might take some time. (mediacollectionservice.py:59)
2023-12-23 08:35:42,959 [    INFO] initialized image DB, added 8 valid images (mediacollectionservice.py:76)
2023-12-23 08:35:42,959 [    INFO] -- process time: 0.0s to initialize mediacollection (mediacollectionservice.py:77)
2023-12-23 08:35:42,968 [    INFO] on_enter_state self.current_state.id='idle'  (processingservice.py:103)
2023-12-23 08:35:42,969 [    INFO] state idle entered. (processingservice.py:126)
2023-12-23 08:35:42,971 [    INFO] gpio not enabled - enable for gpio support on raspberry pi (gpioservice.py:60)
[... 10 lines cut ...]
2023-12-23 08:35:42,993 [    INFO] psutil.virtual_memory()=svmem(total=952446976, available=582533120, percent=38.8, used=302440448, free=252108800, active=370130944, inactive=115113984, buffers=21606400, cached=376291328, shared=364544, slab=95535104) >
2023-12-23 08:35:42,994 [    INFO] __debug__=False (informationservice.py:60)
2023-12-23 08:35:42,994 [    INFO] initialized information service (informationservice.py:62)
2023-12-23 08:35:42,995 [    INFO] keyboardservice not enabled - enable for keyboard triggers (keyboardservice.py:55)
2023-12-23 08:35:42,996 [    INFO] shareservice disabled, start aborted. (shareservice.py:52)
2023-12-23 08:35:42,997 [    INFO] FileTransferService disabled, start aborted. (filtetransferservice.py:30)
2023-12-23 08:35:42,997 [    INFO] WledService disabled (wledservice.py:40)
2023-12-23 08:35:42,998 [    INFO] Welcome to photobooth-app (__main__.py:28)
2023-12-23 08:35:42,998 [    INFO] multiprocessing_start_method=None (__main__.py:35)
2023-12-23 08:35:43,003 [   DEBUG] Using selector: EpollSelector (selector_events.py:54)
2023-12-23 08:35:43,079 [    INFO] Started server process [56096] (server.py:76)
2023-12-23 08:35:43,081 [    INFO] Waiting for application startup. (on.py:46)
2023-12-23 08:35:43,082 [    INFO] Application startup complete. (on.py:60)
[... cut ...]
2023-12-23 08:38:38,570 [   DEBUG] change to capture mode requested (picamera2_.py:304)
2023-12-23 08:38:38,571 [    INFO] switch_mode invoked, stopping stream encoder, switch mode and restart encoder (picamera2_.py:329)
2023-12-23 08:38:38,991 [    INFO] Camera stopped (picamera2.py:1180)

The application will work for a few hours and it will stop working after a while.

The system is working on Raspberry Pi CM4 with custom carrier board. 

pi@rpi-cm4-3:~ $ uname -ran
Linux rpi-cm4-3 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux

mgineer85, 2023-12-23:
Hi, thanks for using the software.

In first log it seems there is a normal shutdown going on (maybe system shutdown or service shutdown triggered) at 2023-12-22 20:19:11,060 [    INFO] Shutting down (server.py:264).   The following error can be ignored because it's forcing the server to shutdown quite hackish way and aborts processes to do so. So we can ignore that one probably.

Second log is fine up to the last line, where the log ends - is the application indeed stuck there or just cut of the logs? I have seen similar in the past, but fixed.

You use bookworm and are on latest apt update/upgrades?

There were recently some bugs found in picamera2 that could be related. Last release was early november, so maybe the next picamera2 release would fix it for you?

https://github.com/raspberrypi/picamera2/pull/864
https://github.com/raspberrypi/picamera2/pull/876

How often does this happen to you?
Maybe you can upgrade to the latest picamera2 from git repo? (I did not test it)

RobertLukan, 2023-12-25:
Thank you for very quick response. Right now I am not at home, so I will test it more in a few days. I am using 64 bit bookworm. I noticed that everything works for a while(maybe like 2 hours) and after a while when there are no pictures taken = api calls are failing - connection is refused.

I will upgrade RPI to the latest version(patches) and lets see if something changes.

I would like to say thank you for very nice software. I am using it for a slightly different purpose. I will update this ticket in a few days once I come back home.

mgineer85, 2023-12-25:
Ok, wait for it. 
maybe you do no actually need to switch between two modes for capture/video?
It could be removed for you, only little to modify.

also the full logs would be great.
You only send the one api key to the server, nothing else?

RobertLukan, 2024-01-18:
I am not sure what happened. It was working for a while now. But I got a few crashes yesterday. Here is the log:

I have not changed anything really, after those crashes in December. It was working for 3 weeks now perfectly. The camera shutdown itself after 6 hours and it causes an python exception. It recovers itself after about 12 hours.

I will try update/upgrade RPI OS. I will be testing it more. Otherwise I am very happy with this software.

2024-01-17 17:56:29,849 [ WARNING] force switchmode to capture config right before taking picture (picamera2_.py:373)
2024-01-17 17:56:29,850 [   DEBUG] change to capture mode requested (picamera2_.py:304)
2024-01-17 17:56:29,851 [    INFO] switch_mode invoked, stopping stream encoder, switch mode and restart encoder (picamera2_.py:329)
2024-01-17 17:56:30,241 [    INFO] Camera stopped (picamera2.py:1180)
2024-01-17 17:56:30,255 [    INFO] Configuration successful! (picamera2.py:1051)
2024-01-17 17:56:30,310 [   DEBUG] Allocated 2 buffers for stream 0 with fds [38, 41] (dmaallocator.py:64)
2024-01-17 17:56:30,314 [   DEBUG] Allocated 2 buffers for stream 1 with fds [44, 47] (dmaallocator.py:64)
2024-01-17 17:56:30,339 [   DEBUG] Allocated 2 buffers for stream 2 with fds [50, 53] (dmaallocator.py:64)
2024-01-17 17:56:30,366 [    INFO] Camera started (picamera2.py:1126)
2024-01-17 17:56:30,388 [    INFO] switchmode finished successfully (picamera2_.py:344)
2024-01-17 17:56:31,269 [    INFO] Saved <picamera2.request.Helpers object at 0x7f74aed850> to file <_io.BytesIO object at 0x7f74b3b3d0>. (request.py:289)
2024-01-17 17:56:31,270 [    INFO] Time taken for encode: 222.92680099963036 ms. (request.py:290)
2024-01-17 17:56:31,280 [   DEBUG] change to preview mode requested (picamera2_.py:309)
2024-01-17 17:56:31,284 [    INFO] aquired still_image, 1992445bytes to be sent to client (aquisition.py:58)
2024-01-17 17:56:31,287 [    INFO] 192.168.123.10:37010 - "GET /aquisition/still HTTP/1.1" 200 (h11_impl.py:478)
2024-01-17 17:56:31,386 [    INFO] switch_mode invoked, stopping stream encoder, switch mode and restart encoder (picamera2_.py:329)
2024-01-17 17:56:31,744 [    INFO] Camera stopped (picamera2.py:1180)
2024-01-17 17:56:31,761 [    INFO] Configuration successful! (picamera2.py:1051)
2024-01-17 17:56:31,800 [   DEBUG] Allocated 2 buffers for stream 0 with fds [38, 41] (dmaallocator.py:64)
2024-01-17 17:56:31,805 [   DEBUG] Allocated 2 buffers for stream 1 with fds [44, 47] (dmaallocator.py:64)
2024-01-17 17:56:31,809 [   DEBUG] Allocated 2 buffers for stream 2 with fds [50, 53] (dmaallocator.py:64)
2024-01-17 17:56:31,830 [    INFO] Camera started (picamera2.py:1126)
2024-01-17 17:56:31,856 [    INFO] switchmode finished successfully (picamera2_.py:344)
2024-01-17 17:56:32,258 [ WARNING] force switchmode to capture config right before taking picture (picamera2_.py:373)
2024-01-17 17:56:32,259 [   DEBUG] change to capture mode requested (picamera2_.py:304)
2024-01-17 17:56:32,259 [    INFO] switch_mode invoked, stopping stream encoder, switch mode and restart encoder (picamera2_.py:329)
2024-01-17 17:56:32,295 [    INFO] Camera stopped (picamera2.py:1180)
2024-01-17 17:56:32,303 [    INFO] Configuration successful! (picamera2.py:1051)
2024-01-17 17:56:32,345 [   DEBUG] Allocated 2 buffers for stream 0 with fds [38, 41] (dmaallocator.py:64)
2024-01-17 17:56:32,349 [   DEBUG] Allocated 2 buffers for stream 1 with fds [44, 47] (dmaallocator.py:64)
2024-01-17 17:56:32,370 [   DEBUG] Allocated 2 buffers for stream 2 with fds [50, 53] (dmaallocator.py:64)
2024-01-17 17:56:32,391 [    INFO] Camera started (picamera2.py:1126)
2024-01-17 17:56:32,415 [    INFO] switchmode finished successfully (picamera2_.py:344)
2024-01-17 17:56:33,309 [    INFO] Saved <picamera2.request.Helpers object at 0x7f74aed850> to file <_io.BytesIO object at 0x7f6d683fb0>. (request.py:289)
2024-01-17 17:56:33,310 [    INFO] Time taken for encode: 228.71558599945274 ms. (request.py:290)
2024-01-17 17:56:33,320 [   DEBUG] change to preview mode requested (picamera2_.py:309)
2024-01-17 17:56:33,323 [    INFO] aquired still_image, 2026802bytes to be sent to client (aquisition.py:58)
2024-01-17 17:56:33,326 [    INFO] 192.168.123.10:37020 - "GET /aquisition/still HTTP/1.1" 200 (h11_impl.py:478)
2024-01-17 17:56:33,425 [    INFO] switch_mode invoked, stopping stream encoder, switch mode and restart encoder (picamera2_.py:329)
2024-01-17 17:56:33,781 [    INFO] Camera stopped (picamera2.py:1180)
2024-01-17 17:56:33,799 [    INFO] Configuration successful! (picamera2.py:1051)
2024-01-17 17:56:33,840 [   DEBUG] Allocated 2 buffers for stream 0 with fds [38, 41] (dmaallocator.py:64)
2024-01-17 17:56:33,843 [   DEBUG] Allocated 2 buffers for stream 1 with fds [44, 47] (dmaallocator.py:64)
2024-01-17 17:56:33,849 [   DEBUG] Allocated 2 buffers for stream 2 with fds [50, 53] (dmaallocator.py:64)
2024-01-17 17:56:33,870 [    INFO] Camera started (picamera2.py:1126)
2024-01-17 17:56:33,892 [    INFO] switchmode finished successfully (picamera2_.py:344)
2024-01-17 17:56:43,810 [ WARNING] force switchmode to capture config right before taking picture (picamera2_.py:373)
2024-01-17 17:56:43,811 [   DEBUG] change to capture mode requested (picamera2_.py:304)
2024-01-17 17:56:43,812 [    INFO] switch_mode invoked, stopping stream encoder, switch mode and restart encoder (picamera2_.py:329)
2024-01-17 17:56:44,230 [    INFO] Camera stopped (picamera2.py:1180)
[... cut ...]
2024-01-18 11:19:33,923 [    INFO] multiprocessing_start_method=None (__main__.py:35)
2024-01-18 11:19:33,931 [   DEBUG] Using selector: EpollSelector (selector_events.py:54)
2024-01-18 11:19:34,019 [    INFO] Started server process [23161] (server.py:76)
2024-01-18 11:19:34,020 [    INFO] Waiting for application startup. (on.py:46)
2024-01-18 11:19:34,021 [    INFO] Application startup complete. (on.py:60)
2024-01-18 11:19:34,022 [    INFO] Uvicorn running on http://0.0.0.0:8000 (Press CTRL+C to quit) (server.py:218)

mgineer85, 2024-01-19:
Hi,

is there any additional logs in between the lines
```
2024-01-17 17:56:48,368 [    INFO] switchmode finished successfully (picamera2_.py:344)
2024-01-17 23:23:16,233 [    INFO] Shutting down (server.py:264)
```
?

What was causing the shutdown? This is usually an external signal like restart of the service or reboot of the server.

It seems you want to take a picture every second? You also use the livestream now or was this only for setup?

Until the picamera2 gets updated including the bugfixes, I changed the camera implementation so it should recover itself from such errors. Update to the latest app version that includes the changes already.

RobertLukan, 2024-01-19:
Not really, also log rotation happened. I am using double-take software to take a snapshot when a person is detected by simply using following api call: /aquisition/still

I configured double-take to take only one photo. I am not using live stream through photobooth, only taking one photo once a person is detected by using a second camera.

I will update the software and lets see.

Thank you for your help.

mgineer85, 2024-01-20:
Hey, I changed some internal handling to avoid switching between the camera modes after each call to /aquisition/still.

If you want, you can test it by upgrading to the current dev-branch:
https://photobooth-app.org/setup/update/#update-to-development-versions

RobertLukan, 2024-01-20:
Sure, I will give a try. I have upgraded to version 1.2 and for **now** everything is OK. But I will switch to dev also and let you know.

RobertLukan, 2024-01-24:
Ok, I upgraded(Installed) dev version: pipx install --system-site-packages git+https://github.com/photobooth-app/photobooth-app.git@dev --force 

I did not allow me to upgrade so I force installed it. I tested taking one picture over http interface and it works. Lets see what happens when API calls will be made.

This is the log from the dev version:

2024-01-24 11:32:19,169 [   DEBUG] loggingservice __init__ finished (loggingservice.py:114)
2024-01-24 11:32:19,196 [   DEBUG] registered handlers: [<StreamHandler <stderr> (NOTSET)>, <FileHandler /home/pi/photobooth-data/log/photobooth_20240124.log (NOTSET)>, <EventstreamLogHandler (NOTSET)>] (loggingservice.py:115)
2024-01-24 11:32:19,200 [    INFO] platform.system()='Linux' (informationservice.py:34)
2024-01-24 11:32:19,201 [    INFO] platform.release()='6.1.0-rpi4-rpi-v8' (informationservice.py:35)
2024-01-24 11:32:19,201 [    INFO] platform.machine()='aarch64' (informationservice.py:36)
2024-01-24 11:32:19,202 [    INFO] platform.python_version()='3.11.2' (informationservice.py:37)
2024-01-24 11:32:19,202 [    INFO] platform.node()='rpi-cm4-3' (informationservice.py:38)
2024-01-24 11:32:19,202 [    INFO] psutil.cpu_count()=4 (informationservice.py:39)
2024-01-24 11:32:19,204 [    INFO] psutil.cpu_count(logical=False)=4 (informationservice.py:40)
2024-01-24 11:32:19,226 [    INFO] psutil.disk_partitions()=[sdiskpart(device='/dev/mmcblk0p2', mountpoint='/', fstype='ext4', opts='rw,noatime', maxfile=255, maxpath=4096), sdiskpart(device='/dev/mmcblk0p1', mountpoint='/boot/firmware', fstype='vfat', opts='rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro', maxfile=1530, maxpath=4096)] (informationservice.py:41)
2024-01-24 11:32:19,226 [    INFO] psutil.disk_usage('/')=sdiskusage(total=14793007104, used=10589331456, free=3432239104, percent=75.5) (informationservice.py:43)
2024-01-24 11:32:19,228 [    INFO] [('lo', ['127.0.0.1']), ('wlan0', ['192.168.1.160']), ('br-0653ed2e77f9', ['172.18.0.1']), ('docker0', ['172.17.0.1']), ('veth38efdcf', []), ('veth965e39c', []), ('eth0', [])] (informationservice.py:46)
2024-01-24 11:32:19,229 [    INFO] psutil.virtual_memory()=svmem(total=952446976, available=512475136, percent=46.2, used=370958336, free=87281664, active=613666816, inactive=76632064, buffers=25636864, cached=468570112, shared=1835008, slab=88715264) (informationservice.py:55)
2024-01-24 11:32:19,230 [    INFO] __debug__=False (informationservice.py:57)
2024-01-24 11:32:19,230 [    INFO] initialized information service (informationservice.py:59)
2024-01-24 11:32:19,232 [    INFO] init database and creating missing scaled images. this might take some time. (mediacollectionservice.py:55)
2024-01-24 11:32:19,233 [    INFO] watching for filetypes: ['.jpg', '.gif'] (mediacollectionservice.py:58)
2024-01-24 11:32:19,244 [    INFO] initialized image DB, added 9 valid images (mediacollectionservice.py:74)
2024-01-24 11:32:19,245 [    INFO] -- process time: 0.01s to initialize mediacollection (mediacollectionservice.py:75)
2024-01-24 11:32:19,253 [    INFO] on_enter_state self.current_state.id='idle'  (processingservice.py:104)
2024-01-24 11:32:19,253 [    INFO] state idle entered. (processingservice.py:127)
2024-01-24 11:32:19,254 [    INFO] initialized systemservice (systemservice.py:29)
2024-01-24 11:32:19,255 [    INFO] gpio not enabled - enable for gpio support on raspberry pi (gpioservice.py:60)
2024-01-24 11:32:19,556 [    INFO] photobooth directory: /home/pi/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth (__main__.py:43)
2024-01-24 11:32:19,557 [    INFO] working directory: /home/pi/photobooth-data (__main__.py:45)
2024-01-24 11:32:19,557 [    INFO] app version started: 1.2.0 (__main__.py:46)
2024-01-24 11:32:19,558 [    INFO] Welcome to the photobooth-app (__main__.py:49)
2024-01-24 11:32:19,560 [    INFO] started LoggingService (container.py:90)
2024-01-24 11:32:19,560 [    INFO] WledService disabled (wledservice.py:40)
2024-01-24 11:32:19,561 [    INFO] started WledService (container.py:90)
2024-01-24 11:32:21,785 [    INFO] aquisition main backend: <class 'photobooth.services.backends.picamera2.Picamera2Backend'> (aquisitionservice.py:61)
2024-01-24 11:32:21,786 [    INFO] aquisition live backend: None (aquisitionservice.py:62)
2024-01-24 11:32:21,798 [    INFO] connect_thread device status=EnumDeviceStatus.initialized, so trying to start (abstractbackend.py:122)
2024-01-24 11:32:21,804 [    INFO] started AquisitionService (container.py:90)
2024-01-24 11:32:21,809 [    INFO] setting device status from initialized to starting (abstractbackend.py:102)
2024-01-24 11:32:21,831 [    INFO] trying to start backend (abstractbackend.py:127)
2024-01-24 11:32:21,837 [    INFO] started InformationService (container.py:90)
2024-01-24 11:32:21,842 [    INFO] device is available (abstractbackend.py:131)
2024-01-24 11:32:21,853 [    INFO] started MediaprocessingService (container.py:90)
2024-01-24 11:32:21,979 [    INFO] started MediacollectionService (container.py:90)
2024-01-24 11:32:22,065 [    INFO] started SystemService (container.py:90)
2024-01-24 11:32:22,065 [    INFO] started PrintingService (container.py:90)
2024-01-24 11:32:22,066 [    INFO] keyboardservice not enabled - enable for keyboard triggers (keyboardservice.py:54)
2024-01-24 11:32:22,066 [    INFO] started KeyboardService (container.py:90)
2024-01-24 11:32:22,067 [    INFO] started GpioService (container.py:90)
[... cut ...]
2024-01-24 11:33:02,892 [    INFO] 192.168.0.37:51765 - "GET /js/90.80f4c3bc.js HTTP/1.1" 200 (h11_impl.py:478)
2024-01-24 11:33:56,639 [    INFO] 192.168.0.37:51780 - "GET /css/51.0d14ffdb.css HTTP/1.1" 200 (h11_impl.py:478)
2024-01-24 11:33:56,643 [    INFO] 192.168.0.37:51781 - "GET /js/51.574198bf.js HTTP/1.1" 200 (h11_impl.py:478)
2024-01-24 11:33:56,760 [    INFO] 192.168.0.37:51781 - "GET /admin/config/currentActive HTTP/1.1" 200 (h11_impl.py:478)
2024-01-24 11:33:56,889 [    INFO] 192.168.0.37:51780 - "GET /admin/config/schema?schema_type=dereferenced HTTP/1.1" 200 (h11_impl.py:478)

mgineer85, 2024-01-24:
2024-01-24 11:32:22,726 [CRITICAL] control not available on camera - autofocus not working properly Control AfMode is not advertised by libcamera (picamera2.py:295)

indicates that the arducam camera installation might have been broken during an apt update / apt upgrade.

RobertLukan, 2024-01-24:
This camera does not have AF, so maybe that is the reason for this message. I just tested the DEV version and it works so far.

mgineer85, 2024-01-24:
Ok. Then it’s fine.

mgineer85, 2024-01-27:
Version 1.2.1 released including the fixed you tested in dev.
Closing this now. Please reopen or create a new issue if you still have problems.

closed 2024-01-27 by mgineer85

mgineer85, 2024-02-17:
With ref to https://github.com/photobooth-app/photobooth-app/issues/148#issuecomment-1900548918

Picamera2 released updated version yesterday. With that stalling procedure should definitely not be an issue any more.