photobooth-app/photobooth-app #155: Deleted Images Remain Visible as White Squares in Frontend

Peda1996, 2024-01-05:
After deleting images in the application, they are not immediately removed from the frontend. Instead, they remain visible as a white square (where the photo was). When however the white image is opened, the original one can be seen. Reloading the page, however, removes the deleted images entirely from the frontend.

## Steps to Reproduce
1. Delete an image within the application.
2. Observe that the image becomesa white square post-deletion (and is not entirely gone).
3. Reload the page.
4. Observe the state of the deleted image (image is gone).

## Expected Behavior
Once an image is deleted, it should be instantly removed from the frontend without leaving any traces or placeholders.

## Actual Behavior
- Deleted images remain visible in the frontend when clicked on the white square of the deleted image.
- A page reload is required to fully remove the deleted images from the frontend view.

## Environment
- Application Version: 1.1.1
- Browser: Tested on Firefox and Chromium
- Operating System: Ubuntu

mgineer85, 2024-01-05:
I cannot reproduce unfortunately.
Do you have something in the logs related when you click the button to delete?
Could you check also if there are error sin the browser development tools console (in chrome its CTRL+SHIFT+i)

That you can open the image is probably because it is still in the cache.
During page reload the full database is synchronized again, that is why the white square is gone then. Seems delete works but not the notification to the browser.

Peda1996, 2024-01-05:
Was able to reproduce the same issue on my development environment, with the current main branch.

Server Logs:
```
2024-01-05 18:05:52,542 [    INFO] gallery_delete requested, id=a5279e9246c7f7462284151dafe93780 (mediacollection.py:25)
2024-01-05 18:05:52,542 [    INFO] request delete item id a5279e9246c7f7462284151dafe93780 (mediacollectionservice.py:167)
2024-01-05 18:05:52,543 [   DEBUG] found item=MediaItem Id: a5279e9246c7f7462284151dafe93780, filename image_show_20240105-180541-805308.jpg (mediacollectionservice.py:172)
2024-01-05 18:05:52,543 [    INFO] request delete files of MediaItem Id: a5279e9246c7f7462284151dafe93780, filename image_show_20240105-180541-805308.jpg (mediacollectionservice.py:188)
2024-01-05 18:05:52,544 [    INFO] deleted files of MediaItem Id: a5279e9246c7f7462284151dafe93780, filename image_show_20240105-180541-805308.jpg (mediacollectionservice.py:202)
2024-01-05 18:05:52,544 [   DEBUG] deleted mediaitem from db and files MediaItem Id: a5279e9246c7f7462284151dafe93780, filename image_show_20240105-180541-805308.jpg (mediacollectionservice.py:183)
2024-01-05 18:05:52,544 [    INFO] 127.0.0.1:51596 - "GET /mediacollection/delete?image_id=a5279e9246c7f7462284151dafe93780 HTTP/1.1" 204 (h11_impl.py:478)
2024-01-05 18:05:52,551 [    INFO] livestream requested from main backend (aquisitionservice.py:132)
2024-01-05 18:05:52,552 [    INFO] 127.0.0.1:51596 - "GET /aquisition/stream.mjpg HTTP/1.1" 200 (h11_impl.py:478)
2024-01-05 18:05:52,552 [    INFO] livestream started on backend backend_to_stream_from=<class 'photobooth.services.backends.gphoto2.Gphoto2Backend'> (aquisitionservice.py:204)
```

Frontend Logs:
```
ProcessStateinfo 
Object { state: "captures_completed", typ: "image", total_captures_to_take: 1, remaining_captures_to_take: 0, number_captures_taken: 1, duration: 0, ask_user_for_approval: false, confirmed_captures_collection: (1) […], last_captured_mediaitem: {…} }
[app.223dd9e0.js:1:2939](http://localhost:8000/js/app.223dd9e0.js)
received new item to add to collection: 
Object { mediaitem: {…} }
[app.223dd9e0.js:1:3062](http://localhost:8000/js/app.223dd9e0.js)
ProcessStateinfo 
Object { state: "present_capture", typ: "image", total_captures_to_take: 1, remaining_captures_to_take: 0, number_captures_taken: 1, duration: 0, ask_user_for_approval: false, confirmed_captures_collection: (1) […], last_captured_mediaitem: {…} }
[app.223dd9e0.js:1:2939](http://localhost:8000/js/app.223dd9e0.js)
0 [547.e54f752f.js:1:5933](http://localhost:8000/js/547.e54f752f.js)
starting newitemarrived timer, duration=30 [547.e54f752f.js:1:8236](http://localhost:8000/js/547.e54f752f.js)
ProcessStateinfo 
Object { state: "idle", typ: "image", total_captures_to_take: 1, remaining_captures_to_take: 0, number_captures_taken: 1, duration: 0, ask_user_for_approval: false, confirmed_captures_collection: (1) […], last_captured_mediaitem: {…} }
[app.223dd9e0.js:1:2939](http://localhost:8000/js/app.223dd9e0.js)
Object { data: "", status: 204, statusText: "No Content", headers: {…}, config: {…}, request: XMLHttpRequest }
[547.e54f752f.js:1:7393](http://localhost:8000/js/547.e54f752f.js)
GEThttp://localhost:8000/media/processed/thumbnail/image_show_20240105-180541-805308.jpg
[HTTP/1.1 404 Not Found 2ms]

GEThttp://localhost:8000/media/processed/thumbnail/image_show_20240105-180407-100533.jpg
[HTTP/1.1 404 Not Found 3ms]

GEThttp://localhost:8000/media/processed/thumbnail/image_show_20240105-180335-790325.jpg
[HTTP/1.1 404 Not Found 3ms]
```

In this case I directly captured the photo, and directly deleted it from the preview

Gallery Example:
[image]

mgineer85, 2024-01-05:
Thank you for the elaborated bug report.
I can confirm the issue for the itempresenter (after capture), but not when the viewer is opened from the gallery.
I fixed it in the frontend. Will be included in the next release.

closed 2024-01-05 by mgineer85

Peda1996, 2024-01-05:
Thanks, will test it as soon as possible :)

mgineer85, 2024-01-07:
New version is released: https://github.com/photobooth-app/photobooth-app/releases/tag/v1.2.0