photobooth-app/photobooth-app #156: Bug Report: Filters Not Applied on Click

Peda1996, 2024-01-05:
When attempting to apply filters to images, either newly captured or from the gallery, clicking on the filters does not produce any change. This issue occurs in two scenarios: applying filters to newly captured images and applying filters to existing images in the gallery.

## Steps to Reproduce

### Scenario 1: Newly Captured Image
1. Capture an image using the application.
2. Click on the 'Filter' option.
3. Select any filter from the available options.
4. Observe that the filter is not applied to the image.

### Scenario 2: Image from Gallery
1. Open any image from the gallery within the application.
2. Click on the 'Filter' option.
3. Choose any filter from the list.
4. Notice that the filter is not applied to the image.

## Expected Behavior
Selecting a filter should immediately apply the chosen effect to the image, whether it is a newly captured image or an image from the gallery.

## Actual Behavior
- In both scenarios, selecting a filter does not apply any change to the image.
- The issue is consistent, with no response from the application when a filter is selected.

## Note
Filters are applied when using the Collage function. But are not working with the Filter selection list somehow.

mgineer85, 2024-01-05:
You need to enable the pipelineprocessing in general. You see anything about that in the logs after you clicked?
The filter button should be hidden if disabled or just apply the filter even if the filter pipeline is disabled in future.

Peda1996, 2024-01-05:
Oh, my bad, this was a mistake on my side. Someone at my brothers party must have disabled it, and I only updated just now and assumed it was because of the update.

Thanks

closed 2024-01-05 by Peda1996

mgineer85, 2024-01-05:
OK, no problem.
Still the filter application is a bit weird for Collages and GIFs. I will create a followup for this.