photobooth-app/photobooth-app #163: Main backend runs with live backend set for the live preview at gphoto2

Paradoxa, 2024-01-26:
**Describe the bug**
The live preview runs in the gphoto backend (main backend) with Livepreview Enabled, although Picamera2 is selected for the live preview.

**To Reproduce**
Steps to reproduce the behavior:
1. choose 'admin/config/backends' -> Main Backend -> Gphoto2
2. choose 'admin/config/backends' -> Live Backend -> Picamera2
3. choose 'admin/config/backends' -> Livepreview Enabled -> on
4. See Log on 'admin/status' 

**Expected behavior**
The live preview is in the gphoto backend (main backend) if Livepreview Enabled is on and Picamera2 is selected for the live backend, then the live preview is switched off in the main backend.
Maybe not right but possible solution:
In the file "photobooth-app/photobooth/services/backends/gphoto2.py" line 200 expanded
from "_if appconfig.backends.LIVEPREVIEW_ENABLED:_" -> to "_if appconfig.backends.LIVEPREVIEW_ENABLED and appconfig.backends.LIVE_BACKEND.value == "Disabled":_"

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System (please complete the following information):**
 - Raspberry Pi Bookworm 64bit
 - chrome

**Additional context**
Add any other context about the problem here.

mgineer85, 2024-01-27:
fixed by ec42f08770f87bd16f2da35e71f09b6d0a09a931

closed 2024-01-27 by mgineer85

mgineer85, 2024-01-27:
Version 1.2.1 is released including a fix. Thx for reporting!