photobooth-app/photobooth-app #168: Keep Raw on Camera?

samyness, 2024-02-03:
How to keep RAW pictures on my Camera using gphoto2?

Is there a way to keep RAW's on the Camera, like here:
https://github.com/andreknieriem/photobooth/wiki/FAQ#how-to-keep-pictures-on-my-camera-using-gphoto2

mgineer85, 2024-02-03:
Not implemented currently.

Maybe you want to add it?
We use python gphoto2 https://github.com/jim-easterbrook/python-gphoto2/tree/main/examples to connect the camera.
The logic is here: https://github.com/photobooth-app/photobooth-app/blob/main/photobooth/services/backends/gphoto2.py

mgineer85, 2024-02-07:
additional information on this topic: 
https://github.com/photobooth-app/photobooth-app/discussions/9#discussioncomment-8367046

mgineer85, 2024-02-11:
Hi,

I improved the gphoto2 backend to 

- keep raw on camera and
- avoid stuck if shot with raw+jpeg.

As a playground I used following script which works fine on my canon 1100D.
To keep raw on camera you need to set the capture target to something permanent. For canon it is memory card.
After capture it will flush the eventqueue now that avoids getting stuck gphoto on second capture.

```
import gphoto2 as gp
import os

c = gp.Camera()
c.init()
# defined events http://www.gphoto.org/doc/api/gphoto2-camera_8h.html#a438ab2ac60ad5d5ced30e4201476800b and
# https://github.com/gphoto/libgphoto2/blob/846ae9f2199eb8bcbcb6d8bfe47fc961a033519a/gphoto2/gphoto2-camera.h#L103
event_texts = {}
for name in ('GP_EVENT_UNKNOWN', 'GP_EVENT_TIMEOUT', 'GP_EVENT_FILE_ADDED',
                'GP_EVENT_FOLDER_ADDED', 'GP_EVENT_CAPTURE_COMPLETE', 'GP_EVENT_FILE_CHANGED'):
    event_texts[getattr(gp, name)] = name

print(event_texts)

# change capturetarget
cfg = c.get_config()
capturetarget_cfg = cfg.get_child_by_name('capturetarget')
print(capturetarget_cfg)
capturetarget = capturetarget_cfg.get_value()
print(capturetarget)
capturetarget_cfg.set_value('Internal RAM')
#capturetarget_cfg.set_value("Memory card")
capturetarget = capturetarget_cfg.get_value()
print(capturetarget)
gp.check_result(gp.gp_camera_set_config(c, cfg))

for i in range(2):
	captured_files:list[tuple[str:str]]=[]
	print("Capture #%1d" % i)

	data = c.capture(gp.GP_CAPTURE_IMAGE)
	captured_files.append((data.folder,data.name))

	#empty the event queue
	typ,data = c.wait_for_event(200)
	while typ != gp.GP_EVENT_TIMEOUT:
		
		print(f"Event: {event_texts.get(typ,f'unknown event index: {typ}')}, data: {data}")
		
		if typ == gp.GP_EVENT_FILE_ADDED:
			captured_files.append((data.folder,data.name))

		#try to grab another event
		typ,data = c.wait_for_event(10) # timeout in ms

	print("thats all for now...")

	print(captured_files)

	file_to_download=None
	for captured_file in captured_files:
		_, file_extension = os.path.splitext(captured_file[1])
		print(file_extension)
		if str(file_extension).lower() in (".jpg", ".jpeg"):
			print("ok, this file to download!")
			file_to_download=captured_file

			break
		
	if file_to_download is None:
		raise RuntimeError("no capture or no jpeg captured! shooting in raw-only mode?")
	
	f2 = c.file_get(captured_file[0],captured_file[1],gp.GP_FILE_TYPE_NORMAL)
	dest2 = os.path.join("/tmp",captured_file[1])
	f2.save(dest2)
	print(f"saved {dest2}")

		
c.exit()
```