photobooth-app/photobooth-app #169: BETA: Video Capture in v2

mgineer85, 2024-02-03:
There is a video capture implementation in version 2 coming up🎉✨😊

Still in beta, following items need to finish:
- [ ] Add Rest-Api to capture (like still)
- [ ] Add tests
- [ ] finalize routes (start, stop)
- [ ] add keyboard to trigger video
- [ ] add keyboard to stop video
- [ ] add gpio to trigger video
- [ ] add gpio to stop video
- [ ] refine handling of scaled versions in gallery
- [ ] option to print video? (also true for gif) or just disable for now?
- [ ] share video via QR works?
- [ ] improve countdown in UI

PixelAlarm, 2024-02-13:
Hi,

really looking forward to the final implementation.

I think some parameters like video-lenght and framerate are customizable (at least in the backend-code).
Is there a possibility to add a functionality to export the video in a boomerang-style gif?

mgineer85, 2024-02-13:
Video length default is a parameter in admin backend.
Framerate not yet, but I add it because it's nice.

boomerang is not planned for the first release currently. Also I think mp4/h264 makes more sense than a gif as h264 is widely supported and messenger like whatsapp can convert easily if desired by the user.

Striker62, 2024-04-03:
The video is captured from the mainbackend ?
If the mainbackend is not support liveview no video capture right ?

The new version is great

Striker62, 2024-04-03:
after a while the liveview screen flip to a random screen. Is the a switch in the config to deactivate this flip?

Striker62, 2024-04-03:
Found it "Show Automatic Slideshow Timeout" set to 0

mgineer85, 2024-04-03:
> The video is captured from the mainbackend ? If the mainbackend is not support liveview no video capture right ?
> 
> The new version is great

As long as no second backend is configured, it tries to get video / preview from main backend. If a second one is configured, that is used for live/video exclusively and main only for stills.

Striker62, 2024-04-04:
sorry do not get this .
I have a webcam configured   EnumImageBackendsMain (it is realy named like this) with gphoto2 my Nikon D3100 (no liveview possible) and EnumImageBackendsLive with Webcamcv2 and a webcam .

so any chance to use the webcam for video recording without changing the gphoto2 ?

mgineer85, 2024-04-04:
Yes, use main gphoto and live webcamcv for video. Gphoto will capture only stills in this case.
Earlier you referred to use webcamcv2 two times which is just not working that way.

Striker62, 2024-04-05:
sorry for confusion ( i would prefer to discuss in german ) I'm testing both releases. 
The question regards webcamcv2 was related to the actual release. The idea here was to reduce hardware for a smaler Booth.

The question regards Video is for the version 2

Striker62, 2024-04-05:
[image]
beside the 2 Typos.

The video catrue via Webcam is working :-) I try to view the video not on the raspberry attached screen in chrome but on my pc connected to the webserver on my pi with firefox.

In chrome on localhost:8000 it is shown.

Strange but it is an issue with my Firefox setup

mgineer85, 2024-04-08:
Version 2 is released now: https://github.com/photobooth-app/photobooth-app/releases/tag/v2.0.0
The original issue is closed.

closed 2024-04-08 by mgineer85