photobooth-app/photobooth-app #176: Gphoto2 Disable Viewfinder Before Capture causes timeout receiving frames (abstractbackend.py:253)

samyness, 2024-02-07:
**Describe the bug**
If I activate the Gphoto2 Disable Viewfinder Before Capture, I'm getting an "timeout receiving frames (abstractbackend.py:253)" error and the camera triggers two shots.

**To Reproduce**
Steps to reproduce the behavior:
activate Gphoto2 Disable viewfinder before capture with a D810 

**Expected behavior**
I expacted better Autofocus!

**Screenshots**
If applicable, add screenshots to help explain your problem.

LOG: 

`2024-02-07 18:02:17,325 [    INFO] livestream started on backend backend_to_stream_from=<class 'photobooth.services.backends.gphoto2.Gphoto2Backend'> (aquisitionservice.py:169)
2024-02-07 18:02:17,326 [    INFO] setting device enable_lores_flag from False to True on backend Gphoto2Backend (abstractbackend.py:115)
2024-02-07 18:02:17,520 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,542 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,569 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,591 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,613 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,635 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,657 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,680 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,702 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,725 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,747 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,769 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,791 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,814 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,836 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,864 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,886 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,908 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,930 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:17,953 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:20,280 [    INFO] set up job to start: typ=<Typ.image: 'image'>, total_captures_to_take=1 (processingservice.py:85)
2024-02-07 18:02:20,281 [    INFO] start job typ=Typ.image, total_captures_to_take=1, confirmed_captures_collection=[], last_capture=None (processingservice.py:93)
2024-02-07 18:02:20,282 [    INFO] on_exit_state self.current_state.id='idle'  (processingservice.py:112)
2024-02-07 18:02:20,282 [    INFO] on_enter_state self.current_state.id='counting'  (processingservice.py:104)
2024-02-07 18:02:20,283 [    INFO] WledService preset_thrill triggered (wledservice.py:136)
2024-02-07 18:02:20,284 [ WARNING] WLED module not connected, ignoring request (wledservice.py:148)
2024-02-07 18:02:20,284 [    INFO] setting custom iso value: 50 (gphoto2.py:168)
2024-02-07 18:02:20,284 [ WARNING] cannot set iso, command ignored [-110] I/O in progress (gphoto2.py:171)
2024-02-07 18:02:20,285 [    INFO] setting custom shutter speed: 1/250 (gphoto2.py:178)
2024-02-07 18:02:20,285 [ WARNING] cannot set shutter speed, command ignored [-110] I/O in progress (gphoto2.py:181)
2024-02-07 18:02:20,286 [    INFO] start countdown, duration_user=duration=5.0, offset_camera=0.25 (processingservice.py:161)
2024-02-07 18:02:24,991 [    INFO] on_exit_state self.current_state.id='counting'  (processingservice.py:112)
2024-02-07 18:02:24,993 [    INFO] on_enter_state self.current_state.id='capture'  (processingservice.py:104)
2024-02-07 18:02:24,994 [    INFO] current capture (1/1, remaining 0) (processingservice.py:177)
2024-02-07 18:02:24,995 [   DEBUG] capture to filepath_neworiginalfile=PosixPath('media/original/image_show_20240207-180224-995267.jpg') (processingservice.py:191)
2024-02-07 18:02:24,995 [    INFO] WledService preset_shoot triggered (wledservice.py:141)
2024-02-07 18:02:24,996 [ WARNING] WLED module not connected, ignoring request (wledservice.py:148)
2024-02-07 18:02:24,997 [    INFO] disable viewfinder before capture (gphoto2.py:239)
2024-02-07 18:02:25,107 [    INFO] (_get_config [config.c:11188]) Type of property 'Lens Name' expected: 0x0002 got: 0x0004 (port_log.py:20)
2024-02-07 18:02:25,108 [    INFO] (_get_config [config.c:11204]) Widget get of property 'Lens Name' failed, trying to see if we have another... (port_log.py:20)
2024-02-07 18:02:25,113 [    INFO] (_get_config [config.c:11188]) Type of property 'Available Shots' expected: 0x0006 got: 0x0004 (port_log.py:20)
2024-02-07 18:02:25,393 [    INFO] (_get_config [config.c:11188]) Type of property 'Exposure Program' expected: 0x0002 got: 0xffff (port_log.py:20)
2024-02-07 18:02:25,457 [    INFO] (_get_config [config.c:11188]) Type of property 'Live View AF Mode' expected: 0x0001 got: 0x0002 (port_log.py:20)
2024-02-07 18:02:25,458 [    INFO] (_get_config [config.c:11204]) Widget get of property 'Live View AF Mode' failed, trying to see if we have another... (port_log.py:20)
2024-02-07 18:02:25,475 [    INFO] (_get_config [config.c:11204]) Widget get of property 'Live View Exposure Preview' failed, trying to see if we have another... (port_log.py:20)
2024-02-07 18:02:25,626 [    INFO] (_get_config [config.c:11188]) Type of property 'Auto White Balance Bias' expected: 0x0002 got: 0x0004 (port_log.py:20)
2024-02-07 18:02:25,630 [    INFO] (_get_config [config.c:11188]) Type of property 'Tungsten White Balance Bias' expected: 0x0002 got: 0x0004 (port_log.py:20)
2024-02-07 18:02:25,635 [    INFO] (_get_config [config.c:11188]) Type of property 'Fluorescent White Balance Bias' expected: 0x0002 got: 0x0004 (port_log.py:20)
2024-02-07 18:02:25,640 [    INFO] (_get_config [config.c:11188]) Type of property 'Daylight White Balance Bias' expected: 0x0002 got: 0x0004 (port_log.py:20)
2024-02-07 18:02:25,653 [    INFO] (_get_config [config.c:11188]) Type of property 'Flash White Balance Bias' expected: 0x0002 got: 0x0004 (port_log.py:20)
2024-02-07 18:02:25,658 [    INFO] (_get_config [config.c:11188]) Type of property 'Cloudy White Balance Bias' expected: 0x0002 got: 0x0004 (port_log.py:20)
2024-02-07 18:02:25,668 [    INFO] (_get_config [config.c:11188]) Type of property 'Shady White Balance Bias' expected: 0x0002 got: 0x0004 (port_log.py:20)
2024-02-07 18:02:26,479 [    INFO] taking hq picture (gphoto2.py:243)
[... 159 lines cut ...]
2024-02-07 18:02:35,094 [    INFO] 192.168.0.19:61599 - "GET /mediaprocessing/preview/2c5dbd943da67f894865cc25d4dc5f23/_1977 HTTP/1.1" 200 (h11_impl.py:474)
2024-02-07 18:02:35,123 [    INFO] 192.168.0.19:61601 - "GET /mediaprocessing/preview/2c5dbd943da67f894865cc25d4dc5f23/aden HTTP/1.1" 200 (h11_impl.py:474)
2024-02-07 18:02:35,152 [    INFO] 192.168.0.19:61600 - "GET /mediaprocessing/preview/2c5dbd943da67f894865cc25d4dc5f23/ashby HTTP/1.1" 200 (h11_impl.py:474)
2024-02-07 18:02:35,603 [    INFO] Camera file path: //capt0001.jpg (gphoto2.py:247)
[... cut ...]
2024-02-07 18:02:36,179 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:36,202 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:36,224 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)
2024-02-07 18:02:36,247 [    INFO] (ptp_usb_getresp [usb.c:516]) PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019) (port_log.py:20)`

mgineer85, 2024-02-07:
Yeah, maybe the viewfinder disable is only good for canon and harmful for nikon?
How is the capture speed compared if you disable liveview to liveview enabled?
How are the absolute times?

samyness, 2024-02-08:
I need to check the absolute times first. Just based on feeling, there's no difference. 

Do you know what triggers the second photo? Actually, taking the second photo is the only mistake. The correct, first capture is displayed and saved and the Camera focus is **much** better than with activ viewfinder.

mgineer85, 2024-02-08:
Is it actually taking a second photo or maybe just the control sequence that sounds like it but no photo taken?

samyness, 2024-02-08:
It takes a second Picture 6seconds later:
```
2024-02-07 18:02:26,479 [    INFO] taking hq picture (gphoto2.py:243)
2024-02-07 18:02:32,165 [    INFO] taking hq picture (gphoto2.py:243)
```

samyness, 2024-02-08:
Should I try to set the Timeout receicing frames in gphoto.py to 10sec.? 

```
def _wait_for_hq_image(self):
        with self._hires_data.condition:
            self._hires_data.request_ready.set()
            if not self._hires_data.condition.wait(timeout=10):
                raise TimeoutError("timeout receiving frames")

        self._hires_data.request_ready.clear()
        return self._hires_data.data
```

mgineer85, 2024-02-08:
That’s very Long. 
you can try but I prefer to find a way with shorter capture time.
It would be interesting if it’s faster with only jpeg capture.

samyness, 2024-02-08:
Thats only jpeg capture, but in highest quality (35mb/picture).

mgineer85, 2024-02-08:
Holy, ok, I think transfer time comes here to an reasonable amount. Also when switching to raw plus storing to the sd card it will take even longer until all is settled.

I think the modes used are not so reasonable for a PhotoBooth like this or might need heavy optimization. Processing the images afterwards might need some time also?

samyness, 2024-02-08:
```
2024-02-07 18:02:32,995 [    INFO] -- process time: 1.59s to create scaled images (processingservice.py:237)
2024-02-07 18:02:33,355 [    INFO] -- process time: 0.36s to process singleimage (processingservice.py:242)
```

mgineer85, 2024-02-08:
> ```
> 2024-02-07 18:02:32,995 [    INFO] -- process time: 1.59s to create scaled images (processingservice.py:237)
> 2024-02-07 18:02:33,355 [    INFO] -- process time: 0.36s to process singleimage (processingservice.py:242)
> ```

Hehe, ok the turbojpeg implementation is really good 🙂

samyness, 2024-02-08:
1. I'll try to set the quality lower and check what happens. (Strange that the delay only occurs when the viewfinder is deactivated. Maby the focusing takes to much time, the Imagequality stays the same.)
2. If that doesn't work, I'll try to set the timeout to 10sec. and see what happens. 

Is it normal that "device timed out deliver lores image" during taking hq picture?

mgineer85, 2024-02-08:
I understood the absolute capture time is about the same with viewfinder enabled/disabled?

You can elongate the timeout, no problem. But it will give a bad user experience with such long times. To compare, on my Canon it takes about 0.5-1seconds until the capture is done.

The lores image deliver timeout is normal. The app keeps asking for the livestream during the capture. When the camera is busy, it should just ignore the requests...
Try to disable liveview and see if the camera behaves better/faster.

samyness, 2024-02-08:
**I understood the absolute capture time is about the same with viewfinder enabled/disabled?**
No, the capture time is about 3sec. with enabled viewfinder and about 6sec. with disabled viewfinder!

**Try to disable liveview and see if the camera behaves better/faster.**
With disabled liveview the capture takes about 5.5sec.

samyness, 2024-02-08:
With a 7-second timeout, usually only one photo is taken now. It's rare that the camera triggers twice. 
I'm a bit confused; in the admin settings, I set the Takepic Msg Time to 0.5, but the white screen appears already at a countdown time of 1.3.

samyness, 2024-02-08:
I have set the retries to 40. Currently, it takes me 5.5 seconds to capture the image."
` def wait_for_lores_image(self, retries: int = 40):` 
in abstractbackend.py

mgineer85, 2024-02-10:
this prerelease comes with improvements for gphoto2:
- timeout raised from 4s to 8s
- during still capture, the lores stream requests are paused.

https://github.com/photobooth-app/photobooth-app/releases/tag/v2.0a3

mgineer85, 2024-03-26:
I'm closing this as solved. If you have issues, please reopen.

closed 2024-03-26 by mgineer85