photobooth-app/photobooth-app #182: Improve translation integration

mgineer85, 2024-02-10:
To ensure good translation quality (completeness) and lower the barrier to contribute additional languages for users consider:
- [ ] creating a github actions workflow, testing for missing translations
- [ ] fallback to englisch (which needs 100% coverage)
- [ ] integrate with a service like crowdin.com

andrii-bodnar, 2024-02-14:
Hey, just accidentally stumbled across this issue. 

I would recommend using the Crowdin's [GitHub Action](https://github.com/crowdin/github-action) to automate the localization synchronization between the repository and a Crowdin project.

Also, Crowdin offers a [free license for Open-Source projects](https://crowdin.com/page/open-source-project-setup-request).

mgineer85, 2024-02-14:
Hi! Yes, implanting GitHub action is the way to go 👍

mgineer85, 2024-04-10:
### Help Translate the App

We use [automated pre-translation powered by Crowdin](https://crowdin.com/project/photobooth-app) now.
Help to improve the accuracy by proofreading the translation.
It's very easy to translate and proofread the current translations:

- [Open the photobooth-app translation project in Crowdin](https://crowdin.com/project/photobooth-app/invite?h=b00f8c8abec20ed573058db633f2452c2057822).
- Click the dropdown button next to your language to open language details.
- Click the button "translate" or "proofread".
- Work through the items, improve the translation and click the check mark to approve.
- If the language is 100% translated and proofread, it will be included in the next release.
- Leave a short notice in the discussions that you improved the translation to let us know. Thank you!

closed 2024-04-10 by mgineer85