photobooth-app/photobooth-app #191: USB copy does not work on RPI4 ?

hillerpa, 2024-03-22:
**Describe the bug**
Maybe I'm the idiot but I already renamed my USB to photobooth and created a folder with photobooth, 

**Expected behavior**
expectation would have been that it is automatically copying the pictures if the service is updated but it is not doing it . Do I need to formate it somehow special or something?

**Screenshots**
USB Folder is Empty

**System (please complete the following information):**
 - OS: Raspberry Pi Bookworm 64bit
 - Browser chrome / Firefox
 - Version latest

Thank you for the great work

mgineer85, 2024-03-22:
Hey,
Right now as per implementation the files are copied one time only when the stick is plugged in.
when you plug in /out you see messages in the logs? 

here is the logic for your reference
https://github.com/photobooth-app/photobooth-app/blob/main/photobooth/services/filtetransferservice.py

hillerpa, 2024-03-23:
Thank you for the fast response. I used in the past other photobooth which was automatically syncing all X min - got it now at least how it should work but on my side it does not. 
The Log does not show anything in this direction but the USB device is detected by the OS.
Format is FAT32 - I also tried other USB devices but it did not work...

 I also setted up the whole box again now I can at least access all the files and everything else looks fine, beside the filetransfer. I only activated the USB FIle Transfer Service this should be everything in theory?
 
 I found a small other topic where a "gap" is missing on the documentation:
 
[image]

mgineer85, 2024-03-23:
If the system detects the stick well it might be that python does not.
Can you try a script running this line in python https://github.com/photobooth-app/photobooth-app/blob/main/photobooth/services/filtetransferservice.py#L80

run with stick attached and removed. There should be a difference.

hillerpa, 2024-03-23:
maybe the virus is in front of the PC .... but unfortunately I'm not that well with python: 

I created a test.py script with following code
`print('hello')
print (device for device in psutil.disk_partitions(all=False) if "removable" in device.opts)
get_current_removable_media()
@staticmethod
def get_current_removable_media():
 """Returns set of removable drives detected on the computer."""
 print (device for device in psutil.disk_partitions(all=False) if "removable" in device.opts)`

and then started it with `python test.py` 

Terminal gives back:
NameError: name 'psutil' is not defined

Feedback is also `get_current_removable_media is not defined`
Sorry I tried the last 45min to make it anyhow different working...

mgineer85, 2024-03-26:
please try following code. 

save it as test.py.
run by `python test.py`
if psutil is missing install it by `pip install psutil`
```
import psutil

print("devices filtered for removable:")
print([device for device in psutil.disk_partitions(all=False) if "removable" in device.opts])
print("devices without any filter:")
print([device for device in psutil.disk_partitions(all=False)])

```

hillerpa, 2024-03-26:
Thank you very much for your patience:

Here is the outcome without and with - so it is detected by Python (sda1):
[image]

so the issue seems somewhere else - as told the service seems not automatically starting in the Box.

- Activated the Filetransfer into photobooth folder
- Created a folder USB Device: Fotobox Folder: photobooth
- Also tried a Devicename Photbooth with Folder photobooth

hillerpa, 2024-03-26:
It is an USB-Stick  - I tried now 5 different USB sticks I had laying around also the very common.
[image]

Not one is detected as removable. 

I also changed the Format between NTFS, exFAT and FAT32 also makes no difference.
I also changed the interface from a USB3 to a USB2 on the Raspberry PI 4

mgineer85, 2024-03-27:
I see, thank you for testing. I think the underlying system has changed recently in a way that removable is not signaled any more. Maybe we just remove the filter in the future or find a different way.

Striker62, 2024-03-31:
silly question would it help to name the stick "removable"
is the format relevalt or is extfat ok

mgineer85, 2024-04-07:
So it seems the underlying system changed in Bookworm a bit and this leads to usb sticks can not detect as those.

Maybe @Peda1996 can help track down the issue and solve it as the original implementation is by Peda1996.

What is your usecase exactly?
Since there are different ways to access the files on the device, I added a short overview to the documentation. Maybe this works for you also:
https://photobooth-app.org/reference/bulktransferfiles/

hillerpa, 2024-04-07:
For me this is just the easiest way when I use the box on different locations it is not always the same wifi same for SCP.
It is much more easy also to non technical people to use a USB ^^.

mgineer85, 2024-04-07:
If the box itself is not in any WiFi the Pi could also create a hotspot from which you login and still use the filebrowser. Just another way ;)

Peda1996, 2024-04-07:
Hey thanks for the Mention, wouldn't have seen this issue otherwise. I think the issue might have to do with the if 
`if "removable" in device.opts` in the filetransferservice.py. Commenting this if statement worked somehow in my development environment.

```
    @staticmethod
    def get_current_removable_media():
        """Returns set of removable drives detected on the  computer."""
        return {
            device
            for device in psutil.disk_partitions(all=False)
            #    if "removable" in device.opts
        }
```

Tested it with my other USB sticks and somehow I only got it working on one? of my USB sticks with this line of code. (Somehow this was the one I used for my booth to transfer files) So removing this one if statement should work.

Will request a PR shortly.

mgineer85, 2024-04-08:
Patch merged and version 2 released now, close this. If it still not works, please reopen.

closed 2024-04-08 by mgineer85