photobooth-app/photobooth-app #193: Raspi Camera Modul3 not working on raspi4

dtimes, 2024-03-24:
**Describe the bug**

Raspi Camera Modul3 not working on raspberry pi 4 / same setup was working before update from quite old version of photobooth (sorry I forgot to write down the old version number before upgrading) 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to admin page, select picamera2 backend for live view or snapshot 
2. See error

```
/photobooth-app $ photobooth
2024-03-24 17:26:46,530 [   DEBUG] loggingservice __init__ basicConfig set (loggingservice.py:79)
2024-03-24 17:26:46,530 [   DEBUG] loggingservice __init__ started (loggingservice.py:80)
2024-03-24 17:26:46,531 [   DEBUG] loggingservice __init__ finished (loggingservice.py:114)
2024-03-24 17:26:46,534 [   DEBUG] registered handlers: [<StreamHandler <stderr> (NOTSET)>, <FileHand                                                                                                                                                                                                                        ler /home/photo/photobooth-app/log/photobooth_20240324.log (NOTSET)>, <EventstreamLogHandler (NOTSET)                                                                                                                                                                                                                        >] (loggingservice.py:115)
2024-03-24 17:26:46,536 [    INFO] platform.system()='Linux' (informationservice.py:34)
2024-03-24 17:26:46,537 [    INFO] platform.release()='6.6.20+rpt-rpi-v8' (informationservice.py:35)
2024-03-24 17:26:46,537 [    INFO] platform.machine()='aarch64' (informationservice.py:36)
2024-03-24 17:26:46,538 [    INFO] platform.python_version()='3.11.2' (informationservice.py:37)
2024-03-24 17:26:46,539 [    INFO] platform.node()='photopy' (informationservice.py:38)
2024-03-24 17:26:46,539 [    INFO] psutil.cpu_count()=4 (informationservice.py:39)
2024-03-24 17:26:46,543 [    INFO] psutil.cpu_count(logical=False)=4 (informationservice.py:40)
2024-03-24 17:26:46,546 [    INFO] psutil.disk_partitions()=[sdiskpart(device='/dev/mmcblk0p2', mount                                                                                                                                                                                                                        point='/', fstype='ext4', opts='rw,noatime', maxfile=255, maxpath=4096), sdiskpart(device='/dev/mmcbl                                                                                                                                                                                                                        k0p1', mountpoint='/boot/firmware', fstype='vfat', opts='rw,relatime,fmask=0022,dmask=0022,codepage=4                                                                                                                                                                                                                        37,iocharset=ascii,shortname=mixed,errors=remount-ro', maxfile=1530, maxpath=4096)] (informationservi                                                                                                                                                                                                                        ce.py:41)
2024-03-24 17:26:46,549 [    INFO] psutil.disk_usage('/')=sdiskusage(total=30794645504, used=52249886                                                                                                                                                                                                                        72, free=23984816128, percent=17.9) (informationservice.py:43)
2024-03-24 17:26:46,550 [    INFO] [('lo', ['127.0.0.1']), ('wlan0', ['192.168.188.253']), ('eth0', [                                                                                                                                                                                                                        ])] (informationservice.py:46)
2024-03-24 17:26:46,552 [    INFO] psutil.virtual_memory()=svmem(total=3976253440, available=35355361                                                                                                                                                                                                                        28, percent=11.1, used=352174080, free=2928783360, active=595476480, inactive=335163392, buffers=3090                                                                                                                                                                                                                        4320, cached=664391680, shared=16949248, slab=54939648) (informationservice.py:55)
2024-03-24 17:26:46,552 [    INFO] __debug__=True (informationservice.py:57)
2024-03-24 17:26:46,553 [    INFO] initialized information service (informationservice.py:59)
2024-03-24 17:26:46,556 [    INFO] init database and creating missing scaled images. this might take                                                                                                                                                                                                                         some time. (mediacollectionservice.py:55)
2024-03-24 17:26:46,557 [    INFO] watching for filetypes: ['.jpg', '.gif'] (mediacollectionservice.p                                                                                                                                                                                                                        y:58)
2024-03-24 17:26:46,559 [    INFO] initialized image DB, added 0 valid images (mediacollectionservice                                                                                                                                                                                                                        .py:74)
2024-03-24 17:26:46,560 [    INFO] -- process time: 0.0s to initialize mediacollection (mediacollecti                                                                                                                                                                                                                        onservice.py:75)
2024-03-24 17:26:46,576 [    INFO] on_enter_state self.current_state.id='idle'  (processingservice.py                                                                                                                                                                                                                        :104)
2024-03-24 17:26:46,577 [    INFO] state idle entered. (processingservice.py:127)
2024-03-24 17:26:46,578 [    INFO] initialized systemservice (systemservice.py:29)
2024-03-24 17:26:46,579 [    INFO] gpio not enabled - enable for gpio support on raspberry pi (gpiose                                                                                                                                                                                                                        rvice.py:60)
2024-03-24 17:26:46,899 [    INFO] photobooth directory: /home/photo/.local/lib/python3.11/site-packa                                                                                                                                                                                                                        ges/photobooth (__main__.py:43)
2024-03-24 17:26:46,900 [    INFO] working directory: /home/photo/photobooth-app (__main__.py:45)
2024-03-24 17:26:46,901 [    INFO] app version started: 1.2.1 (__main__.py:46)
2024-03-24 17:26:46,901 [    INFO] Welcome to the photobooth-app (__main__.py:49)
2024-03-24 17:26:46,905 [    INFO] started LoggingService (container.py:90)
2024-03-24 17:26:46,906 [    INFO] WledService disabled (wledservice.py:40)
2024-03-24 17:26:46,906 [    INFO] started WledService (container.py:90)
2024-03-24 17:26:49,230 [    INFO] aquisition main backend: <class 'photobooth.services.backends.pica                                                                                                                                                                                                                        mera2.Picamera2Backend'> (aquisitionservice.py:61)
2024-03-24 17:26:49,230 [    INFO] aquisition live backend: <class 'photobooth.services.backends.pica                                                                                                                                                                          [... cut ...]
**System (please complete the following information):**
```
 cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
```
photobooth app vesion: tried latest stable + dev snapshot , both not working 

**Additional context**
reinstalled OS and started from scratch, same issue. also tested on second raspi4 ,still the same issue. 
taking pictures with libcamera works normal / without any issues.

dtimes, 2024-03-24:
forgot to add, did a small test to confirm that picamera2 is working in python
script below generates picture / runs without any issues:

```
from picamera2 import Picamera2
import time
picam2 = Picamera2()
camera_config = picam2.create_still_configuration(main={"size": (320, 240)})
picam2.configure(camera_config)
picam2.start()
time.sleep(2)
picam2.capture_file("/tmp/test.png")
picam2.close()
del picam2
```

mgineer85, 2024-03-25:
The first error says the cam was busy. Maybe there is another service running in background blocking the camera access? The following errors could be follow-up errors.
Are apt packages up to date?

mgineer85, 2024-03-25:
Ah, you got picamera as main and additional live backend configured. It will access the camera two times. Disable the second one. Liveview is captured from main backend for a picamera.

dtimes, 2024-03-25:
yes, sorry my bad. I think it was set up this way before updating and it worked fine. I will change the setting and test again later

mgineer85, 2024-03-25:
With current implementation, this is hard to do but there will be improvements in the future. Glad it works again 👍

closed 2024-03-26 by mgineer85