photobooth-app/photobooth-app #203: Frame/background Preview on liveview

Striker62, 2024-03-31:
The frame and background features are great. 

To get more out of it please develop a preview on the liveview.

Idealy it would be like the filter selection in the media screen.
show either a list or  thumbnails of the frame.png or backgound.jpg.  
If one is selected show the liewview with the frame or background.
but it would only make sence to show a buton when the coresponding feature is activated in the config.

 That would make the booth more flexible and inter activ and the people can take the right possition to fit to the frame or background.

hope this is understandable.
If not please come back to me.

mgineer85, 2024-03-31:
This could be extended in version 2 at one point.
Overlay a frame id quite simple. The background is different story since the liveview then needs to process the cpu intensive chromakeying in realtime. The pi is very limited here, so I did not follow this with high priority. Could change with the pi 5 though. Needs testing and an implementation that is still easy to understand by the users.

McKaiver, 2024-06-18:
I'm looking for the same feature. I want to add the wedding couple in the center and anybody can stand next to them in the picture. but it is essential, that they see that in the preview.
I am using a Raspi5, so power should be ok.

McKaiver, 2024-06-18:
for usage, i would suggest the following:
thefiel Img Frame File is already there. you just need an extra toggle with "show in Preview - Yes/no"
It stays intuitive and wil gain a huge amount of extra features.

mgineer85, 2024-06-18:
Would you be able to help improving this feature? Would be glad to accept a pull request.

McKaiver, 2024-06-18:
Yes, i would like. I speak C++ but am interested in Python. So how can i help?

mgineer85, 2024-06-18:
So there is no definition or proposal how to implement. I could think of several different ways:

- Have the frontend load an image that is configurable in the backend and just display it statically any time on the preview screen.
- Have it merged into the live-view. The liveview is actually a series of JPG that is delivered to the frontend in this router: https://github.com/photobooth-app/photobooth-app/blob/b485f4d53547350b6841d6f878417e3c8a907c9c/photobooth/routers/api/aquisition.py#L24
  - Starting from there you could add a live-processing pipeline that merges an image to the preview. Very similar to the processing here: https://github.com/photobooth-app/photobooth-app/blob/b485f4d53547350b6841d6f878417e3c8a907c9c/photobooth/services/mediaprocessingservice.py#L135
- Another idea would be - and that is way less invasive and easier to implement maybe even without coding to use the private.css to modify your preview and add an overlay. You could also modify the config uisettings->FRONTPAGE_TEXT. It can be HTML and you can reference to your image. Maybe this helps already.

McKaiver, 2024-06-18:
ok, that sounds great!
the first would always show the maried couple, so it would be a problem with other setups
the second is also static, so for other usecases also not good.

in the end, there should be 4 Buttons for 4 differend actions.
1 normal picture
2 collage type one
3 Picture with the couple as "frame"
4 second collage

I think it would not work with these 3 options or is it possible to use different settings for different actions?

mgineer85, 2024-06-18:
Yes different actions could have different settings.  But there is no way to know in idle if the frame shall display or not. So only during countdown possible.

I think there could be one additional switch in the settings that enables a live image processing pipeline only after an action is triggered. The pipeline needs to be fast enough to finish calculations within the framerate.

McKaiver, 2024-07-01:
I have set up and installed Fotobooth-app fotobooth. I gave it 4 Buttons. 
Ther is only one single picture and a collage.
But it seems, that it is not possible, to assign 2 functions to one Button like:
in start screen: Button 1 is take a Gallery
when first picture is made: Button 1 is keep picture
when gallery finished: Button 1 is print gallery

It shows always printing not possible, so double assignment is not allowed. That would result in 8 or mor buttons, which make it complex to use. I think this has to be done first, that only the function is made, which makes sense at this moment.
Button 1 should not start a new collage, when you are asked if you want to keep the previous picture.

mgineer85, 2024-07-01:
Please let us keep the topic here to the title.
There is another issue #250 for this.

Currently touchscreens are first class citizens, GPIO is ... available...
Please use the other issue and comment how you would propose to use the 4 buttons. I think there are many options and I would like to implement a simple one. But as I personally do not use the buttons, it would be good if an actual user describes it's operation before implement anything.

McKaiver, 2024-07-01:
Yes, that's right I just wanted to say, why i did not proceed here.(like lots of people do, ask for something and then dissapear)

So this is so interesting, that i would like to help, with everything i can do with my skill. So for now i will try on my own and come back, if i have questions.

Striker62, 2024-09-11:
HI,

I try to add a frame  to the liveview via css.
I found this . Do you think this can work ?
Not sure if the path is visible for the websever for this purpose .

.preview-stream::after {
    content: "";
    position: absolute;
    top: 0;
    left: 0;
    right: 0;
    bottom: 0;
    background-image: url('/media/processed/Photobox_Klein_2.png'); /* Dein PNG als Overlay */
    background-size: cover; /* Deckt den gesamten Bereich ab */
    background-position: center;
    pointer-events: none; /* Erlaubt weiterhin die Interaktion mit dem Stream */
    z-index: 1; /* Stellt sicher, dass das Overlay über dem Stream liegt */
    opacity: 0.9; /* Optional: Transparenz für das Overlay */

Here is my frame
[image]

Striker62, 2024-09-15:
with the css solution for the frame on the main screen I would say request can be closed.
what do you think ?

McKaiver, 2024-09-15:
The Problem here is, that it will be there for all pictures, right? 
So if you put the couple on the screen, they will be there always, so they will be double, if the Real couple is there...
I would prefer the second solution with the parsing, but i habe no idea of Python...

Striker62, 2024-09-15:
I have "stupid?" Idea.

Frist, sorry for bad englisch 😀

- If i'm Right you can define a Share Action Like printing. But instead of puting a Print Command, you can rename the Frame file to another Name or an Not existent file. Assigne a Button via gpio or on the screen . The Same the otherway round if you wont to get the Frame back. I did something Similar to change the Frame.

If you have it a try come back with the result 🥹

McKaiver, 2024-09-30:
Ok, I tried to add this:
`.preview-stream::after {
content: "";
position: absolute;
top: 0;
left: 0;
right: 0;
bottom: 0;
background-image: url('/media/processed/Photobox_Klein_2.png'); /* Dein PNG als Overlay /
background-size: cover; / Deckt den gesamten Bereich ab /
background-position: center;
pointer-events: none; / Erlaubt weiterhin die Interaktion mit dem Stream /
z-index: 1; / Stellt sicher, dass das Overlay über dem Stream liegt /
opacity: 0.9; / Optional: Transparenz für das Overlay */`

here:
`"FRONTPAGE_TEXT": "<div class=\"fixed-center text-h2 text-weight-bold text-center text-white\" style=\"text-shadow: 4px 4px 4px #666;\">Hey!<br>Let's take some pictures! <br>📷💕</div>",`

so now it looks like this:
`    "FRONTPAGE_TEXT": .preview-stream::after {
content: "";
position: absolute;
top: 0;
left: 0;
right: 0;
bottom: 0;
background-image: url('/media/processed/Photobox_Klein_2.png'); /* Dein PNG als Overlay /
background-size: cover; / Deckt den gesamten Bereich ab /
background-position: center;
pointer-events: none; / Erlaubt weiterhin die Interaktion mit dem Stream /
z-index: 1; / Stellt sicher, dass das Overlay über dem Stream liegt /
opacity: 0.9; / Optional: Transparenz für das Overlay */,`

that had no effect...

Striker62, 2024-09-30:
hi, I'm  not sure but in another disussion @mgineer85  explained that only the Userdata folder can be seen from the webserver. please try the same setup as i wrote.

Striker62, 2024-09-30:
You can find my sample here Frame selection Howto #356

McKaiver, 2025-01-12:
i made now a few tests, unfortunately it is quite complex to change it.
so if you are using different frames, switching between them always needs a refresh and the button action.

If i can help with anything regarding the JPEG Stream, let me know, because that would be the smoothest feature for on site usage. 

the only problem is, that i can only speak C++

mgineer85, 2025-01-12:
I think the easiest and most compute efficient option would be to implement overlaying the preview stream in html/css/js.
Maybe I can add it to v5, not sure yet.

McKaiver, 2025-01-19:
> I think the easiest and most compute efficient option would be to implement overlaying the preview stream in html/css/js. Maybe I can add it to v5, not sure yet.

that would be perfect, the additional possibilities would be enormous.

The first real use of this box is the wedding of my sister in April, so there is time left :-)

mgineer85, 2025-01-19:
So implemented in v5 now.   The red circled text/logo is a transparent png that is overlayd the livestream. 
Since I use the mirroreffect, the text is mirrored also. This is intended. 
The final image will look correct.

[image]

Also added some nice glass-effect to the buttons so they do not stick out that much but are still clearly visible.
Ah, and also added the preview stream to the background a second time but blurred. So the screen is always nicely covered.

closed 2025-01-19 by mgineer85

mgineer85, 2025-01-24:
It is available as release candidate already, so if you choose the v5 explicitly, you could get it.
But I will release it hopefully this weekend anyways, so it would be better to wait.

I am fixing some minor bugs right now.

mgineer85, 2025-01-24:
v5 is released just now: https://github.com/photobooth-app/photobooth-app/releases/tag/v5.0.0

McKaiver, 2025-01-24:
Made directly the Update, there is one thing, i found out: for the framepath, you have to add userdata/ in the beginning of the path, otherwise, it does not work. (could cause problems, because an error appears.

For this functionality i was hoping, that the frame changes in liveview to the frame, which is used for the action.
for example an action with this frame:

[image]

The users need to know, where to put the head, that the frame is exactly over theirs. so during the countdown, it has to be this frame.

but for other actions, it shoould not be there, so in standart live stream again the normal Frame, which could be this:

[image]

mgineer85, 2025-01-24:
This function is not (yet) implemented, but with the recent improvements behind the scenes it could be easy to add. No timeline yet.

mgineer85, 2025-01-24:
I opened a new issue about that.

mgineer85, 2025-01-26:
There is a first implementation now: https://github.com/photobooth-app/photobooth-app/issues/405

commit 56b14d0e (2025-01-19): bump frontend, fixes #203