photobooth-app/photobooth-app #204: add frame and background selection to the media section

Striker62, 2024-03-31:
in the media section you have provide the function to add filter to the pictures.

would it be possible to do the same with frames and backgrounds ?

but only if the corresponding function is active in the config.

mgineer85, 2024-03-31:
I consider the frames to be more static. with regards to lower power systems like the pi I try to avoid unnecessary computationally costly operations in postprocessing. So the user shall get the result as fast as possible.

 One could precalculate the result one time but to me I think it is out of scope. Still if there is a ready to merge pull request including cache I would accept it.

Striker62, 2024-03-31:
yes you're right the background is too much to render on the liveview.
the more interesing part is the frame

to be a little more flexible i'll rty to setup seperate config files per theme and write a small script to repalce the actual and restart the service.

Striker62, 2024-03-31:
sorry but this request was not for the live view screen it is for the media screen.
but if you have the same concerns also on the media screen it is ok for me.

mgineer85, 2024-03-31:
Ah, I see. You mean in the gallery when you open and image? That could be possible to add later in v2 probably quite easy. 
but there is no timeline for this unfortunately.

mgineer85, 2024-07-14:
I think with the recent implementation to choose between different actions when a capture is started, this functionality is covered to like 90%.
I am gonna close this.
Maybe this topic can be revisited later again when there will be more customizations than "just" the filter.

closed 2024-07-14 by mgineer85