photobooth-app/photobooth-app #207: Gphoto2 ISO-Settings for stills not working / set correctly

PixelAlarm, 2024-04-04:
**Describe the bug**
In the backends-settings (gphoto2), ISO values can be set separately for still images and live view.
The value for still images isn‘t working (maybe not given correctly to gphoto2). (Canon 700D)

I set live ISO to 6400 and still ISO to 800.
Final picture has 6400 in Metadata.

**To Reproduce**
Steps to reproduce the behavior:
1. Set different ISOs in settings „backends“
2. Take Picture
3. Check Metadata

**Expected behavior**
ISO values for still images are forced in Camera and are actually used.

**System (please complete the following information):**
 - OS: Xubuntu 22.04
 - Futro S940
 - Firefox
 - Python 3.10.x

mgineer85, 2024-04-04:
Can you confirm that another setting instead 6400 for live would work. To ensure it’s not a default in the current cam settings.
Can you provide the logs?

PixelAlarm, 2024-04-04:
I tried for the last hour, but I think my skillset is not enough.

What would I have to move?
def _on_configure_optimized_for_hq_capture
def _iso
def _worker_fun

mgineer85, 2024-04-05:
Jah, could be a bit complicated like that. Modify following for now like this:
```
def _on_configure_optimized_for_hq_capture(self):
        pass
        #self._iso(appconfig.backends.gphoto2_iso_capture)
        #self._shutter_speed(appconfig.backends.gphoto2_shutter_speed_capture)
```
And then you can place just the iso command somewhere you think makes sense:
```
self._iso(appconfig.backends.gphoto2_iso_capture)
```
I think at the red position would be best probably: 
[image]

PixelAlarm, 2024-04-05:
Works now with your mentioned position. I moved both settings for iso and shutter there. Thank you!
(I did the right thing myself, but in the wrong path :D)

Would now also be really nice to trigger the autofocus to off for live view (to disable "focus breathing") and on for hq capture. (Focus point of last picture would be present in live view)
I tried to add another setting for set_config with "autofocusdrive"="0"/"1", but it wouldn't work.

mgineer85, 2024-04-05:
Please check what’s in the logs.
Maybe the camera doesn’t support it?

mgineer85, 2024-04-05:
Latest main will have the config update procedure for the camera at the right place. If you find out how to use autodrive on your camera, we can add this also later.
https://github.com/photobooth-app/photobooth-app/commit/f753535f08a8f29b016cbb1c783c7923b9c5b24e

mgineer85, 2024-04-06:
Since the original problem will be solved in latest main, I close this issue.
If you find a way to configure autofocus for your camera I am happy to integrate this. Just put the information in a new issue.
Thank you!

closed 2024-04-06 by mgineer85