photobooth-app/photobooth-app #208: Add delay after countdown to trigger capture (and Display Smile-Screen longer)

PixelAlarm, 2024-04-04:
**Is your feature request related to a problem? Please describe.**
My backend (gphoto2 / Canon 700D) is so fast, that the Countdown won‘t run till the end and the „smile-screen“ is often skipped completely.
Countdown will directly go to 9-dot-animation.

**Describe the solution you'd like**
It would be nice to be able to set a delay for the Camera to be triggered.
In the delay-period the smile-screen will be shown.

Maybe additional customizations for smile screen would be nice (e. g. disable liveview, so that the attention goes to the Camera and Not the screen.)

**Describe alternatives you've considered**
Using slower Camera *lol*

**Additional context**
I could add a .gif later

mgineer85, 2024-04-04:
What about raising the timing in uisettings for the message time?

PixelAlarm, 2024-04-04:
So sorry, I missed that implementation, since it's in another place. My bad.

I still think the current implementations isn't very intuitive.

The "Takepic Msg Time" steals the time from the countdown.
In my opinion it would be better, if the Msg Text would be shown after the countdown is finished and can then be modified as desired. But that's just my taste.

**Describe the solution you'd like**
Maybe a compromise would be to have two settings. This way it would be completely configurable:
1. Time to display Msg Text **before** countdown ends
2. Time to display Msg Test **after** countdown ends

Thank you!

mgineer85, 2024-04-04:
It's a bit more complicated and the implementation needs to cover also slower backends. It's built with that in mind.
You can offset the capture when the camera is triggered before count to 0. That is to improve experience with slower backends.
You can also offset the takepic msg time, which steals the time from the overall counter, but you could extend by that amount.
Now the message is hidden in the moment the backend signals that the capture is done and the 9 processing dots appear.
I see that for fast backends the flicker of takepicture message could be confusing.
I am still hesistant to add another option for this and maybe find a better simpler solution.

PixelAlarm, 2024-04-04:
I get your point but also see the advantage of my solution :)

In my opinion it is nicer if the countdown runs to 0 and I see the smile page for like 2-3 seconds and in the worst case the picture is already taken, when the smile is still shown. People who don't want the "extend after the countdown" just could leave it at 0.

Another option would be to configure an offset between real counter and displayed counter. That way the displayed counter could already be at 0 and show the Msg Text but in the background the real counter has time remaining. But I think this is more complex.

In the mean time I think I have to live with the counter to switch to the Msg Text bis X seconds not been displayed counting down.

mgineer85, 2024-04-05:
Actually the offset is also to take into account that the trigger to the camera is offsetted in case the delay of the camera is quite long.
Just extending the smile message is not working. I would not like to have an additional delay just to display the smile longer even if all processing is done. It shall be fast :)
I consider how it can be improved without additional settings.

PixelAlarm, 2024-04-05:
Yes, since speed is a key feature of the software, I totally get your point.

Maybe a solution would be to allow negative values for Capture Offset.
This way the processing can continue right when capture is done...

mgineer85, 2024-04-08:
I'd consider the virtualcamera as fast backend. I used it to improve the countdown. The improved version is in the main branch of this repo. Changes made here: https://github.com/photobooth-app/photobooth-frontend/commit/32a884797af9a67cf307ba023e90ec3bbebe3ee6

Following values give me now way better UX in the latest main:
Countdown Capture First = 3
Countdown Camera Capture Offset = 0,2
Takepic Msg Time = 1

The fix lowers the animation-speed to 100. The default was 600 and led to the circle still showing some remaining time despite countdown was finished already.
Also the takepicture message is now displayed on top of the circle, the number is hidden if the takepicture message is shown.

I think this should be fine for your setup also?

mgineer85, 2024-04-08:
https://github.com/photobooth-app/photobooth-app/assets/3868450/d1424f11-d270-4f88-8169-ea480f4c9588

mgineer85, 2024-04-08:
Version 2 with the improved timing is released just now.

closed 2024-04-08 by mgineer85