photobooth-app/photobooth-app #210: Integrated Photobooth-Project, Photobooth-App and WLED

AlexXLVI, 2024-04-04:
**Is your feature request related to a problem? Please describe.**
I integrated Photobooth-Project and this Photobooth-App. ([https://pypi.org/project/photobooth-app/0.1.0b4/](url))
So my Pi-Camera gives a Preview and the capture too.
Then I get an LED-ring running ([https://photobooth-app.org/extras/wled/](url)) with the Photobooth-App.
How can I combine this WLED with the Photobooth-Project? 

**Describe the solution you'd like**
This [Video](https://photobooth-app.org/extras/wled/) shows the result, I am looking for.

**Describe alternatives you've considered**

**Additional context**

mgineer85, 2024-04-05:
Hey, the integration is not actively supported any more, the information about integration are deleted. Following is the last document: https://github.com/photobooth-app/photobooth-docs/blob/d7379860365c5607f574ee3546e3c2e68f8d0dc8/docs/reference/photoboothprojectintegration.md

Depending on the version you use, it still should work as is.

closed 2024-04-05 by mgineer85

AlexXLVI, 2024-04-09:
Thank you for your quick reply. According to these instructions, I was able to integrate the live stream of the Photobooth app into the Photobooth.Project.
I was also able to integrate an LED ring into the photobooth app via WLED.
In the video, this LED ring also works with the Photobooth.Project.

How can I configure this too?

mgineer85, 2024-04-10:
Earlier there was a dedicated api endpoint to ensure this behaviour. This is removed also.
But you can use the WLED api directly to load presets:

Example to load preset number 2:
```
curl -X POST "http://192.168.10.81/json" -H "Content-Type: application/json" -d "{'ps':2}"
```

Replace IP by actual IP.

AlexXLVI, 2024-04-11:
Thanks, but unfortunately I don't know where to place this code?
Where in the admin panel or where in my.config.inc.php does this code belong?

Can different WLED presets also be called up?
E.g. countdown - captures or  single photo - collage

mgineer85, 2024-04-11:
You would add this as a command in the boothproject.