photobooth-app/photobooth-app #211: Pi Camera V3 not taking pictues (but Live preview in homepage works)

niccolov, 2024-04-05:
I've finally installed all the system and tested it.
Very great job! Congratulation!

I've a little problem with my new Pi Camera Module V3.

**System:**
 - OS: Raspberry Pi bullseye 64bit
 - Hardware: Raspberry Pi 3 B+
 - Camera: Pi Camera V3
- I've made a clean full installation and followed the instruction

Using Photobooth-app I see the live preview on the home page but if I try to take a picture the camera stops working.
Here the copy of the error:

2024-04-05 19:10:00,664 [ WARNING] WLED module not connected, ignoring request (wledservice.py:148)
2024-04-05 19:10:00,668 [   DEBUG] change to capture mode requested (picamera2.py:240)
2024-04-05 19:10:00,671 [    INFO] start countdown, duration_user=duration=2.0, offset_camera=0.25 (processingservice.py:161)
2024-04-05 19:10:00,731 [    INFO] switch_mode invoked, stopping stream encoder, switch mode and restart encoder (picamera2.py:265)
2024-04-05 19:10:00,842 [    INFO] Camera stopped (picamera2.py:1075)
2024-04-05 19:10:01,319 [    INFO] Configuration successful! (picamera2.py:945)
2024-04-05 19:10:01,629 [   ERROR] Uncaught exception in thread <StoppableThread(_worker_thread, started daemon 547520074176)>: <class 'RuntimeError'> Failed to allocate buffers: Impossibile allocare memoria (loggingservice.py:190)
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/fotobutte/.local/lib/python3.9/site-packages/photobooth/services/backends/picamera2.py", line 311, in _worker_fun
    self._switch_mode()
  File "/home/fotobutte/.local/lib/python3.9/site-packages/photobooth/services/backends/picamera2.py", line 273, in _switch_mode
    self._picamera2.switch_mode(self._current_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1241, in switch_mode
    return self.dispatch_functions(functions, wait, signal_function, immediate=True)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1203, in dispatch_functions
    return job.get_result() if wait else job
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 77, in get_result
    return self._future.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 440, in result
    return self.__get_result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 47, in execute
    done, result = self._functions[0]()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1234, in switch_mode_
    self.configure_(camera_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 983, in configure_
    if self.allocator.allocate(stream) < 0:
RuntimeError: Failed to allocate buffers: Impossibile allocare memoria
2024-04-05 19:10:02,950 [    INFO] on_exit_state self.current_state.id='counting'  (processingservice.py:112)
2024-04-05 19:10:02,959 [    INFO] on_enter_state self.current_state.id='capture'  (processingservice.py:104)
2024-04-05 19:10:02,971 [    INFO] current capture (1/1, remaining 0) (processingservice.py:177)
2024-04-05 19:10:03,004 [   DEBUG] capture to filepath_neworiginalfile=PosixPath('media/original/image_show_20240405-191002-985841.jpg') (processingservice.py:191)
2024-04-05 19:10:03,010 [    INFO] WledService preset_shoot triggered (wledservice.py:141)
2024-04-05 19:10:03,016 [ WARNING] WLED module not connected, ignoring request (wledservice.py:148)
2024-04-05 19:10:07,018 [   ERROR] timeout receiving frames (abstractbackend.py:253)
Traceback (most recent call last):
  File "/home/fotobutte/.local/lib/python3.9/site-packages/photobooth/services/backends/abstractbackend.py", line 251, in wait_for_hq_image
    return self._wait_for_hq_image()
  File "/home/fotobutte/.local/lib/python3.9/site-packages/photobooth/services/backends/picamera2.py", line 205, in _wait_for_hq_image
    raise TimeoutError("timeout receiving frames")
TimeoutError: timeout receiving frames
2024-04-05 19:10:07,028 [   ERROR] error capture image. attempt=1/3, retrying (abstractbackend.py:254)
2024-04-05 19:10:11,030 [   ERROR] timeout receiving frames (abstractbackend.py:253)
Traceback (most recent call last):
  File "/home/fotobutte/.local/lib/python3.9/site-packages/photobooth/services/backends/abstractbackend.py", line 251, in wait_for_hq_image
    return self._wait_for_hq_image()
  File "/home/fotobutte/.local/lib/python3.9/site-packages/photobooth/services/backends/picamera2.py", line 205, in _wait_for_hq_image
    raise TimeoutError("timeout receiving frames")
TimeoutError: timeout receiving frames
2024-04-05 19:10:11,034 [   ERROR] error capture image. attempt=2/3, retrying (abstractbackend.py:254)
2024-04-05 19:10:15,037 [   ERROR] timeout receiving frames (abstractbackend.py:253)
[... 117 lines cut ...]
[0:08:59.831734656] [2333]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:08:59.834838166] [2317]  INFO Camera camera.cpp:840 Pipeline handler in use by another process
ERROR: *** failed to acquire camera /base/soc/i2c0mux/i2c@1/imx708@1a ***

Same error If I tri something like:
libcamera-jpeg -o prova3.jpg

If I stop the photobooth-app libcamera works again.

$ libcamera-jpeg -o prova3.jpg
Made X/EGL preview window
[0:34:58.331291190] [2775]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:34:58.559265288] [2776]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media2 and ISP device /dev/media0
[0:34:58.559456428] [2776]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:34:58.564874022] [2775]  INFO Camera camera.cpp:1033 configuring streams: (0) 2048x1152-YUV420
[0:34:58.566279344] [2776]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
[0:35:08.835362019] [2775]  INFO Camera camera.cpp:1033 configuring streams: (0) 4608x2592-YUV420 (1) 4608x2592-SBGGR10_CSI2P
[0:35:08.841003099] [2776]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
Still capture image received

Can you help me to complete the process.... making the pictures?

Thanks,

Niccolò

mgineer85, 2024-04-05:
Hey,
for what reason you chose the old raspberry system?
The error is from picamera2/libcamera which is actually only limited updated on the old system.
Please upgrade to bookworm.
Ensure you updated the system also to the latest packages with apt.

mgineer85, 2024-04-05:
It cannot be changed in the photobooth, but I changed to catch the error and automatically restart the app instead to crash.
But this just covers the error only and does not fix it 
https://github.com/photobooth-app/photobooth-app/commit/6ae2cbe43d06ce7940aaac60da0887df1b2d1d59

niccolov, 2024-04-07:
You're right! I've made a fresh installation with Bookworm, this time on a Raspberry 4, and the Pi Camera 3 works great! 
Thanks!

mgineer85, 2024-04-07:
Great, thank you for providing the feedback.
So I close this issue for now.

closed 2024-04-07 by mgineer85