photobooth-app/photobooth-app #212: Shareservice error

Kristian92dk, 2024-04-06:
Hi I am getting this error

2024-04-05 20:58:44,076 [    INFO] stream quality Quality[appconfig.backends.picamera2_stream_quality.name]=<Quality.MEDIUM: 2> (picamera2.py:152)
2024-04-05 20:58:44,216 [    INFO] Camera started (picamera2.py:1134)
2024-04-05 20:58:44,472 [   ERROR] api endpoint error: Expecting value: line 1 column 1 (char 0) (shareservice.py:42)
2024-04-05 20:58:44,473 [    INFO] api endpoint found, URL valid (shareservice.py:45)
2024-04-05 20:58:44,473 [    INFO] starting shareservice worker_thread (shareservice.py:72)
2024-04-05 20:58:44,581 [   DEBUG] autofocus set (picamera2.py:295)
2024-04-05 20:58:44,581 [   DEBUG] photobooth.services.backends.picamera2 started (picamera2.py:168)
2024-04-05 20:58:44,581 [    INFO] setting device status from starting to running (abstractbackend.py:106)
2024-04-05 20:58:44,581 [    INFO] device started, set status running (abstractbackend.py:155)
2024-04-05 20:58:44,930 [    INFO] successfully connected to shareservice dl.php script (shareservice.py:97)
2024-04-05 20:58:44,933 [   ERROR] Uncaught exception in thread <StoppableThread(_shareservice_worker, started daemon 140732367679872)>: <class 'json.decoder.JSONDecodeError'> Expecting value: line 1 column 1 (char 0) (loggingservice.py:190)
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/photo/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/services/shareservice.py", line 116, in _worker_fun
    decoded_line = json.loads(line)
                   ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
2024-04-05 20:58:45,473 [   DEBUG] photobooth.services.shareservice initialized and started (shareservice.py:61)
2024-04-05 20:58:45,473 [    INFO] started ShareService (container.py:90)
2024-04-05 20:58:45,474 [    INFO] FileTransferService disabled, start aborted. (filtetransferservice.py:30)
2024-04-05 20:58:45,474 [    INFO] started FileTransferService (container.py:90)
2024-04-05 20:58:45,474 [    INFO] started container (container.py:95)
2024-04-05 20:58:45,479 [   DEBUG] Using selector: EpollSelector (selector_events.py:54)
2024-04-05 20:58:45,536 [    INFO] Started server process [1984] (server.py:82)
2024-04-05 20:58:45,536 [    INFO] Waiting for application startup. (on.py:48)
2024-04-05 20:58:45,537 [    INFO] Application startup complete. (on.py:62)
2024-04-05 20:58:45,537 [    INFO] Uvicorn running on http://0.0.0.0:8000 (Press CTRL+C to quit) (server.py:214)
2024-04-05 20:58:52,570 [    INFO] 127.0.0.1:40966 - "GET / HTTP/1.1" 200 (h11_impl.py:477)
2024-04-05 20:58:53,435 [    INFO] 127.0.0.1:40966 - "GET /config/ui HTTP/1.1" 200 (h11_impl.py:477)
2024-04-05 20:58:53,451 [   DEBUG] -- process time: 0.0s to compile db_get_images_as_dict output (mediacollectionservice.py:116)
2024-04-05 20:58:53,458 [    INFO] 127.0.0.1:40982 - "GET /mediacollection/getitems HTTP/1.1" 200 (h11_impl.py:477)
2024-04-05 20:58:53,842 [    INFO] SSE subscription added for client Address(host='127.0.0.1', port=40982) (sseservice.py:217)
2024-04-05 20:58:53,842 [   DEBUG] SSE clients listed [<starlette.requests.Request object at 0x7ffee0195ed0>] (sseservice.py:218)
2024-04-05 20:58:53,850 [    INFO] 127.0.0.1:40982 - "GET /sse HTTP/1.1" 200 (h11_impl.py:477)
2024-04-05 20:58:53,937 [    INFO] livestream requested from main backend (aquisitionservice.py:116)
2024-04-05 20:58:53,945 [    INFO] 127.0.0.1:40966 - "GET /aquisition/stream.mjpg HTTP/1.1" 200 (h11_impl.py:477)
2024-04-05 20:58:53,952 [    INFO] livestream started on backend backend_to_stream_from=<class 'photobooth.services.backends.picamera2.Picamera2Backend'> (aquisitionservice.py:169)

System:

OS: Raspberry Pi bookworm 64bit (Release: March 15th 2024)
Hardware: Raspberry Pi 5
Camera: Pi Camera V3

Thanks,

Kristian

mgineer85, 2024-04-06:
Hey, did you upload the dl.php to a Hoster and pointed to that uri in the admin dashboard configuration?
If so, what you see in the browser when you access that uri?
Configuration , please upload also.

mgineer85, 2024-04-08:
Hey @Kristian92dk , did you sort out the issue?

commit 8f87f997 (2024-04-06): improvement: catch error if dl.php webservice is not setup properly instead throwing exceptions. fixes #212
closed 2024-04-19 by mgineer85