photobooth-app/photobooth-app #216: Version 2.0a4 Filesection

Striker62, 2024-04-08:
### Discussed in https://github.com/photobooth-app/photobooth-app/discussions/215

<div type='discussions-op-text'>

<sup>Originally posted by **Striker62** April  8, 2024</sup>
I'm notable to select any folder or file in the file section.
I realiesed this when i try to delete logfiles to start a new one.

I the a new switch in the config that protect file from deletion ?</div>

Striker62, 2024-04-08:
Is there a new switch in the config that protect file from deletion ?

mgineer85, 2024-04-08:
there is nothing that prevents you from deleting any file.
We could create a backup if the config is saved to have it avail for further references instead.
What you mean by not able to select? any screenshots?

Striker62, 2024-04-08:
if i try to klich the checkbox in fornt of the file or folder nothing is selected so the deletion buton is not working

mgineer85, 2024-04-08:
thank you for reporting. fixed in latest main by https://github.com/photobooth-app/photobooth-frontend/commit/4013937f5873fb1ed8c4bed2f1858ecb920f6802