photobooth-app/photobooth-app #220: Fullscreen-Mode / Progressive Web App (PWA)

PixelAlarm, 2024-04-08:
**Is your feature request related to a problem? Please describe.**
When exporting photobooth-app to home screen and open in Kiosk Mode (Testet iPadOS 17), the address bar is still visible.
So the Kiosk-Mode is pretty useless, since everyone could put in another link.

**Describe the solution you'd like**
Make app available as PWA, so the address bar is not visible.
Kiosk-Mode will then allow the photobooth-app to be the only usable tablet-function.

**Describe alternatives you've considered**
I tried fullscreen browser, but the url can be changes easily by the user.

**Additional context**
Maybe this is easy to integrate, since vite5 is used: https://github.com/vite-pwa/vite-plugin-pwa
Alternative: https://gist.github.com/irae/1042167

For old Apple Devices, adding this to the index.html would help (will try myself)
<meta name="apple-mobile-web-app-capable" content="yes" />
<meta name="apple-mobile-web-app-status-bar-style" content="black" />

mgineer85, 2024-04-09:
Quasar has all it needs to build as PWA: https://quasar.dev/quasar-cli-vite/developing-pwa/introduction

But if it is not neccessary and just some html tags control the behaviour, that would be better, since there is no use of all the other PWA features.

PixelAlarm, 2024-04-09:
Adding the meta tags as stated in the link worked:
https://developer.apple.com/library/archive/documentation/AppleApplications/Reference/SafariWebContent/ConfiguringWebApplications/ConfiguringWebApplications.html#//apple_ref/doc/uid/TP40002051-CH3-SW1

I did a pull request:
https://github.com/photobooth-app/photobooth-app/pull/223

This issue can be closed.

mgineer85, 2024-04-10:
Feature implemented in https://github.com/photobooth-app/photobooth-frontend/commit/4853df864de4b955b80edf9304088372aaf50e21

Included in next release; release just now.

closed 2024-04-10 by mgineer85