photobooth-app/photobooth-app #233: Video function broken/unstable/no preview with gphoto2

PixelAlarm, 2024-04-15:
**Describe the bug**
Since a recent update (tested 2.0.2) the Video function became very unstable for me using gphoto2 with liveview

1. sometimes it throws error. Log:
FileNotFoundError: ‚None‘ video not found

2. Sometimes Video works but cannot be displayed on result screen (Media corrupt) and Gallery (just blank item)

(3. Never works as intended for me.)

**To Reproduce**
Steps to reproduce the behavior:
1. Use gphoto2
2. Enable Live View
3. Take Video

**Expected behavior**
Working Function

If I can help testing something, please contact.
Really Looking forward to this awesome Feature and the implementation of boomerang in the Future :)

**System (please complete the following information):**
 - OS: Xubuntu 22.04
 - Firefox Linux and Safari iOS

**Additional context**
If I can help testing something, please contact.
Really Looking forward to this awesome Feature and the implementation of boomerang in the Future :)

mgineer85, 2024-04-15:
Hey,
could you upload a logfile from a restring, please?
You mean only the video does not work? Live preview is good all the times or has issues with your camera?

you installed ffmpeg?

PixelAlarm, 2024-04-15:
Hey,

the live preview works.

To be more precise for "2.":
What is not working is the result-screen after video recording, so media playback on result-screen is broken.
And in the gallery there are only white items, when they are clicked, there is broken media also.

ffmpeg is installed. The video files are indeed stored in the media folders and are playable with vlc.

I will post the log, will have to change computers.

mgineer85, 2024-04-15:
Any errors in the browser console?
Would chrome work(that is what I use to develop)?

PixelAlarm, 2024-04-15:
This is log entry from working video with broken media in result / gallery:

2024-04-15 13:19:41,359 [    INFO] _videoworker_fun start (abstractbackend.py:388)
2024-04-15 13:19:41,360 [    INFO] writing to ffmpeg stdin (abstractbackend.py:429)
2024-04-15 13:19:41,361 [   DEBUG] -- ffmpeg startuptime: 0.0s  (abstractbackend.py:362)
ffmpeg started on 2024-04-15 at 13:19:41
Report written to "./log/ffmpeg-last.log"
Log level: 32
ffmpeg version 4.4.2-0ubuntu0.22.04.1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.2.0-19ubuntu1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.22.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-librsvg --enable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
2024-04-15 13:19:44,361 [    INFO] no user input within 3s so stopping video (processingservice.py:425)
2024-04-15 13:19:44,362 [    INFO] transition after: counting--(_counted)-->record (processingservice.py:226)
2024-04-15 13:19:44,363 [    INFO] exit: record from stop_recording (processingservice.py:237)
2024-04-15 13:19:44,363 [    INFO] WledService preset_standby triggered (wledservice.py:132)
2024-04-15 13:19:44,363 [   DEBUG] stop recording (abstractbackend.py:372)
2024-04-15 13:19:44,431 [    INFO] writing to ffmpeg stdin finished (abstractbackend.py:448)
2024-04-15 13:19:44,431 [   DEBUG] -- process time: 3.07s  (abstractbackend.py:449)
Input #0, image2pipe, from 'pipe:':
  Duration: N/A, start: 1713179981.480000, bitrate: N/A
  Stream #0:0: Video: mjpeg (Baseline), yuvj422p(pc, bt470bg/unknown/unknown), 960x640, 25 fps, 25 tbr, 25 tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (libx264))
[libx264 @ 0x5e72e1f59000] using cpu capabilities: MMX2 SSE2Fast SSSE3 SSE4.2
[libx264 @ 0x5e72e1f59000] profile High 4:2:2, level 3.1, 4:2:2, 8-bit
[libx264 @ 0x5e72e1f59000] 264 - core 163 r3060 5db6aa6 - H.264/MPEG-4 AVC codec - Copyleft 2003-2021 - http://www.videolan.org/x264.html - options: cabac=1 ref=1 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=2 psy=1 psy_rd=1.00:0.00 mixed_ref=0 me_range=16 chroma_me=1 trellis=0 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=0 threads=6 lookahead_threads=2 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=1 keyint=250 keyint_min=6 scenecut=40 intra_refresh=0 rc_lookahead=10 rc=abr mbtree=1 bitrate=3000 ratetol=1.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to 'tmp/Gphoto2Backend_9e8ad11ff0034e75aa972f077ae1938f.mp4':
  Metadata:
    encoder         : Lavf58.76.100
  Stream #0:0: Video: h264 (avc1 / 0x31637661), yuvj422p(pc, bt470bg/unknown/unknown, progressive), 960x640, q=2-31, 3000 kb/s, 6 fps, 12288 tbn
    Metadata:
      encoder         : Lavc58.134.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/3000000 buffer size: 0 vbv_delay: N/A
[mp4 @ 0x5e72e1f5fb80] Starting second pass: moving the moov atom to the beginning of the file
frame=   20 fps=0.0 q=-1.0 Lsize=    2444kB time=00:00:02.83 bitrate=7065.5kbits/s dup=0 drop=13 speed=4.66x    
video:2443kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.044976%
[libx264 @ 0x5e72e1f59000] frame I:1     Avg QP: 0.33  size:276851
[libx264 @ 0x5e72e1f59000] frame P:5     Avg QP:10.14  size:113050
[libx264 @ 0x5e72e1f59000] frame B:14    Avg QP: 8.61  size:118466
[libx264 @ 0x5e72e1f59000] consecutive B-frames:  5.0%  0.0% 15.0% 80.0%
[libx264 @ 0x5e72e1f59000] mb I  I16..4:  8.7% 59.6% 31.7%
[libx264 @ 0x5e72e1f59000] mb P  I16..4: 12.8% 57.0%  2.9%  P16..4: 17.3%  3.8%  2.6%  0.0%  0.0%    skip: 3.7%
[libx264 @ 0x5e72e1f59000] mb B  I16..4:  8.0% 40.4%  0.7%  B16..8:  9.6%  5.4%  1.1%  direct:25.8%  skip: 9.0%  L0:26.9% L1:45.0% BI:28.1%
[libx264 @ 0x5e72e1f59000] final ratefactor: 15.05
[libx264 @ 0x5e72e1f59000] 8x8 transform intra:79.1% inter:63.5%
[libx264 @ 0x5e72e1f59000] coded y,uvDC,uvAC intra: 89.5% 96.4% 91.2% inter: 58.7% 72.4% 55.9%
[... cut ...]
2024-04-15 13:19:45,089 [    INFO] enter: finished from _finish (processingservice.py:230)
2024-04-15 13:19:45,089 [    INFO] transition after: present_capture--(_finish)-->finished (processingservice.py:226)
2024-04-15 13:19:45,089 [   DEBUG] job finished (processingservice.py:64)
2024-04-15 13:19:45,090 [   DEBUG] _process_fun left (processingservice.py:84)
2024-04-15 13:19:45,280 [    INFO] 127.0.0.1:48258 - "GET /media/processed/preview/video_show_20240415-131945-055246.mp4 HTTP/1.1" 200 (h11_impl.py:477)

PixelAlarm, 2024-04-15:
This is for Error during video recording without result:

2024-04-15 13:44:52,816 [    INFO] _videoworker_fun start (abstractbackend.py:388)
2024-04-15 13:44:52,822 [    INFO] writing to ffmpeg stdin (abstractbackend.py:429)
2024-04-15 13:44:52,822 [   DEBUG] -- ffmpeg startuptime: 0.01s  (abstractbackend.py:362)
ffmpeg started on 2024-04-15 at 13:44:52
Report written to "./log/ffmpeg-last.log"
Log level: 32
ffmpeg version 4.4.2-0ubuntu0.22.04.1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.2.0-19ubuntu1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.22.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-librsvg --enable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
2024-04-15 13:44:53,023 [   ERROR] timeout receiving frames (abstractbackend.py:441)
Traceback (most recent call last):
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/photobooth/services/backends/abstractbackend.py", line 437, in _videoworker_fun
    ffmpeg_subprocess.stdin.write(self._wait_for_lores_image())
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/photobooth/services/backends/gphoto2.py", line 176, in _wait_for_lores_image
    raise TimeoutError("timeout receiving frames")
TimeoutError: timeout receiving frames
2024-04-15 13:44:53,024 [   ERROR] Failed to create video! Error: timeout receiving frames (abstractbackend.py:442)
2024-04-15 13:44:53,025 [    INFO] leaving _videoworker_fun (abstractbackend.py:472)
2024-04-15 13:44:57,823 [    INFO] no user input within 5s so stopping video (processingservice.py:425)
2024-04-15 13:44:57,824 [    INFO] transition after: counting--(_counted)-->record (processingservice.py:226)
2024-04-15 13:44:57,825 [    INFO] exit: record from stop_recording (processingservice.py:237)
2024-04-15 13:44:57,826 [    INFO] WledService preset_standby triggered (wledservice.py:132)
2024-04-15 13:44:57,827 [   DEBUG] stop recording (abstractbackend.py:372)
2024-04-15 13:44:57,828 [    INFO] _video_worker_thread stopped and joined (abstractbackend.py:375)
2024-04-15 13:44:57,828 [    INFO] enter: captures_completed from stop_recording (processingservice.py:230)
2024-04-15 13:44:57,830 [    INFO] start postprocessing phase 2 (processingservice.py:439)
2024-04-15 13:44:57,830 [   ERROR] 'None' video not found! pls check logs (processingservice.py:66)
Traceback (most recent call last):
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/photobooth/services/processingservice.py", line 63, in _process_fun
    self._state_machine.start()
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/statemachine/event.py", line 54, in trigger_event
    return event_instance.trigger(self, *args, **kwargs)
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/statemachine/event.py", line 29, in trigger
    return machine._process(trigger_wrapper)
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/statemachine/statemachine.py", line 288, in _process
    return self._processing_loop()
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/statemachine/statemachine.py", line 303, in _processing_loop
    result = trigger()
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/statemachine/event.py", line 27, in trigger_wrapper
    return self._trigger(trigger_data)
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/statemachine/event.py", line 39, in _trigger
    if transition.execute(event_data):
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/statemachine/transition.py", line 140, in execute
    result = machine._activate(event_data)
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/statemachine/statemachine.py", line 336, in _activate
    self._callbacks_registry[target.enter].call(
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/statemachine/callbacks.py", line 228, in call
    return [
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/statemachine/callbacks.py", line 229, in <listcomp>
    callback(*args, **kwargs)
[... cut ...]
    temp_videofilepath = self._aquisition_service.get_recorded_video()
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/photobooth/services/aquisitionservice.py", line 145, in get_recorded_video
    return self._get_video_backend().get_recorded_video()
  File "/home/PixelAlarm/.local/lib/python3.10/site-packages/photobooth/services/backends/abstractbackend.py", line 385, in get_recorded_video
    raise FileNotFoundError(f"'{self._video_recorded_videofilepath}' video not found! pls check logs")
FileNotFoundError: 'None' video not found! pls check logs

mgineer85, 2024-04-15:
I think it could be because Ubuntu installation is different to Debian, maybe filter are missing…
Please also check if there is something in
Report written to "./log/ffmpeg-last.log"

it’s always only last run.

In case it breaks, it would be good to have log from before starting the job also.

PixelAlarm, 2024-04-15:
> Any errors in the browser console? Would chrome work(that is what I use to develop)?

Chrome show a different picture:
1. Video throws errors more often than Firefox (ErrorReceivingFrames)
2. When Video is being recorded correctly, it will be shown in resultscreen and gallery correctly in chrome. Also the old media is being displayed in gallery. So this seems indeed to be browser-related.

Just also tested with Firefox and chrome on macOS, preview is working there ok.

mgineer85, 2024-04-15:
> Chrome show a different picture:
>
> 1. Video throws errors more often than Firefox (ErrorReceivingFrames)
> 2. When Video is being recorded correctly, it will be shown in resultscreen and gallery [...]

So safari is the only browser with issues, right?

mgineer85, 2024-04-15:
Maybe for your cam this timeout is just too tight:   

https://github.com/photobooth-app/photobooth-app/blob/f099eabbc1cc4fca39b0fee05824972483b00de6/photobooth/services/backends/gphoto2.py#L175

self._lores_data.condition.wait(timeout=0.2):

PixelAlarm, 2024-04-15:
For Firefox in Linux, I get following log:

[Child 22427, MediaDecoderStateMachine #3] WARNING: Decoder=722f61dcda00 Decode error: NS_ERROR_DOM_MEDIA_DECODE_ERR (0x806e0004) - virtual MediaResult mozilla::FFmpegVideoDecoder<58>::DoDecode(MediaRawData *, uint8_t *, int, bool *, MediaDataDecoder::DecodedData &): avcodec_send_packet error: Invalid data found when processing input: file /build/firefox/parts/firefox/build/dom/media/MediaDecoderStateMachineBase.cpp:166

PixelAlarm, 2024-04-18:
Hi,

I think I found the problem. The mjpeg stream has a pixel format with chroma 4:2:2 (yuvj422p).
This seems not to be supported in some browsers. When not specified, ffmpeg will contain the input chroma. This was working with VirtualCamera (yuv420p) but not with the GPhoto2-Stream (yuvj422p).

After forcing yuv420p in ffmpeg (adding a parameter "-pix_fmt yuv420p") through the abstract backend, it now works on all browsers I've testet.

in abstractbackend.py:
...
"-preset",
"veryfast",
**"-pix_fmt",
"yuv420p",**
"-b:v",  # bitrate
f"{appconfig.misc.video_bitrate}k",
"-movflags",
"+faststart",

mgineer85, 2024-04-18:
Nice finding! Thank you!
In the very end of this post your suspicion is confirmed. https://unix.stackexchange.com/a/741124

Will add it to next release. Not sure if there is an impact on cpu use to encode. Will test and maybe its selectable.

mgineer85, 2024-04-18:
Despite higher cpu usage I added this one.
Will be included in next release.
Thx!
Maybe the browser at one time will be compatible with the 422 format also

closed 2024-04-18 by mgineer85

mgineer85, 2024-04-19:
Some additional context on this one:
iOS seems not to play videos other than 4:2:0 subsampling for h264.

4:2:0 can be forced by ffmpg profile base|main|high or pix_fmt yuv420p to ensure compatibility.

For now I chose pix_fmt to set (despite there are deprecation warnings in the output) so ffmpeg can choose the profile automatically.

- https://trac.ffmpeg.org/wiki/Encode/H.264#Encodingfordumbplayers
- https://de.wikipedia.org/wiki/H.264#Profile

For testing if needed, the virtualcamera can be forced to output 422 jpeg also: 
`img.save(jpeg_buffer, format="jpeg", quality=90)  # add subsampling="4:2:2" could be useful to test for #233`

mgineer85, 2024-04-19:
Updated version including a fix is released.
Please enable the compatibility mode in video settings to make it work for you.

commit 33c7e789 (2024-04-15): raise timeout for lores image a bit so backend has bit more time to deliver. might fix #233
commit d234088a (2024-04-18): fix #233