photobooth-app/photobooth-app #240: Keyboard input is not working

Striker62, 2024-04-21:
I connect via usb my keyboard to the pi.
-Test keyboard in the editor , works
enabled keyboard input in the admin view 
- c is defined for takeing the picture
- saved and restart the pi
[image]

start the browser an press c in the main view 
- nothing happend
- klick the photo button , photo taken

tested also with gpio enabled and dissabled

nothing found in the logfile

 - OS: raspberry Pi Bullseye64bit
 - Browser chrome
 - Version  2.0.2

originaly I want to use the Buzzer via bluetooth.
see Bluetooth Buzzer connection esp to raspberry #239
after the connection worked I try to use it but did not work.
therefore i did the test with the USB keyboard

mgineer85, 2024-04-21:
The buzzer emulates a keyboard. When you put the curser in any textfield and push the buzzer, you get a c?

Striker62, 2024-04-21:
yes, same as with the usb keyboard. Works

mgineer85, 2024-04-21:
Hm, so either the underlying keyboard library cannot catch the keypresses from the bluetooth stack with blueman or maybe the keyboard is not detected because it was connected after the start of the photobooth.

Could you please try to connect the buzzer, ensure it is sending keys to any other textfield and then restart the service? Does it work then?

Striker62, 2024-04-21:
It is not only the bluetooth buzzer that is not working . It is the whole keyboard capture. aas I wrote I test with an USB keyboard as well. And this doesn't also not working in the app but in the editor

Striker62, 2024-04-21:
When I use the buzzer I get a char in the editor on the pi
Same as with the usb keyboard

The app doen't catch any keyboard input

mgineer85, 2024-04-21:
I see. It did not get clear to me that keyboard is also not working.
Did you follow the installation procedure actually completely?
Especially this one is important:
https://photobooth-app.org/setup/installation/#tweak-system-settings-rpi

Please post the full logs in a file also.

Striker62, 2024-04-21:
Ok I read this when I setup first and wanted to have in mind for later. But I missed to follow this step.
USB kebbord is now working . Buzzer not. (but in the editor)

I'll do some new researches and come back to you if i can not fix it on my own.
thanks

mgineer85, 2024-04-21:
I just released a new version.
Please check if it helps for your case.
There are new settings in the config -> uisettings at the very bottom of the page. Enable this keyinput and disable the other one.
The other one will be removed later anyways probably.

Let me know if this helps for your case.

Striker62, 2024-04-21:
now every thnig is working. 
I've decoupeld and did a new pairing now the buzzer is working.

by the way can you give me a hit how I can add more buttons to the esp ?
I tryed to define 2 more buttons in the main.cpp but I miss something

closed 2024-04-21 by mgineer85