photobooth-app/photobooth-app #242: PiCam 3 - oh no - stream error :( RuntimeError, no preview from cam

AlexXLVI, 2024-04-21:
I combined Photobooth.Project with Photoboot-App (V 1.2.1) and it works - Preview an Capturing.
Now I changed from PiCam 2 to PiCam 3. Therefore I changed the resolution in config/backends/Picamera2.

Sometimes it works but the most time I get the RuntimeError.
What goes wrong?

Here is a section from log:
```
2024-04-21 22:40:46,558 [   DEBUG] loggingservice __init__ finished (loggingservice.py:114)
2024-04-21 22:40:46,559 [   DEBUG] registered handlers: [<StreamHandler <stderr> (NOTSET)>, <FileHandler /home/Alex/photobooth-data/log/photobooth_20240421.log (NOTSET)>, <EventstreamLogHandler (NOTSET)>] (loggingservice.py:115)
2024-04-21 22:40:46,560 [    INFO] platform.system()='Linux' (informationservice.py:34)
2024-04-21 22:40:46,560 [    INFO] platform.release()='6.6.20+rpt-rpi-v8' (informationservice.py:35)
2024-04-21 22:40:46,560 [    INFO] platform.machine()='aarch64' (informationservice.py:36)
2024-04-21 22:40:46,560 [    INFO] platform.python_version()='3.11.2' (informationservice.py:37)
2024-04-21 22:40:46,561 [    INFO] platform.node()='fotobox' (informationservice.py:38)
2024-04-21 22:40:46,561 [    INFO] psutil.cpu_count()=4 (informationservice.py:39)
2024-04-21 22:40:46,562 [    INFO] psutil.cpu_count(logical=False)=4 (informationservice.py:40)
2024-04-21 22:40:46,563 [    INFO] psutil.disk_partitions()=[sdiskpart(device='/dev/mmcblk0p2', mountpoint='/', fstype='ext4', opts='rw,noatime', maxfile=255, maxpath=4096), sdiskpart(device='/dev/mmcblk0p1', mountpoint='/boot/firmware', fstype='vfat', opts='rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro', maxfile=1530, maxpath=4096)] (informationservice.py:41)
2024-04-21 22:40:46,564 [    INFO] psutil.disk_usage('/')=sdiskusage(total=14698274816, used=8828321792, free=5102612480, percent=63.4) (informationservice.py:43)
2024-04-21 22:40:46,564 [    INFO] [('lo', ['127.0.0.1']), ('eth0', ['192.168.180.67']), ('wlan0', ['10.42.0.1'])] (informationservice.py:46)
2024-04-21 22:40:46,565 [    INFO] psutil.virtual_memory()=svmem(total=8189411328, available=7497850880, percent=8.4, used=499712000, free=6541418496, active=1057136640, inactive=399634432, buffers=45346816, cached=1102934016, shared=71237632, slab=75579392) (informationservice.py:55)
2024-04-21 22:40:46,565 [    INFO] __debug__=False (informationservice.py:57)
2024-04-21 22:40:46,565 [    INFO] initialized information service (informationservice.py:59)
2024-04-21 22:40:46,566 [    INFO] init database and creating missing scaled images. this might take some time. (mediacollectionservice.py:55)
2024-04-21 22:40:46,566 [    INFO] watching for filetypes: ['.jpg', '.gif'] (mediacollectionservice.py:58)
2024-04-21 22:40:46,574 [    INFO] initialized image DB, added 16 valid images (mediacollectionservice.py:74)
2024-04-21 22:40:46,574 [    INFO] -- process time: 0.01s to initialize mediacollection (mediacollectionservice.py:75)
2024-04-21 22:40:46,580 [    INFO] on_enter_state self.current_state.id='idle'  (processingservice.py:104)
2024-04-21 22:40:46,581 [    INFO] state idle entered. (processingservice.py:127)
2024-04-21 22:40:46,581 [    INFO] initialized systemservice (systemservice.py:29)
2024-04-21 22:40:46,581 [    INFO] gpio not enabled - enable for gpio support on raspberry pi (gpioservice.py:60)
2024-04-21 22:40:46,778 [    INFO] photobooth directory: /home/Alex/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth (__main__.py:43)
2024-04-21 22:40:46,779 [    INFO] working directory: /home/Alex/photobooth-data (__main__.py:45)
2024-04-21 22:40:46,779 [    INFO] app version started: 1.2.1 (__main__.py:46)
2024-04-21 22:40:46,779 [    INFO] Welcome to the photobooth-app (__main__.py:49)
2024-04-21 22:40:46,781 [    INFO] started LoggingService (container.py:90)
2024-04-21 22:40:46,781 [    INFO] WledService disabled (wledservice.py:40)
2024-04-21 22:40:46,781 [    INFO] started WledService (container.py:90)
2024-04-21 22:40:47,485 [    INFO] aquisition main backend: <class 'photobooth.services.backends.picamera2.Picamera2Backend'> (aquisitionservice.py:61)
2024-04-21 22:40:47,485 [    INFO] aquisition live backend: None (aquisitionservice.py:62)
2024-04-21 22:40:47,486 [    INFO] setting device enable_lores_flag from False to False on backend Picamera2Backend (abstractbackend.py:115)
2024-04-21 22:40:47,497 [    INFO] connect_thread device status=EnumDeviceStatus.initialized, so trying to start (abstractbackend.py:135)
2024-04-21 22:40:47,503 [    INFO] started AquisitionService (container.py:90)
2024-04-21 22:40:47,508 [    INFO] setting device status from initialized to starting (abstractbackend.py:106)
2024-04-21 22:40:47,524 [    INFO] started InformationService (container.py:90)
2024-04-21 22:40:47,524 [    INFO] trying to start backend (abstractbackend.py:140)
2024-04-21 22:40:47,539 [    INFO] started MediaprocessingService (container.py:90)
2024-04-21 22:40:47,555 [    INFO] device is available (abstractbackend.py:144)
2024-04-21 22:40:47,565 [    INFO] started MediacollectionService (container.py:90)
2024-04-21 22:40:47,643 [    INFO] started SystemService (container.py:90)
2024-04-21 22:40:47,644 [    INFO] started PrintingService (container.py:90)
2024-04-21 22:40:47,645 [    INFO] keyboardservice not enabled - enable for keyboard triggers (keyboardservice.py:54)
2024-04-21 22:40:47,645 [    INFO] started KeyboardService (container.py:90)
2024-04-21 22:40:47,645 [    INFO] started GpioService (container.py:90)
2024-04-21 22:40:47,645 [    INFO] Initialization successful. (picamera2.py:447)
2024-04-21 22:40:47,646 [    INFO] shareservice disabled, start aborted. (shareservice.py:51)
2024-04-21 22:40:47,646 [    INFO] Camera now open. (picamera2.py:468)
2024-04-21 22:40:47,646 [    INFO] started ShareService (container.py:90)
2024-04-21 22:40:47,651 [   DEBUG] change to preview mode requested (picamera2.py:245)
2024-04-21 22:40:47,651 [    INFO] FileTransferService disabled, start aborted. (filtetransferservice.py:30)
2024-04-21 22:40:47,653 [    INFO] started FileTransferService (container.py:90)
2024-04-21 22:40:47,653 [    INFO] Camera configuration has been adjusted! (picamera2.py:1054)
2024-04-21 22:40:47,654 [    INFO] started container (container.py:95)
2024-04-21 22:40:47,657 [    INFO] Configuration successful! (picamera2.py:1059)
2024-04-21 22:40:47,660 [   DEBUG] Using selector: EpollSelector (selector_events.py:54)
2024-04-21 22:40:47,691 [   DEBUG] Allocated 3 buffers for stream 0 with fds [31, 34, 37] (dmaallocator.py:64)
2024-04-21 22:40:47,702 [   DEBUG] Allocated 3 buffers for stream 1 with fds [40, 43, 46] (dmaallocator.py:64)
2024-04-21 22:40:47,707 [    INFO] Started server process [3201] (server.py:77)
2024-04-21 22:40:47,708 [    INFO] Waiting for application startup. (on.py:48)
2024-04-21 22:40:47,708 [    INFO] Application startup complete. (on.py:62)
[... 56 lines cut ...]
2024-04-21 22:40:55,126 [    INFO] setting device status from running to fault (abstractbackend.py:106)
2024-04-21 22:40:55,126 [   DEBUG] stopping (picamera2.py:175)
2024-04-21 22:40:55,127 [   DEBUG] stopped (picamera2.py:177)
2024-04-21 22:40:58,150 [   DEBUG] device timed out deliver lores image in 19 attempts. One last try now or raise exception. (abstractbackend.py:293)
2024-04-21 22:40:58,351 [   ERROR] timeout receiving frames (abstractbackend.py:300)
Traceback (most recent call last):
  File "/home/Alex/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/services/backends/abstractbackend.py", line 297, in wait_for_lores_image
    return self._wait_for_lores_image()  # blocks 0.2s usually. 10 retries default wait time=2s
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Alex/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/services/backends/picamera2.py", line 235, in _wait_for_lores_image
    raise TimeoutError("timeout receiving frames")
TimeoutError: timeout receiving frames
2024-04-21 22:40:58,353 [CRITICAL] finally failed after 20 attempts to capture lores image! (abstractbackend.py:301)
2024-04-21 22:40:58,354 [   ERROR] failing to get lores images from device is considered as error, set fault flag to recover. (abstractbackend.py:304)
2024-04-21 22:40:58,355 [    INFO] ignored to flag device as faulty because not in running mode (abstractbackend.py:242)
2024-04-21 22:40:58,355 [   ERROR] device raised exception (aquisitionservice.py:185)
Traceback (most recent call last):
[... cut ...] ckend.py", line 297, in wait_for_lores_image
    return self._wait_for_lores_image()  # blocks 0.2s usually. 10 retries default wait time=2s
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Alex/.local/pipx/venvs/photobooth-app/lib/python3.11/site-packages/photobooth/services/backends/picamera2.py", line 235, in _wait_for_lores_image
    raise TimeoutError("timeout receiving frames")
TimeoutError: timeout receiving frames

The above exception was the direct cause of the following exception:

mgineer85, 2024-04-22:
I think most likely is a hardware connection issue. Please double check the camera connected well. Maybe there is additional system log which could help and shows additional error from camera?

mgineer85, 2024-04-23:
hey you found if the Hardware is an issue?

mgineer85, 2024-04-28:
It seems this is related to the system setup, I close this.
Please reopen if needed.

closed 2024-04-28 by mgineer85