photobooth-app/photobooth-app #246: New function in collage capture

Striker62, 2024-04-24:
today I update to 2.0.5 looks good.

I try to use the booth only with bluethooth Buzzer, therfore I setup a 3 Button one.
Photo
Collage
Print

I have 2 requests:

1. can you add a config for a button to restart the photobooth-app service. The reason is that it is defined when the bluetooth buzzer is conneted to the raspberry . Most of the time the photobooth-app is 1. when this happens the Buzzer is not knowen by the photobooth-app only by the pi. With a restart button I can connenct the buzzer when ever I want and restart the photobooth-app
I would be great if you can add also a GPIO Pin for restart the Service, I than have the choice to restart via the Buzzer or a GPIO Button in the cover of the photobooth

3. You added in the Collage pipline a desicion if the photo taken is ok or not. can you add the desicion to the collage keyboard key. e.g. if the photo is ok do nothing if not press the collage key aggain. Otherwiese it would need to mre buttons one for yes one for no. That would lead for me to setup a 5 - 6 button buzzer, what is not the problem but I have to do a new 3D design.

mgineer85, 2024-04-24:
1) in 2.0.5 the keyboard is handled by the browser, so if the keyboard is connected later, this is not a problem any more.
Also to restart the app by gpio or keyboard from within is not possible per working principle unfortunately.

2) it’s just the default that changed, you can disable the approval screen in the config again.
But you are right, approval should be on existing buttons, not on new ones.

Striker62, 2024-04-24:
1. I'm ussing a buzzer via Bluetooth. for me it looks like it is not defined what service start first Bluetooth or photobooth-app
if it is the bluetooth service the photobooth-app has no info about the Buzzer. It is as it is. OK thanks. I'll check if I can use the api.
2. sorry but I can not find the switch in the admin section to disable the approval. please tell me;-)

mgineer85, 2024-04-24:
1) 
you should be able to see the keypresses from keyboard/buzzer in the browser console. If you cannot see it, the connection is not working properly. 
[image]

2) it's in admin config -> common -> collage_automatic_capture_continue -> enable that one.

Striker62, 2024-04-24:
Ok the Connection of the Buzzer is recocnized at and time 😀 Perfect.
 
Collage captureing is now working without confrimation. 😀