photobooth-app/photobooth-app #250: Approve/Decline images using GPIO/keyboard

mgineer85, 2024-04-24:
Allow more flexible configuration of gpio and keyboard input to approve/decline images.

mgineer85, 2024-07-01:
reference to another post: https://github.com/photobooth-app/photobooth-app/issues/203#issuecomment-2199200479

McKaiver, 2024-07-01:
Yes, as i said in the other topic, It would be nice, that you could controll the box with a few buttons.

Touch screens have a big dissadvantage in "Public" areas, because everybody touches it, even with dirty fingers and then there is sauce on the screen and so on, especially, because this device will be used at a party, with drunk people.

mgineer85, 2025-04-07:
So you have 3 buttons.    What are these buttons labeled like to the user?

A: Caputure Image / Approve(?)
B: Reject(?)
C: Print(?)

The api endpoints used by the touchscreen have 3 additional routes: confirm+next, reject, abort. 
The GPIO service (and for equality the keyboard input then also in the end) needs to have similar, like:
```

    def _handle_processing_next_confirm_button(self):
        try:
            self._processing_service.continue_process()
            return "OK"
        except Exception as exc:
            # other errors
            logger.critical(exc)

    def _handle_processing_reject_button(self):
        try:
            self._processing_service.reject_capture()
            return "OK"
        except Exception as exc:
            # other errors
            logger.critical(exc)

    def _handle_processing_abort_button(self):
        try:
            self._processing_service.abort_process()

        except Exception as exc:
            # other errors
            logger.critical(exc)
```

Hardest part will be the matching between the buttons and the function if the GPIO pin is already occupied by another function (like confirm and capture on one button).

mgineer85, 2025-04-22:
There are changes to the GPIO handling now made in the dev branch.

There are 3 additional gpio inputs now that can be defined:
- next
  - is to confirm the capture
  - or for a video to stop capture
- reject
  - is to reject the capture and repeat it
  - or for a video it stops the capture (but it's not rejected, just the capture is stopped and the result presented)
- abort: aborts the job completely.

With the changes added, it is possible to reuse a GPIO pin. That allows to use 2 buttons that share logic.
If no user input is required (means, no approval state), the input next/reject/abort is ignored. If the GPIO has an action assigned it will be triggered - for example create a collage.

If a user input would be required in the moment the gpio is active, it will ignore the action but continue as requested by the user (approve/reject/abort).

Hope that makes sense?

Would be great if somebody else would test the dev branch to confirm it's working as intended...

[image]