photobooth-app/photobooth-app #251: indepentent second screen to show e.g. slideshow

Striker62, 2024-04-25:
today I found the slide show function. I've spend same days to figureout how I can setup such a funtion on my own on the raspberry.
I is already there ;-).

But  my Idea was to have a second monitor that only shows a slide show in a browser window.
I tried it with the buildin slide show, at first it looks good. But when I take a picture the screen go back to the livevie as the primary screen.

Probably a second URL only for the slide show?

mgineer85, 2024-04-25:
https://photobooth-app.org/reference/foldersandurls/#special-urls
?

Striker62, 2024-04-25:
Yes I found it, but as I wrote, id you take a Photo the url Switch to the Main view. What I'm looking for is That the slideshow is indepedant from the actions on the booth

Striker62, 2024-04-25:
what I can imagine is a seperat service for the slideshow with its own url.

mgineer85, 2024-04-25:
http://IP:PORT/#/standalone/slideshow
What about that uri?

Striker62, 2024-04-25:
ok looks perfrect . have I overseen it or where is it documented. 
Is it possible to maintain a css for it ?

mgineer85, 2024-04-25:
I added id's to all pages and layouts. this way you are able to use the private.css and select elements on the standalone layout without affecting the normal pages: https://github.com/photobooth-app/photobooth-frontend/commit/f0240f16821b3b58b9d34837e2e302082373ff67

closed 2024-04-25 by mgineer85

Striker62, 2024-04-25:
you are the best . thanks