photobooth-app/photobooth-app #253: Live Preview doesn't fill screen anymore

PixelAlarm, 2024-04-30:
**Describe the bug**
Since a few releases (maybe since implementing the preview-scaling-option) the preview won't fill the screen anymore.

Previous behavior:
The preview was stretched to fill the whole screen and not display any borders.
The preview was fixed in the middle.
With smaller height, the video was cropped in height and display full width.
With smaller width, the video was cropped in width and display full heigth.

Actual behavior:
The preview isn't cropped and always contains the original aspect ratio.

**To Reproduce**
Enable live preview and resize the window of firefox

**Expected behavior**
Stretch video to fill the whole screen and not display any borders.

**System (please complete the following information):**
Version 2.0.6 fresh install
Firefox and Chrome

PixelAlarm, 2024-04-30:
I added the line in the private.css
`(background-size: cover;)`

Now it's working again. Was it meant to be taken away in commit 09124a2?

mgineer85, 2024-04-30:
Yes I changed it because it will hide a part of the image while people position.
Any idea how to handle that better?

PixelAlarm, 2024-05-05:
I thought the "cover"-setting was a good one for all landscape oriented displays, since the positioning in front of the screen often depends on the width of the scene and not on the height.
(people standing next to each other instead of in front of each other.)

For portrait oriented screens, it's suboptimal no matter what (small image vs. cropped image)

I will just force the old functionality through the private.css for my setup.

mgineer85, 2024-05-05:
I added a section to the documentation how to adjust: https://photobooth-app.org/reference/customizetheme/#adjust-livestream-to-cover-the-full-screen

There are some cameras like the pi camera 3 that have very wide aspect ratio, so contain could be good choice also.

closed 2024-05-05 by mgineer85