photobooth-app/photobooth-app #255: Canon Selphy 1200 not excepting files

DIY89, 2024-05-01:
**Describe the bug**
Canon Selphy 1200 not excepting precessed files of photobox but printing own background files.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to media folder `/Photobooth/photobooth-app/media/processed/preview`
2. Type in manual print command `lp -d Canon_SELPHY_CP1200_USB -o landscape -o fit-to-page collage_show_20240501-215510-723490.jpg` 
3. See error: [image]

**Expected behavior**
Proper working printing

**Screenshots**
[image]

**System (please complete the following information):**
 - OS:  Raspberry Pi 4 Bookworm 64bit
 - Browser Firefoy
 - Version [e.g. 22]

**Additional context**
Printer is working with own pictures uploaded to the Raspi.
Maybe something wrong with MetaData/Resolution ...?

DIY89, 2024-05-02:
Not printable picture because `Encoding Process                : Baseline DCT, Huffman coding  `

`exiftool collage_show_20240501-133832-896857.jpg`
`ExifTool Version Number         : 12.84`
`File Name                       : collage_show_20240501-133832-896857.jpg`
`Directory                       : . `
`File Size                       : 339 kB `
`File Modification Date/Time     : 2024:05:01 13:56:04+02:00 `
`File Access Date/Time           : 2024:05:01 21:07:27+02:00 `
`File Creation Date/Time         : 2024:05:01 13:56:04+02:00 `
`File Permissions                : -rw-rw-rw- `
`File Type                       : JPEG `
`File Type Extension             : jpg ` 
`MIME Type                       : image/jpeg `
`JFIF Version                    : 1.01`
`Resolution Unit                 : None  `
`X Resolution                    : 1  `
`Y Resolution                    : 1  `
`Image Width                     : 1575  `
`Image Height                    : 1050  `
`Encoding Process                : Baseline DCT, Huffman coding  `
`Bits Per Sample                 : 8  `
`Color Components                : 3  `
`Y Cb Cr Sub Sampling            : YCbCr4:2:0 (2 2)  `
`Image Size                      : 1575x1050  `
`Megapixels                      : 1.7 `

Printable picture because `Encoding Process                : Progressive DCT, Huffman coding `
`exiftool Hintergrund_1920x1280.jpg `
`ExifTool Version Number         : 12.84`
`File Name                       : Hintergrund_1920x1280.jpg`
`Directory                       : .`
`File Size                       : 920 kB`
`File Modification Date/Time     : 2024:05:01 21:52:09+02:00`
`File Access Date/Time           : 2024:05:01 22:10:50+02:00`
`File Creation Date/Time         : 2024:05:01 21:52:09+02:00`
`File Permissions                : -rw-rw-rw-`
`File Type                       : JPEG`
`File Type Extension             : jpg`
`MIME Type                       : image/jpeg`
`JFIF Version                    : 1.01`
`Exif Byte Order                 : Little-endian (Intel, II)`
`Image Description               : Created with GIMP`
`Orientation                     : Horizontal (normal)`
`X Resolution                    : 300`
`Y Resolution                    : 300`
`Resolution Unit                 : inches`
`Software                        : GIMP 2.10.36`
`Modify Date                     : 2024:05:01 21:52:06`
`User Comment                    : Created with GIMP`
`Color Space                     : sRGB`
`XMP Toolkit                     : XMP Core 4.4.0-Exiv2`
`Document ID                     : gimp:docid:gimp:b9f0e5af-a759-42d1-a50a-7e1f9ce7b77a`
`Instance ID                     : xmp.iid:282a485c-b968-4035-9eac-c0544d65d340`
`Original Document ID            : xmp.did:30fa4bd4-2ae8-427d-a862-d163357a2e44`
`Format                          : image/jpeg`
`Api                             : 2.0`
`Platform                        : Windows`
`Time Stamp                      : 1714593129543545`
`Version                         : 2.10.36`
`Creator Tool                    : GIMP 2.10`
`Metadata Date                   : 2024:05:01T21:52:06+02:00`
`History Action                  : saved`
`History Changed                 : /`
`History Instance ID             : xmp.iid:c88d7467-ba97-41ce-bd25-2e98240b095d`
`History Software Agent          : Gimp 2.10 (Windows)`
`History When                    : 2024:05:01 21:52:09`
`Profile CMM Type                : Little CMS`
`Profile Version                 : 4.4.0`
`Profile Class                   : Display Device Profile`
`Color Space Data                : RGB`
`Profile Connection Space        : XYZ`
`Profile Date Time               : 2024:05:01 11:58:10
`Profile File Signature          : acsp
`Primary Platform                : Microsoft Corporation
`CMM Flags                       : Not Embedded, Independent
`Device Manufacturer             :
`Device Model                    :
`Device Attributes               : Reflective, Glossy, Positive, Color
`Rendering Intent                : Perceptual
`Connection Space Illuminant     : 0.9642 1 0.82491
`Profile Creator                 : Little CMS
`Profile ID                      : 0
`Profile Description             : GIMP built-in sRGB
`Profile Copyright               : Public Domain
`Media White Point               : 0.9642 1 0.82491
`Chromatic Adaptation            : 1.04788 0.02292 -0.05022 0.02959 0.99048 -0.01707 -0.00925 0.01508 0.75168
`Red Matrix Column               : 0.43604 0.22249 0.01392
`Blue Matrix Column              : 0.14305 0.06061 0.71393
`Green Matrix Column             : 0.38512 0.7169 0.09706
`Red Tone Reproduction Curve     : (Binary data 32 bytes, use -b option to extract)
`Green Tone Reproduction Curve   : (Binary data 32 bytes, use -b option to extract)
`Blue Tone Reproduction Curve    : (Binary data 32 bytes, use -b option to extract)
`Chromaticity Channels           : 3
`Chromaticity Colorant           : Unknown
`Chromaticity Channel 1          : 0.64 0.33002
`Chromaticity Channel 2          : 0.3 0.60001
`Chromaticity Channel 3          : 0.15001 0.06
`Device Mfg Desc                 : GIMP
`Device Model Desc               : sRGB
`Comment                         : Created with GIMP
`Image Width                     : 1920
`Image Height                    : 1280
`Encoding Process                : Progressive DCT, Huffman coding
`Bits Per Sample                 : 8
`Color Components                : 3
`Y Cb Cr Sub Sampling            : YCbCr4:4:4 (1 1)
`Image Size                      : 1920x1280
`Megapixels                      : 2.5

Is there any chance to change the encoding process during the creation of pictures?

mgineer85, 2024-05-02:
If you have the command ready in python I can implement a compatibility layer probably.
Can you reduce it to 1 or 2 lines python code with which ensures proper encoding?

mgineer85, 2024-05-02:
In my opinion it’s more likely because of the subsampling Y Cb Cr Sub Sampling            : YCbCr4:2:0 (2 2)

DIY89, 2024-05-02:
I think it is somewhere around here in the `mediaprocessingservice.py` 
`## create mediaitem`
`        canvas = canvas.convert("RGB") if canvas.mode in ("RGBA", "P") else canvas`
`        filepath_neworiginalfile = get_new_filename(type=MediaItemTypes.collage)`
`        canvas.save(filepath_neworiginalfile, format="jpeg", quality=appconfig.mediaprocessing.HIRES_STILL_QUALITY, optimize=True)`
`        mediaitem = MediaItem(os.path.basename(filepath_neworiginalfile))  # instanciate mediaitem with new original file`

Tried a few things but didn't work yet:
[[Color space conversion from YCbCr to RGB using PIL library?](https://stackoverflow.com/questions/59332411/color-space-conversion-from-ycbcr-to-rgb-using-pil-library)](https://stackoverflow.com/questions/59332411/color-space-conversion-from-ycbcr-to-rgb-using-pil-library)

mgineer85, 2024-05-02:
You could just test it separately , operate directly on the files.
On the other side, did you use a custom ppd for your printer? The ones delivered with the system can be quite old and limited functionality.

DIY89, 2024-05-02:
Just created a small script to test the subsampling 

`import io`
`import logging`
`import os`
`import time`
`from typing import Union`
`from PIL import Image`

`canvas_size = (1920, 1280)`
`canvas = Image.new("RGBA", canvas_size, color=None)`
`print("Canvas mode: ", canvas.mode)`
`canvas = canvas.convert("YCbCr")`
`print("Canvas mode: ", canvas.mode)`
`canvas.save("test.jpg", format="jpeg", quality=90, subsampling=0, optimize=False)`

and it creates an file with the correct subsampling

`exiftool test.jpg`
`ExifTool Version Number         : 12.84`
`File Name                       : test.jpg`
`Directory                       : .`
`File Size                       : 68 kB`
`File Modification Date/Time     : 2024:05:02 15:05:07+02:00`
`File Access Date/Time           : 2024:05:02 15:05:07+02:00`
`File Creation Date/Time         : 2024:05:02 14:45:16+02:00`
`File Permissions                : -rw-rw-rw-`
`File Type                       : JPEG`
`File Type Extension             : jpg`
`MIME Type                       : image/jpeg`
`JFIF Version                    : 1.01`
`Resolution Unit                 : None`
`X Resolution                    : 1`
`Y Resolution                    : 1`
`Image Width                     : 1920`
`Image Height                    : 1280`
`Encoding Process                : Baseline DCT, Huffman coding`
`Bits Per Sample                 : 8`
`Color Components                : 3`
`Y Cb Cr Sub Sampling            : YCbCr4:4:4 (1 1)`
`Image Size                      : 1920x1280`
`Megapixels                      : 2.5`

Tried to insert the same thing `subsampling=0` in `mediaprocessingservice.py` but this is not working

mgineer85, 2024-05-02:
But the test.jpg would print on your printer with no issue?

DIY89, 2024-05-02:
Tried several picture conversions but a least found out that *.png is working.
Is there a chance to create png files instead of jpg?

mgineer85, 2024-05-02:
You could write a bash script and use imagick to convert and afterwards call the print command.

DIY89, 2024-05-03:
Tried to chance a ghost. Used another printer driver. Was able to choose the new driver after installation of `cups-driver-gutenprint`. Sorry for wasting your time and thank you for support.

closed 2024-05-05 by mgineer85