photobooth-app/photobooth-app #259: Error - when I select a filter

Louloubiwan, 2024-05-16:
Hello :D
When I try to pick any filter in the dropdownlist, I get this error :

check following fields:
body -> uisettings -> gallery_filter_userselectable -> 0: Input should be 'original', '_1977', 'aden', 'ashby', 'amaro', 'brannan', 'brooklyn', 'charmes', 'clarendon', 'crema', 'dogpatch', 'earlybird', 'gingham', 'ginza', 'hefe', 'helena', 'hudson', 'inkwell', 'juno', 'kelvin', 'lark', 'lofi', 'ludwig', 'maven', 'mayfair', 'moon', 'nashville', 'perpetua', 'poprocket', 'reyes', 'rise', 'sierra', 'skyline', 'slumber', 'stinson', 'sutro', 'toaster', 'valencia', 'walden', 'willow' or 'xpro2'

**To Reproduce**
1. Go to admin -> uisettings.
2. Click on Pic1 Filter Userselectable.
3. Choose any filter.
4. The Error appear.

**Expected behavior**
To enable the filter that I have choosen

**System :'**
 - OS: Raspberry Pi Bookworm 64bit
 - Browser chromium
 - Version 22

mgineer85, 2024-05-16:
Which app version you used? I think there was an issue in older version. You need to choose at least one value I think.

mgineer85, 2024-05-17:
new version 2.0.7 is released and should fix it.

closed 2024-05-17 by mgineer85

Louloubiwan, 2024-05-17:
Hello, I confirm that it's working, thank you for your fast reply.
Have a nice day