photobooth-app/photobooth-app #260: Printing not Working - „GET /api/print/…/0 HTTP/1.1“ 404

PixelAlarm, 2024-05-18:
**Describe the bug**
Printing Not working on 3.0a2
Log says „GET /api/print/…/0 HTTP/1.1“ 404 

Printing is working via Terminal.

**To Reproduce**
Steps to reproduce the behavior:
1. setup Printer via cups 
2. Print From Gallery

mgineer85, 2024-05-18:
The endpoint is slightly different:
http://localhost:8000/api/doc
[image]

mgineer85, 2024-05-18:
First and second print latest, the third prints specific id.

PixelAlarm, 2024-05-18:
So the Print Button in Gallery or result screen will lead to an error in newest Versions.

I don‘t Know Anything about the API.

I just want to trigger the Print from the screen.

PixelAlarm, 2024-05-18:
The full line is:

2024-05-18 16:27:33,828 [    INFO] 127.0.0.1:47404 - "GET /api/print/b357cb2ee31861f3c789be61a0cc01be/0 HTTP/1.1" 404 (httptools_impl.py:481)

mgineer85, 2024-05-18:
Dang, very correct. Updated in https://github.com/photobooth-app/photobooth-frontend/commit/a8caa659b2f15aa2695d92ce5b806703eae40ab9

Will be fixed in next release.

PixelAlarm, 2024-05-18:
Any way to fix it manuelly?

I want to use the photobooth today :)

mgineer85, 2024-05-18:
Maybe not the best idea to use a prerelease version if urgently needed... ;)

I just released a new version, wait until its done and you can update as usual: https://github.com/photobooth-app/photobooth-app/actions/runs/9140662352

closed 2024-05-18 by mgineer85