photobooth-app/photobooth-app #262: Disabling "Show Takecollage/Takeanimation/Takevideo On Frontpage" has no effect

foujax, 2024-05-19:
**Describe the bug**
Seems like if i decide to hide the Collage, Animation and Video buttons on the frontpage using the admin settings (uisettings), I still see the buttons even after an app restart or a system restart

**To Reproduce**
1. Go to admin/uisettings
2. Turn off the settings (Takecollage/Takeanimation/Takevideo) and save
3. Go back to the front page. The buttons are still there

**Expected behavior**
The buttons not being there... I guess...

**System (please complete the following information):**
 - Raspberry Pi Bookworm 64bit
 - Chromium
 - Photobooth-app 3.0.0

**Additional context**
Note that the admin and gallery buttons are correctly disabled. Maybe I'm missing something here... Thanks!

mgineer85, 2024-05-19:
Oh, Right. That’s probably a bug.
In v3 you can remove the buttons in the actions settings by removing the icon and label per each configuration set.

mgineer85, 2024-05-19:
v3.0.1 is in the release process.
The old settings are removed. 

Instead there is a new setting "show button" in the configuration now. Earlier it was just assumed, if you clear the icon and the title to not display the button, but a separate setting is more reasonable:
[image]

closed 2024-05-19 by mgineer85

foujax, 2024-05-20:
Well that was fast!
Thanks a lot! Seems to be working!

commit da3639c4 (2024-05-19): fix #262 in backend